dresden-elektronik/deconz-rest-plugin #3135: Phoscon: Scene that turns off lights does nothing

Cornelius-Riemenschneider, 2020-08-13:
For one of my groups I have several scenes, all of them except one have the lights turned off.
Recalling these scenes works fine.
However, when recalling the scene that turns off all the lights nothing happens, and the lights stay at whatever state they were.
Manually turning on and off the lights via the on/off switch in the GUI works, though.

ChrisPrefect, 2020-08-25:
Is there any update available on this? Will it be fixed some day? Scenes are only about 30% useful if lights can't be turned off. I really hope this gets solved soon. Thanks!

f0086, 2020-09-04:
Any update on this? It is very annoying that I can't use the scenes at all. It is pretty useless for me if I can't turn off lights.
Is there any further data/test/log I can provide to help debug (and finally fix) this issue?

Mimiix, 2020-09-06:
Forwarded it.

f0086, 2020-10-09:
Any insights so far @Mimiix ?

espen4001, 2020-10-13:
Have same issue. When versting a scene in phoscon webapp, not all lights are turned off Even if set to minimum in scene.
Wanted to figure out why, and did an API-call to get the scene data from DeConz. Turned out that some lights in my scenes are set to brightness=1 and on=true. And some others are set to brightness=<whatever> and on=false.
Those who are set to on=false, are always turned off when recalling the scene. Those with brightness=1 and on=true are randomly on or off.
Did an API PUT to modify scene and changed those with brightness=1 and on=true to on=false.
So far my testing shows that this solved the issue.
So my conclusion is that the webgui for creating scenes in phoscon is not always able to set the correct value of lights. Maybe there should be a separate option to select to switch light off, instead of just setting brightness to 0%?

Mimiix, 2020-10-20:
@f0086 This one somehow fell trough my net. Sorry! I monitor all issues, this one slipped :( 

I pmed @manup again.

mmaraa, 2020-10-31:
Same problem here 😔 Wondered, should I move scenes away from deconz to home assistant itself, but perhaps I'll wait a moment before doing the migration.

manup, 2020-11-01:
For the on=true issue, can you please check that the sliders are really on the bottom, the icons on the main page should indicate that the lights are indeed off=false.

Related: https://github.com/dresden-elektronik/deconz-rest-plugin-v2/issues/5

ChrisPrefect, 2020-11-01:
Yes, the sliders really are at 0%. And the scenes have been saved multiple times after pulling each light to 0% again.

But please: Add "real" switches to the On/Off state and not rely on the sliders in Phoscon. I often am not able to reach 0% or 100%, due to resolution limits of the browser window I guess. There should be a off/on (100%) switch for each light and each group. Or how do you turn on or off groups in the mobile version of Phoscon?

manup, 2020-11-01:
Have you tried to use the light symbols, they act as on/off switches?

sorenjacobi, 2020-11-10:
I have the same problem. If I turn off the lights with the symbol and saves the scene, the same thing happens: I can turn all lights on to 100% with "turn on" scene, but when I try the "turn off" scene, IKEA colour bulbs stay lit. If I look into deconz, the IKEA bulbs are shown as turned off.

If I then try to turn the entire group on, by clicking on the group bulb icon, all lights will turn on but at the minimum light level. The "turn on" scene still works to get them to 100%.

f0086, 2020-11-23:
@manup I don't think this is a user problem (sliders not dragged to the bottom, not tried to turn the bulb completely off by hitting the bulb symbol, etc.) as proved by a lot of people here. When I set up the scene with lights off, the lights turn off as it should and the bulb turned gray. When I save the scene, switch all lights on and recall the scene, the lights do not turn off.

This is a very annoying bug and makes the scenes unusable :cry: 
PLEASE give us some instructions how to debug this further. What information do you need to reveal the root cause?

I will do the same as @mmaraa if this bug is not taken seriously.

gongevangen, 2020-11-28:
I have the same problem, 3 out of 10 Ikea bulb's I cannot turn off. Really make scenes useless

FreshJonte, 2020-12-08:
I am having a simular issue with the ikea color bulbs as well. When toggling the light off through my dashboardnsometimes the state changes to off but the light is still on - Deconz also shows off. Super anoying when lights are not turning off reliably.

JurjenBraam, 2020-12-15:
I too face the same issue where my lights do turn on correctly with a scene but only one of the lights won't turn off with a scene. 

Work correctly:
6x TRADFRI bulb GU10 WW 400lm 
1x TRÅDFRI bulb E27 CWS opal 600lm
2x TRÅDFRI control outlet

Won't work correctly:
1x TRADFRI bulb E27 WS opal 1000lm

The Phoscon and Homekit state is that the light is off however it's still turned on..

Drumdevil, 2020-12-24:
Just got the Conbee II stick and have the same problem with my IKEA bulbs.

tjfoth, 2020-12-26:
Have to agree... this is broken for me.  I'm using Ikea switches, too, with gledopto LED strip controllers.  Note that if I use the old web app and press the Off button for the LED strips, the lights actually go completely off.

In the pwa, if you set the brightness to 0 (and the lights go out) and then save the scene, the brightness pops up to 1.

espen4001, 2020-12-27:
I want to point out the need for a change in the scene settings page to be able to select if a light should be on or off, in addition to set the brightness to any value between 1 and 255. As the scene report from DeConz is inconsistent in the light state report on="true" or on="false" when setting the brightness slider to a minimum.
Would recommend a switch to be able to set the light to be on or off, and this switch should override the brightness slider setting if set to "off".

eSoares, 2021-03-02:
I'm having the same issue.

I have in scenes two lightbulbs from IKEA TRÅDFRI:
- TRÅDFRI bulb E27 CWS opal 600lm (version 1.3.002) : works correctly with scenes to turn on or off
- TRADFRI bulb E27 WS opal 1000lm (version 2.0.022) : unable to turn off

My version of deconz is 2.10.00 (12/23/2020) and gateway firmware 26660700.

After clicking in a scene to turn light off the light stays on but says is off in the UI. Turning light on and then off in the UI changes the light state to off. 

Maybe the problem is with the fade out to off?
If more logs are needed, feel free to ask.

keks, 2021-03-29:
BTW I have the same issue with the official IKEA gateway and app, so it might not be your fault (not asking you to fix my situation, just a heads up that you are not alone and it could be an upstream issue)

falkenbt, 2021-04-06:
Same issue here with some Ikea bulbs, would really appreciate any hints or workarounds that could apply.

erik3rik, 2021-04-29:
I have the same issue. I have both IKEA TRÅDFRI and Philips Hue lights. It works fine to have the Philips Hue lights off in a scene (and recalling that scene will turn the light gray in the Phoscon app and the light will physically turn off as expected). 

However, for the IKEA lights, the Phoscon app will also display them as gray and turned off but in reality the lights are actually on.  It happens EVERY time for the IKEA lights, it never happens with the Philips Hue lights.

If I connect the IKEA lights to the IKEA TRÅDFRI GATEWAY, and create scenes in the IKEA app that includes lights that are off, it works as expected to recall the scenes.

To summarize:
- It is not really the Phoscon GUI that is problematic. It displays the (wanted) state correctly; with the lights that are supposed to be turned off as grayed out.
- It is not really the GUI or any general property of the ConBee II, as it works every time for the Philips Hue lights.
- It is not really the IKEA lights, as they do turn off when included with an "off state" in the scene as expected when connected to the IKEA GATEWAY.
- But it never works with the IKEA lights if they are connected to the ConBee II / is configured to be off in a scene in the Phoscon app.

So there must be something different that the Phoscon app / ConBee II does when interacting with the IKEA lights compared to what the native IKEA TRÅDFRI GATEWAY does. I have no idea if that difference is IKEA not following the standards, or have glitchy firmware, or if it is the deconz / ConBee / Phoscon app that does not do things right. However, since a lot of people obviously has serous issues with this, perhaps a compensation or workaround for this glitchyness could be added to the deconz / ConBee / Phoscon thingy as IKEA will probably not fix anything. They are good at making cheap furniture, but not at making good software.

MagnusKazen, 2021-05-01:
Hi.
Got similar problems that appeared quite recently after using Deconz to upgrade the firmware OTA of my Tradfri lightbulbs.
Started with just one light in one room but then spread. Found this thread searching for answers.
So I did some digging. 
I'm no programmer but work as a technician finding and solving hardware problems in cameras and lenses etc. 
I use a Conbee stick in my Home Assistant setup connected to Coogle Assistant through HA cloud (Nabu Casa).
Hope my findings can be of some use for someone to dig further.

The problems emerged when my tradfri lights were updated to latest FW through Deconz OTAU.
Two lamps with the same model, TRÅDFRI bulb E14 opal 400lm, one with FW 1.2.221 and one with 2.3.050, the old FW worked, the new FW didn't. While testing, the old FW was updated and stopped working. 

With Google Assistant I command to set a light at a specific percentage, (Set readinglamp to 30%), and it also works to set the lamp to 0%, that will turn the light off. 
I can also increase a light with a percentage or to a percentage. These commands seem to be handled differently. 
When i Increase reading light to 30% it will stay on 30% as expected, while increase with 30% will do just that.
The opposite, decrease to 0% will turn the light off while decrease with 100%, or whatever percentage it was lit with, will decrease the light to 1% but not turn it off. However, this goes for any light, even non-Tradfri and also Tradfri with old FW, 

If I do the same from Home Assistant it will result in a turned off light, but there the command for brightness and the command for color is sent separately, thus the color is changed and then the brightness is adjusted. If I add a transition time of one second, this will change only the color, the brightness will never change. Seems the command for changing the color is blocking the command for brighness. (Actually one reason to keept the scenes withing the conbee bridge.)

So far there is nothing really peculiar here, apart from the decreace with 100% that will leave the light at 1%. 
 
Moving to the Phoscon app and scene settings, the results are a bit different.
Sending a scene command changing both brightness and color will do that, both, so that command is not sent the same way as from HA. Also the transition is smooth for both color and brightness. 
I've tested this with another lamp, a TRADFRI Bulb E27 WS Clear 950lm.
The only issue is that the slider for the color temperature is not always going to the preset position but really close. I tested sliding it to a warm white as possible but when recalling that scene, the slider was about 10% from the end. 

Now sending a scene command where the brighness is 0 and the color is the same will result in nothing. 
The qustion is what actually is sent, and why the new FW can't interprate it correctly while the old could. 
One possibility is that the newer FW regards a command containing both a color and a 0 brightness as invalid since the reaction is nothing, which acutally ought to be correct, A no-light at all can't really have a color. 

The solution in my humble opinion should be to correct the scenes to not send a color when the brightness is set to 0.
Actually, when wishing for changes, it should be possible to choose if both color and brightnes or only one of those parameters should be changed by the scene. And a possibility to enter a value for those settings to make it easier to set to lamps beside each other to the same values in a scene.

espen4001, 2021-05-02:
> Hi.
>
> Got similar problems that appeared quite recently after using Deconz to upgrade the firmware OTA of my Tradfri lightbulbs.
>
> Started with just one light in one room but then spread. Found this [...]

I have similar issues after performing OTA updates to my IKEA lights. Upgraded to FW 2.0.029. Scene recalls does not work at all. But individual control of each light works fine.
I have created my scenes by manually creating them through API-POST commands, to make sure the states are correct.
As I have mentioned earlier in this thread, creating them through the Phoscon GUI doesnt always set the states correctly when wanted state is "off".

MagnusKazen, 2021-05-27:
> I have similar issues after performing OTA updates to my IKEA lights. Upgraded to FW 2.0.029. Scene recalls does not work at all. But individual control of each light works fine.
> I have created my [...]

Using API-POST commands sounds as nice way to go, is there some information available on how to do that for a total rookie?

woutput, 2021-07-10:
FYI: I use Tradfri GU10 spots with IKEA's Home smart app, also in this "normal" case the default "All off" scene doesn't work correctly: it only turns off 2 of my 14 lights. Also removing & re-adding the lights doesn't solve this.
So I think there is a big chance that the cause is in the IKEA FW/SW itself.

nicobuelk, 2021-07-24:
Hi,

I have similar problems but after updating the ConBee II firmware to 266B0700 and deconz to 2.12.03 the behaviour changed.
My TRADFRI light bulbs also received an firmware update. I am only listing the bulbs that are causing problems after the update:
- 1000lm WS E27 -> 0x20029623
- GU10 W -> 0x23050631
- Driver for built in lights (hp and lp) -> 0x23068631

Now activating a scene does not turn the light ON. Turning the light off by activating a scene works fine. Also the color temperature of the 1000lm WS E27 bulbs does not change with a scene change.

Both the 1000lm and the GU10 bulb can be controlled individually. The driver for built in lights does nothing even when controlled individually.

woutput, 2021-07-24:
> FYI: I use Tradfri GU10 spots with IKEA's Home smart app, also in this "normal" case the default "All off" scene doesn't work correctly: it only turns off 2 of my 14 lights. Also removing & [...]

A firmware update (released somewhere during the last two weeks) by IKEA solved the problem for my configuration.

fillilutten, 2021-08-04:
I still have this problem with my "TRÅDFRI bulb E27 CWS opal 600lm". Running fw version 1.3.013.

nicobuelk, 2021-08-06:
For me nothing changed. I am checking for firmware updates almost every day. Is there a way of downloading older firmware?

wischnu, 2021-12-13:
As I still have this issue, I downgraded my TRADFRI lamps to the original firmware. Here is a link to older firmwares. 
https://github.com/basilfx/TRADFRI-Hacking/tree/master/firmwares/ikea/otau

Now I have the issue that some lamps are not turning on with the scenes... Still better like this.

wischnu, 2021-12-17:
> Using API-POST commands sounds as nice way to go, is there some information available on how to do that for a total rookie?

If someone is interested I could post some python lines on how I automated it using the REST Websocket interface. In the scenes all lamps with brightness of 1 will be turned off automatically by the script. It is a workaround, but it works quite well.

falkenbt, 2021-12-18:
I would be interested! When / how is the script called?

wischnu, 2021-12-18:
Condition: 
Scenes that call a TRADFRI lamp to turn off do not run correctly. All other scenes where the lamps stay "on" at whatever brightness work perfectly.

Solution: 
I have put together a simple script that checks if a lamp has been switched to brightness 1 (lowest possible brightness). If yes, then the script turns off the lamp(s). By this, you can create working scenes with the tradfri lamps where a brightness of 1 equal to an "off" state. 

For the script to run, you need to have python available. Then install the deconzpy library (sudo pip3 install deconzpy) that handles the websocket interface with the Deconz server.

The config is a config.json file at the location with the script with the following content: 
{"gatewayIP": "1.1.1.1", "username": "user"} 

The user is the API Access token that you can get from the Phoscon help page under API Information.

When switching the scene, the deconzpy library throws an exception, but it still works.

```
from deconzpy import Router
import time

router = Router() # Router is a singelton, can be called in multiple places and will return the same Router Object
router.startAndRunThread() # subscribe to websocket (for updates)

def onChange_Light(light, key_that_changed, old_bri, new_bri): # Acts if brightness of light changes
    print("Brightness switched from: " + str(old_bri) + " to: " + str(new_bri))
    if new_bri == 1: #if brightness is switched to 1 turn off light
        time.sleep(1)
        light.actionOff()

Light1=router.getLightByName("IKEA Tradfri Lamp 1") # subscribe to brightness changes of the defined light
Light1.subscribeToAttribute("state_bri", onChange_Light)

while True: # run main thread in a loop
    time.sleep(5)
```

manup, 2022-01-13:
Side since a while deCONZ contains a small workaround to send unicast "Off with effect" commands in addition to the group scene command to lights which are "off" in the scene, in my tests this was the only way to turn the damn lights off, since they wouldn't even react to normal "off" commands.

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/4209428e9c428271ca0ddc1175ef5469b4be7470/rest_groups.cpp#L2704-L2717

I have this scenario in my bathroom with 4 Ikea GU10 where in one scene 2 lights are off, without the workaround the 2 off lights won't turn off (which used to work in older Ikea firmware versions). Usually the workaround is applied automatically as long as the light state in the REST API scene is `false`.

easybeat, 2023-07-20:
Was this bug ever fixed?

I have a similar problem with new lights added to my system that I just can't setup them in new scenes. It looks like the scene is just not stored into the light at all. Any hints on this?