dresden-elektronik/deconz-rest-plugin #3139: Zemismart New Design Zigbee 3.0 Wall Push Light Switch

allanlopesprado, 2020-08-13:
Product name: Zemismart New Design Zigbee 3.0 Wall Push Light Switch SmartThings Control US AU Physical Switches
Manufacturer: FeiBit (ZemisSmart)
Model identifier: FNB56-ZSW01LX2.0
Device type : Switch (One Gang)
Link: [https://www.zemismart.com/products/zemismart-au-wifi-wallush-light-switch-alexa-google-home-enable-smart-life-app-controlone-gang-two-gangs-three-gangs](Device)

**Screenshot of the Basic cluster**
[image]

**Screenshot of the Node Info Panel**
[image]

Smanar, 2020-08-14:
But this device don't work natively ?

allanlopesprado, 2020-08-14:
> But this device don't work natively ?

Yes it works, but it is being recognized as Light and not as Switchs, is that correct?

[image]

Smanar, 2020-08-14:
No it s normal ^^.

It s a long story, with different idea, my personnal explanation is switches are actuators and light is receiver. With switch you can send command, but your device receive order to open or close the circuit.

Battery switch are sensor/switch.
Wired switch are light.

allanlopesprado, 2020-08-18:
Perfect, so we can close the request

Thanks for your time and explanation

closed 2020-08-18 by allanlopesprado