dresden-elektronik/deconz-rest-plugin #3142: Non stop logging (Home Assistant)

ASNNetworks, 2020-08-14:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi 4 (4GB)
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26580700
 - deCONZ version: 2.05.79 / 22-05-2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes 

### deCONZ Logs

## Additional context

Mimiix, 2020-08-14:
AS far as i knew, we were going to remove this message. I just did a PR to do this. Thanks for the headsup!

Also: You should check on the device with mac: 00158D0004B275E8 

It gives a error:
`11:52:09:616 0x00158D0004B275E8 error APSDE-DATA.confirm: 0xE9 on task`

For reference on this: 
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log

ASNNetworks, 2020-08-14:
@Mimiix thanks for your quick reply!

The error you mention is one of two INNR E14 lightbulbs. I also saw 0x00158D0004B2EA6D  as an error, which is the second E14. I removed both and paired them again, restarted HA. Still gives that line though... How can I debug what that error is?

Mimiix, 2020-08-14:
Your welcome. My PR: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3144

The 0x00 is not a error , the 0xE9 on task is the error. The 0x00 line is the MacAdress of the device.

This seems like a connection issue but it is hard to tell what the issue is. Do you have discord? It might be easier to help you there

ASNNetworks, 2020-08-14:
> Your welcome. My PR: #3144
> 
> The 0x00 is not a error , the 0xE9 on task is the error. The 0x00 line is the MacAdress of the device.
> 
> This seems like a connection issue but it is hard to tell what the issue is. Do you have discord? It might be easier to help you there

I'm sorry, I meant I saw the same error for that device. So:

13:21:15:113 0x00158D0004B2EA6D error APSDE-DATA.confirm: 0xE9 on task
13:21:33:712 0x00158D0004B275E8 error APSDE-DATA.confirm: 0xE9 on task

I have Discord yes. Can I add you somehow or go to a specific server?

Mimiix, 2020-08-14:
There we go! 
https://discord.gg/KgXkgv

ASNNetworks, 2020-08-14:
> There we go!
> https://discord.gg/KgXkgv

thank you! Besides the error I'm getting with both lights: is it normal for Deconz to keep logging and having actitvies? It's as if it's non stop refreshing and looking for firmware updates and saying current channel is 25 etc.

Mimiix, 2020-08-14:
Yes. It is normal behaviour. I am not sure if it is due to HA having enabled some parameters on default, but nothing strange on that.

ASNNetworks, 2020-08-14:
> Yes. It is normal behaviour. I am not sure if it is due to HA having enabled parameters on default, but nothing strange on that.

That's good to know, then I;m not worried at least. Since I opened this issue because of that and the other issue you have pointed out can be discussed at Discord, I'll close this.

Thanks again!

closed 2020-08-14 by ASNNetworks