dresden-elektronik/deconz-rest-plugin #3145: Xiaomi Aqara Smart Switch - Long Press Button not available

Maddinthebrain, 2020-08-14:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

### deCONZ Logs

## Additional context

Smanar, 2020-08-14:
Wich one model id is ?
Not all model have same features.

Maddinthebrain, 2020-08-14:
> Wich one model id is ?
> Not all model have same features.

On the bottom there is written
型号: (this means model number) WXKG11LM
CMIIT ID: 2016DP6442
[image]
[image]

Do you need more information? 

Martin

Smanar, 2020-08-14:
I think 2016 is the year, and with the model 2016 you don't have long press.

ebaauw, 2020-08-14:
@Maddinthebrain I’m not sure what you’re saying, not what you’re asking.

> By the way the API recongizes the 1x long press yet. I can see and also use in FHEM.

Are you telling that the API does support long press for this switch?  In that case, there is no issue with the REST API plugin.

> But deconz only offers two possibilties to use it. 1x short press and 2x press

What do you mean by “deconz”?  The GUI?  The API?  The Phoscon web app?  See https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-for-Dummies for the differences between these.

Smanar, 2020-08-14:
You can too take a look on model id, if you have "lumi.sensor_switch.aq2" > 2016 > no long press
```
    { VENDOR_JENNIC, "lumi.sensor_switch.aq2", jennicMacPrefix }, // Xiaomi WXKG11LM 2016
    { VENDOR_JENNIC, "lumi.remote.b1acn01", jennicMacPrefix },    // Xiaomi WXKG11LM 2018
```

Maddinthebrain, 2020-08-15:
The Problem is, that in Phoscon you cannot choose long press for an action. But in Fhem you can use it as the status is recognized properly. The long press option is just missing in the Phoscon GUI and not anywhere else. It is where you define the actions of a light depending on switch actions. The "Schaltereditor" . There are only  the two possibilities 1x short and 2x short. 

Maybe this button has a long press  because in Phoscon a different and newer ID appears. 

[image]
[image]

Smanar, 2020-08-15:
For me, your device don't have long press, it s in the device firmware, no software.

ebaauw, 2020-08-15:
I would never have done this, but they hacked something to emulate long press for the `lumi.sensor_switch`.
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/f245b37194add12c23e1be3f7b96190351da1452/de_web_plugin.cpp#L3758-L3785

Maddinthebrain, 2020-08-15:
> For me, your device don't have long press, it s in the device firmware, not

Sorry, you are wrong, why Fhem then can use it? Fhem gets the data from 
deconz. And deconz from the switch. Fhem shows different status messages 
for 1x short, 2x short and as well as for 1x long. Please check it. It must 
be value 1003. There's indeed a 1x long! I use it in Fhem, so it's there.

Martin

ebaauw, 2020-08-15:
Anyways, the API already issues `buttonevent` 1003, so this is an issue for Phoscon.  @Mimiix can you transfer it?

Mimiix, 2020-08-15:
Your wish is my command @ebaauw