dresden-elektronik/deconz-rest-plugin #3146: Effect "colorloop" not working for groups when accompanied by an "on" action

klada, 2020-08-14:
## Describe the bug

When PUTing `{"effect": "none", "on": true}` to a groups action endpoint nothing happens, despite the fact that a HTTP 200 is returned:

```python
>>> requests.put("http://deconz-gw/api/KEY/groups/7/action", json={"effect": "none", "on": True})
<<< <Response [200]>
```
This is a perfectly valid request (as by the API docs). When omitting the `on=true` thing the colorloop is started.

The funny part is that the same request is working on the lights endpoint, where it properly starts the colorloop:

```python
>>> requests.put("("http://deconz-gw/api/KEY/lights/5/state", json={"effect": "colorloop", "on": True})
<<< <Response [200]>
```

## Steps to reproduce the behavior

1. Create a lights group
2. Start "colorloop" effect like mentioned above
3. "colorloop" does not start

## Expected behavior

Colorloop should be started on groups action endpoint, even if `on=true` is also present in the request body.

## Environment

 - Host system: PC
 - Running method: Installed through Debian Package
 - Firmware version: 26580700
 - deCONZ version: 2.05.79
 - Device: ConBee II
 - Do you use an USB extension cable: yes

## Additional context

I stumbled across this because Home Assistant always adds the `on=true` when starting effects. This means that until this issue is resolved the colorloop effect cannot be set through Home Assistant for light groups.

wvuyk, 2020-11-03:
Same issue here for my plugin. The combination with `"on"=true` results in not executing the` "effect"="colorloop"`.

xcorp, 2020-12-13:
Affects me as well, running deconz version 2.7.1 as an addon using hassio/supervisor, version there is 6.6.0

flemmingss, 2021-07-18:
Anyone update on this? I have the same problem.