dresden-elektronik/deconz-rest-plugin #3147: Calibrate Ubisys J1

jan666, 2020-08-15:
I just paired my Ubisys J1 (using it for a Awning) with deCONZ. Controling via deCONZ GUI (Window Covering Cluster) works, but not via HomeKit (HomeBridge). I wanted to calibrate it like described here: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/813 but I dont have a Sensor, just a Light. put config to Light does not work:

```
[
  {
    "error": {
      "address": "/lights/22",
      "description": "method, PUT, not available for resource, /lights/22",
      "type": 4
    }
  }
]
```

Did anything change here?

[image]

ebaauw, 2020-08-15:
I'm unfamiliar with the J1 and the calibration procedure.  _Window Covering_ devices need to be exposed as `/lights` resources for group support.  I don't know if the J1 was initially exposed as a `/sensors` resource and when this was changed.

Homebridge Hue should support _Window Covering_ devices; I'm using it for a Xiaomi curtain controller and an IKEA FYRTUR.  I don't think I've seen anyone trying the J1 yet.  Could you please open an issue in the Homebridge Hue repository, attaching the debug dump file that Homebridge Hue creates on startup and the Homebridge debug log file?

jan666, 2020-08-16:
@ebaauw thanks for your help. I can see it in the home app, I was able to open it once. But now its stuck in „öffnen“ mode forever. I guess it’s because its not yet calibrated so it does not know its state. I’ll try to provide some homebrige logs tomorrow but I thinks homebridge is not the problem here

SwoopX, 2020-08-16:
I got one and it exposes as least one sensor. However, you can find the manual calibration steps here, at the beginning as comments:

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/window_covering.cpp

jan666, 2020-08-17:
@SwoopX I tried like described here: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters with all 4 "Basic" Clusters (while search for new sensor in phoscon). Nothing happens. I even tried all the others cluster. What am I doing wrong here?

closed 2020-08-17 by jan666