dresden-elektronik/deconz-rest-plugin #3148: How to set a different vnc port

pippo73, 2020-08-15:
Hello, I've installed the deconz package on a Debian headless server but I cannot find the way to set the vnc port.

Could someone help me please

ebaauw, 2020-08-15:
That has nothing to do with deCONZ, but with how you configure the VNC server.
[image]

pippo73, 2020-08-15:
What server are you talking about?
In the deconz/phoscon system there is a part accessible by a vnc client.
I would like to know how to setup the port of this part

ebaauw, 2020-08-15:
The VNC client talks to a VNC server.  This is the standard screen sharing (or, in Windows speak, remote desktop) software that comes with Raspbian.  You can access the deCONZ GUI through the shared screen, but the screen sharing itself is not part of deCONZ.

You change the port that the VNC server uses (and to which the VNC client needs to connect) through the _Options_ in the VNC server, as the above screenshot shows.  Click the VNC Server icon on the top right of your screen to popup the VNC window.  Then select _Options_ from the menu icon on the right.  Never tried this myself, though, and not sure if you can change this remotely.

Mimiix, 2020-08-18:
@pippo73 You need a dedicated server to do this. Please follow at @ebaauw his comment.

I am closing this, as you haven't used a template to open this post and information is missing.

closed 2020-08-18 by Mimiix

pippo73, 2020-08-18:
I would be glad to use the template you are talking about if only I would find any... I've looked also in the wiki parte and there is any. Could you be so kind to link me to such template, in order to respect your request. Thank you

Mimiix, 2020-08-18:
@pippo73 It's fine, dont worry ;) No hard feelings.

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new/choose

On this page, you were asked to select a problem. Yours would be "user question".

However, Erik's comment should help you.  Or this would: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Access-deCONZ-GUI-in-headless-setups

If you still can't make it, open the issue i mentioned 😄