dresden-elektronik/deconz-rest-plugin #3150: tradfri 600lm e26 color bulb no color temp

thedannymullen, 2020-08-15:
Hello, I have two tradfri color bulbs.  In the latest release it appears the color temp does not appear in the phoscon application.  Not sure what happened to it.  Attached is a video it shows my color light that when the temperature is clicked it only has the brightness channel.  I have factory reset the bulb and re-added.  I am running 2.05.80 beta.  

https://youtu.be/tnvubJAR5bA

Please advise I am happy to try any suggestions.  I also tried to manually install an older version of deconz, but the issue still appeared.  Not sure that I uninstalled and re-installed properly.  I am happy to test again if needed.

ebaauw, 2020-08-15:
I don't know this particular light, but the Trådfri E27 CWS doesn't support colour temperature.  This light advertises that it also doesn't, only `xy` (`colorcapabilities` of 8).  The API rightly doesn't expose `ct`, and wrongly exposes `hue`, `sat`, and `effect`, which the light doesn't support either.

thedannymullen, 2020-08-15:
@ebaauw according to the box you are correct..  funny thing i am fairly confident it worked in the past.  Unfortunately I don’t have my time machine.   
I suppose the real question is how do I make it go back to just white?  That is my issue, when I change to a color it stays on color.   I can’t figure out how to get back to white.

thedannymullen, 2020-08-16:
[image]

Funny thing picture says white spectrum.  I am going to pair with my tradfri hub and see what I can do.  This is one of those weird issues...  I know I tested this when I installed them.  Again, no time machine.  
I will get back with you

ebaauw, 2020-08-16:
> I suppose the real question is how do I make it go back to just white?

Set the `xy` values to white, probably something like `[0.35, 0.35]`.  Alternatively, power cycle the light and see what `xy` values it supports.  In Phoscon, you want to set the _Saturation_ to 0.

[image]

> i am fairly confident it worked in the past

You might want to double-check in the GUI, if the light responds to _Move to Color Temperature_, and if it supports the _Color Temperature_ attribute.  This wouldn’t be the first light that advertises the wrong capabilities, but IKEA are usually correct here.

You might also want to double-check using the API, if you can still do a PUT to the light `state`, setting `ct`.  There is some obscure (at least, to me) code in the API plugin to translate `ct` values to `xy`, and the API still accepts setting `ct`, even if the light resource doesn’t include it.  There could have been a change in Phoscon, showing the colour temperature slider only when `state.ct` is exposed.

thedannymullen, 2020-08-16:
@ebaauw  I am Closing issue for now.  More research will be done and I will open the issue again if necessary.  
Thanks for help!

closed 2020-08-16 by thedannymullen

thedannymullen, 2020-08-19:
@ebaauw question for you on this topic.  Is there a way for me to make the rest api think this bulb is an extended color light for testing?  I can't seem to figure out where in the source code the model is linked to type?

I am looking to test if this works.  It looks like according to the product id on my box and the product id on the website the bulb supports white spectrum.  Which i believe would make it a type extended color light vs just a color light.  

I am happy to compile the code and make my own changes I was looking for direction on where they type gets pulled from.

ebaauw, 2020-08-19:
> I can't seem to figure out where in the source code the model is linked to type?

Because it isn't.  The type is reported by the light (it's the _Device Type_ in the _Simple Descriptor_, which the GUI shows at the endpoint).

> It looks like according to the product id on my box and the product id on the website the bulb supports white spectrum.

You can test in the GUI if the light supports the _Move to Color Temperature_ command and the _Color Temperature_ attribute.  In theory, the light could expose the wrong device type and _Color Capabilities_, but I would expect better from IKEA (maybe I'm too optimistic).  If the light does support these, we could whitelist it and treat it as an _Extended Color Light_ in the REST API plugin.

thedannymullen, 2020-08-19:
Thanks for your info.  

Yes, the light supports color temp in the ikea app using their gateway.  As a second test, I use the ikea binding in openhab2 it is discovered as extended color and works perfectly.  Note, I am not sure how openhab handles the model identification.  

I will reconnect to deconz and look at the attributes later today and see what they say.  It would be nice to whitelist the bulb if in fact it is reporting incorrectly.

thedannymullen, 2020-08-19:
I did find something interesting today.  When I joined it to the deconz network, the bulb joined and said extended color light.  I had to reset it again because I forgot to delete it out when I reset it.  Some of the info like firmware of the bulb didnt match what was in it.  When I rejoined deconz it joined as a color bulb.

I am not sure what ikea is doing with the bulb description, but it is definitely an extended color light.  Is it possible to whitelist it?

ebaauw, 2020-08-19:
I think the bulb’s firmware might have been updated while it was connected to the Trådfri hub.  If it now exposes itself as _Extended Color Light_, no action would be needed.  Can you double-check what color capabilities and version the light now reports?

thedannymullen, 2020-08-20:
I have tried repeating the earlier process of pairing to tradfri then deconz but the bulb always shows as just a color light.   I believe deconz is seeing the bulb correct based on ID, but the ID is actually wrong for this bulb.   I say this because the tradfri app support the white spectrum as discussed on the ikea website.  

Can we try a whitelist, I can compile the code and test, but don't know how/where to make the changes.

ebaauw, 2020-08-20:
Overwriting the device type is done here: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/f245b37194add12c23e1be3f7b96190351da1452/de_web_plugin.cpp#L2270

You might want to double-check under what conditions this routine is called, might be only for OSRAM/LEDVANCE lights.

thedannymullen, 2020-08-24:
[image]

Good Morning/Evening @ebaauw, 

I have been trying some different configurations and learning how the code is written.  I have posted the cluster response from the bulb.  One thing I was uncertain about, I am wondering if you know.  Does the type reported "color temperature" mean that it should support adjusting the kelvin temperature of the white?  

I am a little confused on the types reported color / color temperature/ extended color etc.  

As of now, you stand correct that this appears to not support the color temperature directly.  In some research, I found there is another 600lm bulb with a dfferent product code.  My product code internal is 1642xxx vs a 17xx or 18xx according to the supported devices on the zigbee2mqtt page this bulb supports the color xy as you stated prior.  But the bulbs of the different product code support color temperature.  I am thinking Ikea may have upgraded their bulbs or sell different bulbs to different countries, but sell them under the same UPC code.   This is why my box reports 2700kelvin but the newer boxes show white spectrum.   

So for now I will stop spending time making this one work, I will make a trip to the ikea store and review the latest boxes and possibly purchase one.  Maybe I will get one of the new product codes.  These bulbs are very usable still and I am unable to explain what software update made them stop functioning the way they did for the last year.  

Thanks for all your assistance!!!  I learned many things about zigbee and deconz in the last few weeks!

ebaauw, 2020-08-24:
_Color Mode_ indicates how the colour was last set: using _Current X_ and _Current Y_, using _Color Temperature_, or using _Current Hue_ and _Current Saturation_.  The current state is reflected by the corresponding attributes.  I don't think the standard defines what the other attributes should be.  Hue lights update all attributes; some other lights only update the attributes corresponding to the _Color Mode_.

Did you get the second screenshot after issuing a _Move to Color Temperature_ from the GUI, or after putting `ct` through the API?  Are you sure you force-read the attributes before making the screenshot (so it's actually the values reported by the light, and not the values set by the REST API plugin)?  It might be prudent to disable the REST API plugin while sending commands from the GUI, to make sure you interact with the light only.
If so, this light supports _Move to Color Temperature_ (given that _Color Mode_ reports _Color Temperature_), but it doesn't support the _Color Temperature_ attribute.  It would seem that the light did update _Current X_ and _Current Y_; I hope with the values corresponding to the set _Color Temperature_.  I haven't seen that behaviour in any other light.  

If the bulb has three channels (RGB), this whole discussion is academic, because setting `ct` results in the same RGB mix as setting the corresponding `xy` values.  If the bulb has five channels (RGB+CCT) however, setting `ct` drives the WW/CW channels whereas setting the corresponding `xy` values drives the RGB values, resulting in a noticeable different colour.

thedannymullen, 2020-08-24:
DO you happen to know if there is a way to obtain the number of channels other than opening the bulb?

ebaauw, 2020-08-24:
Not that I know of.