dresden-elektronik/deconz-rest-plugin #3151: Aqara Vibration Sensor Problemes showing as a light two

clipse2004, 2020-08-15:
## Describe the bug
Hello,
I currently have the latest beta from Deconz running (but my problem was before)
I have two Aqara vibration sensors or two devices:

0x00158d000482dcab and 0x00158d00039e0b30

With these two sensors I have the problem that they are recognized in Deconz, but are then displayed in HomeKit as "Door Lock 16 and 17 (light switch) as well as vibration sensor.
Until yesterday everything worked fine.

Can you help me how to get rid of this? I have already tried to delete these devices in the database and import them again, but after a certain time they are automatically recognized by Deconz.

Oh yes, when I compare the serial numbers of the light and the vibration sensor in HomeKit, they are always the same.

I hope I was able to explain this clearly :)

Merci

  If the problem can't be reproduced and is sporadic, please provide some details
  on how often and when the issue happens.
--> Always when i tried to connect this sensors new

## Screenshots

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Raspbian
 - Firmware version: 26580700
 - deCONZ version: 2.05.80
 - Device:  ConBee II
 - Do you use an USB extension cable: yes

i attach two photos from the devices.

Thank You
[image]
[image]

ebaauw, 2020-08-15:
Can you please list the resources that the API made for these switches?  It's also OK to attach the debug dump file Homebridge Hue creates on startup.

The vibration sensor exposes itself as a _Door Lock_, and uses some manufacturer-specific attributes in the _Door Lock_ cluster.  They did add support for the DanaLock in https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3030.  I fear that might have caused the creation of the _Door Lock_ `/lights` resources.

clipse2004, 2020-08-16:
ok I will post it tomorrow

smulle48, 2020-08-17:
I don't know if it's specific to Homebridge Hue.

I'm using deCONZ with Jeedom and their Homebridge plugin. I have defined the Vibration as a motion sensor, and it's been this way the last year, even with multiple restarts and reinstall.

ebaauw, 2020-08-17:
This has nothing to do with Homebridge Hue.  With the support for the Danalock in v2.05.80, the REST API plugin now creates a `/lights` resource for the _Door Lock_ cluster that the vibration sensor inexplicably uses.  I suppose this resource is created only when you reset and re-pair the sensor.

smulle48, 2020-08-17:
> This has nothing to do with Homebridge Hue.  With the support for the Danalock in v2.05.80, the REST API plugin now creates a `/lights` resource for the _Door Lock_ cluster that the vibration sensor inexplicably uses.  I suppose this resource is created only when you reset and re-pair the sensor.

I'll try to reset and repair my sensor to see if I get the same result.

I don't know how Homebridge Hue handle the characteristics, but in Jeedom, I define it myself from a list, forcing it to use that specific characteristic

Smanar, 2020-08-17:
Can make a filter using manufacture number ?

smulle48, 2020-08-17:
I have just reset my sensor and in both my Danalock and vibration sensor, the door lock is there, but my vibration sensor does not change to a light, in the home app. Don't know if it's because Jeedom force the vibration to mimic a motion sensor? I have also tried to let it parse all information directly, but still no light in home app

smulle48, 2020-08-17:
> Can make a filter using manufacture number ? VENDOR_DANALOCK is rare.

If the Xiaomi sensor is piggybacking on the door lock cluster, then the Danalock is not the culprit :-D

ebaauw, 2020-08-17:
> but my vibration sensor does not change to a light

It wouldn't change; the `ZHAVibration` `/sensors` resource is still there.  You'd just see an additional `/lights` resource.  I wouldn't know how Jeedom would react to that.  You would need to restart Homebridge, before Homebridge Hue would expose the additional resource.

> Can make a filter using manufacture number ? VENDOR_DANALOCK is rare.

Filtering on _Manufacturer Code_ is probably the safest; the _Basic_ cluster attributes won't yet have been read when the `/lights` resource is created.  Problem is that Xiaomi uses `VENDOR_JENNIC` (0x1037), so we're likely to find a legitimate door lock using the same code sooner rather than later.  I think I've already seen a request for support of the Xiaomi door lock?

Smanar, 2020-08-17:
Filtering on Manufacturer Code is probably the safest; the Basic cluster attributes won't yet have been read when the /lights resource is created. Problem is that Xiaomi uses VENDOR_JENNIC (0x1037), so we're likely to find a legitimate door lock using the same code sooner rather than later. I think I've already seen a request for support of the Xiaomi door lock?

It s for that I think it s better to use the VENDOR_DANALOCK, more rare and more stable

with Removing

            // Danalock support. The cluster needs to be defined and whitelisted by setting hasServerOnOff
            else if (i->inClusters()[c].id() == DOOR_LOCK_CLUSTER_ID) { hasServerOnOff = true; }

and making
case DEV_ID_DOOR_LOCK:
{
if (node->nodeDescriptor().manufacturerCode() == VENDOR_DANALOCK)
{
lightNode.setHaEndpoint(*i);
}
}

Like this no more problem for Xiaomi, they can use the vendor number they want.

smulle48, 2020-08-17:
> if (node->nodeDescriptor().manufacturerCode() == VENDOR_DANALOCK)

Just tried the code, but that killed my Ikea tradfri outlet :-/

Reverting to the original code, made the outlet appear again :-)

Smanar, 2020-08-18:
Ha ?

Lol, sorry, I realy don't see the relation. This modification work only for new device inclusion, and I don't see anything about DEV_ID_DOOR_LOCK in the ikea outlet.

wvuyk, 2020-08-19:
Can confirm the light resource is created with 2.05.80. My plugin starts creating a plug now fir the vibration sensors. Name is "Lock nn".

clipse2004, 2020-08-19:
> Can confirm the light resource is created with 2.05.80. My plugin starts creating a plug now fir the vibration sensors. Name is "Lock nn".

Yes but you can delete the lights in the Phoscon App.
 That worked for me

closed 2020-09-11 by clipse2004