dresden-elektronik/deconz-rest-plugin #3152: Kwikset 914 ZigBee smart lock

laurentbeaumer, 2020-08-16:
- Product name: Kwikset 914 ZigBee smart lock
- Manufacturer: Kwikset 
- Model identifier: 914
- Device type : smart lock

I am running headless Deconz, so I can't take screenshots.

Is there a way to get information about the cluster from command line or RestAPI?

Thanks

Smanar, 2020-08-16:
Hello, you miss the cluster list.
But the device is not correctly included yet, you miss the second circle to display the cluster list.

On the first circle, you can trigger some descriptor request, to "enable" the second circle. You propably need to wake up the device in same time.

laurentbeaumer, 2020-08-17:
I have switched to zigbee2mqtt , and it is working. Thanks

zrhutto, 2020-10-09:
Would it be better to reopen this issue @Smanar or create a new one to request device support? Assuming it’s preferable to avoid an extra issue, the cluster list is attached.
[image]

Smanar, 2020-10-09:
Nice thx ^^.
Do you have an unix machine to test code modification ?

zrhutto, 2020-10-09:
I do indeed - running Linux armhf and happy to test

Smanar, 2020-10-10:
Ok so 

To test the new code, on an Unix machine (Hassio is not an "Unix OS")

You have the procedure https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

Just replace the step 1 by
`git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git`
(yeah I m using tuya branch, will change that later if needed) and you can skip the step
`git checkout -b mybranch HEAD`

The last step replace your actual file, so better to close deconz before, or deconz will be closed itself by the change. (you can make a backup if you want but in another folder, file is libde_rest_plugin.so)

After that you just need to re-include the device.

I have some problem ATM with my setting, so the code is not tested, if you have error at compilation can stop direclty.

With this code (if I m right)
- the device will be added in phoscon
- the device will be switched in light section
- not sure the on/off will work yet.

zrhutto, 2020-10-23:
Apologies for the delayed response. I pulled and built the branch you recommended - after removing the door lock and re-adding the device, it still only shows in the deCONZ gui with nothing in the Phoscon web GUI.

Any debugging or other steps you'd suggest trying?

Smanar, 2020-10-24:
Yep, if you can run decon in debug mode, and the log pls ?

`deCONZ --dbg-error=2 --dbg-info=2 --dbg-zdp=0 --dbg-zcl=0 --db-aps=0 --dbg-http=0 --dbg-ota=0`

I just need the part just before you start the inclusion, and 2 mn after.

zrhutto, 2020-10-24:
You'll find the log [here](https://pastebin.com/wPamre6b).

Interestingly, I see some detail in the logs about the ZHATemperature endpoint, and the deCONZ GUI labeling the lock as 'Temperature 11' - despite the fact the device is still recognized as a Door Lock.

Let me know if additional debugging/logging would be useful.

[image]

zrhutto, 2020-10-24:
Interestingly the lock appears in the REST API as a sensor, but not in the Phoscon web app

```11":{"config":{"battery":80,"offset":0,"on":true,"reachable":true},"ep":2,"etag":"0c11e4df1686fce939868a77e9ce5872","lastseen":"2020-10-24T17:03Z","manufacturername":"Kwikset","modelid":"SMARTCODE_CONVERT_GEN1","name":"Temperature 11","state":{"lastupdated":"2020-10-24T17:00:13.986","temperature":2500},"type":"ZHATemperature","uniqueid":"00:24:46:ff:fd:03:3b:06-02-0402"}

Smanar, 2020-10-24:
Yep, can be normal phosocn don't display all device.

Now I know why it s not working ...., look at the first capture, you haven't the same device.
You have 0x1092 as manufacture number, and him 0x0000

I have corrected the code, but I m making another modification on same time (always same branch : tuya) so if you have error during compilation, you can stop directly.

zrhutto, 2020-10-25:
Trying the latest commit - still not seeing a light created for the door lock cluster (0x0101) after unpairing and re-adding the device.

Latest [logs are here](https://pastebin.com/2jFmVZ9v)

Smanar, 2020-10-25:
Sorry , my bad > https://github.com/Smanar/deconz-rest-plugin/commit/18bf1beaeaebc302e3a0ab276878b0b65bbbe467

New correction, I think it will be fine now.

zrhutto, 2020-10-25:
That did it! Thanks for all the help @Smanar

Smanar, 2020-10-25:
Fucking bot.

I will remove the zhathemperature sensor, but can you test the rest ?

Smanar, 2020-10-27:
Have added the code to prevent temperature device be created, but still waiting if all is ok before making the PR.

Mdleal, 2020-11-01:
I'll try to get this tested this week. Currently running deconz in docker so have to migrate it to a Linux install.

Mdleal, 2020-11-02:
[image]

Smanar, 2020-11-03:
You have only the temperature sensor ?
You are using this branch ?
>git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git

Mdleal, 2020-11-03:
Yep. Built the package off the tuya branch.

[image]

Smanar, 2020-11-03:
Nothing in light section ?
Can you show the device JSON ?

Mdleal, 2020-11-03:
Nothing under lights. - Maybe because I didn't add it as a light?
Device .json  - Where is this located?

Smanar, 2020-11-03:
No, it s same under light or under sensor.

On HA, itdk if it s possible to have it, just go in phoscon, read the ip and the port and use
`http://IP:PORT/api/API_KEY/sensors`

For the api key, HA use one, you can use the same.

It s probably visible in the log too.

Because I can be wrong for the modification to supress the temperature device, but the doorlock device have worked with zrhutto

Mdleal, 2020-11-03:
`{"1":{"config":{"configured":true,"on":true,"sunriseoffset":30,"sunsetoffset":-30},"etag":"cad2f2f1e59f6ed76744c69e1a268a65","manufacturername":"Philips","modelid":"PHDL00","name":"Daylight","state":{"dark":false,"daylight":true,"lastupdated":"2020-11-03T13:07:15.066","status":160,"sunrise":"2020-11-03T12:33:41","sunset":"2020-11-03T23:29:00"},"swversion":"1.0","type":"Daylight","uniqueid":"00:21:2e:ff:ff:04:2a:e6-01"},"10":{"config":{"on":true,"reachable":true},"ep":1,"etag":"a88624c1ba037665586e5baf902ce70c","lastseen":"2020-11-03T15:48Z","manufacturername":"Samjin","modelid":"outlet","name":"outlet","state":{"current":19,"lastupdated":"2020-11-03T15:48:05.065","power":0,"voltage":124},"type":"ZHAPower","uniqueid":"28:6d:97:00:01:06:a5:9c-01-0b04"},"11":{"config":{"on":true,"reachable":true},"ep":1,"etag":"0e89cea13586aa32890e3702cc569e1b","lastseen":"2020-11-03T15:48Z","manufacturername":"Samjin","modelid":"outlet","name":"Adapter plug","state":{"current":12,"lastupdated":"2020-11-03T15:48:06.761","power":0,"voltage":125},"type":"ZHAPower","uniqueid":"28:6d:97:00:01:04:a6:3e-01-0b04"},"12":{"config":{"battery":null,"offset":0,"on":true,"reachable":true},"ep":2,"etag":"ee795c0c2f574acf33ca099837c94621","lastseen":"2020-11-03T15:48Z","manufacturername":"Kwikset","modelid":"SMARTCODE_CONVERT_GEN1","name":"SMARTCODE_CONVERT_GEN1","state":{"lastupdated":"none","temperature":null},"type":"ZHATemperature","uniqueid":"00:24:46:ff:fd:03:23:d3-02-0402"},"3":{"config":{"battery":100,"duration":90,"on":true,"reachable":true,"temperature":2600},"ep":1,"etag":"44a89f6320723e2c3fe6a2bbeb06afb3","lastseen":"2020-11-03T14:31Z","manufacturername":"LUMI","modelid":"lumi.sensor_motion.aq2","name":"Kitchen Motion Sensor 2","state":{"lastupdated":"2020-11-03T14:32:58.866","presence":false},"swversion":"20170627","type":"ZHAPresence","uniqueid":"00:15:8d:00:01:e5:50:a6-01-0406"},"4":{"config":{"battery":100,"on":true,"reachable":true,"temperature":2600,"tholddark":12000,"tholdoffset":7000},"ep":1,"etag":"44a89f6320723e2c3fe6a2bbeb06afb3","lastseen":"2020-11-03T14:31Z","manufacturername":"LUMI","modelid":"lumi.sensor_motion.aq2","name":"Kitchen Motion Sensor 2","state":{"dark":false,"daylight":true,"lastupdated":"2020-11-03T14:31:28.675","lightlevel":21004,"lux":126},"swversion":"20170627","type":"ZHALightLevel","uniqueid":"00:15:8d:00:01:e5:50:a6-01-0400"},"5":{"config":{"battery":100,"on":true,"pending":[],"reachable":true,"temperature":2300},"ep":1,"etag":"812c9ca26a1f5216ba3941f1ea73b070","lastseen":"2020-11-03T15:42Z","manufacturername":"LUMI","modelid":"lumi.sensor_wleak.aq1","name":"DS Hot Water Heater","state":{"lastupdated":"2020-11-03T15:42:21.242","lowbattery":false,"tampered":false,"water":false},"swversion":"20170721","type":"ZHAWater","uniqueid":"00:15:8d:00:02:83:4e:a0-01-0500"},"6":{"config":{"battery":100,"on":true,"pending":[],"reachable":true,"temperature":2300},"ep":1,"etag":"a3aa07efbc5229151dc53a3672528c66","lastseen":"2020-11-03T15:27Z","manufacturername":"LUMI","modelid":"lumi.sensor_wleak.aq1","name":"Kitchen sink","state":{"lastupdated":"2020-11-03T15:27:43.105","lowbattery":false,"tampered":false,"water":false},"swversion":"20170721","type":"ZHAWater","uniqueid":"00:15:8d:00:02:87:9b:1b-01-0500"},"7":{"config":{"battery":88,"on":true,"reachable":true,"temperature":2900,"tholddark":12000,"tholdoffset":7000},"ep":1,"etag":"0ddb77ada406c03bc54ae0e78a929221","lastseen":"2020-11-03T15:24Z","manufacturername":"LUMI","modelid":"lumi.sensor_motion.aq2","name":"Kitchen Motion Sensor","state":{"dark":false,"daylight":true,"lastupdated":"2020-11-03T14:31:29.533","lightlevel":24378,"lux":274},"swversion":"20170627","type":"ZHALightLevel","uniqueid":"00:15:8d:00:02:25:16:ec-01-0400"},"8":{"config":{"battery":88,"duration":90,"on":true,"reachable":true,"temperature":2900},"ep":1,"etag":"0ddb77ada406c03bc54ae0e78a929221","lastseen":"2020-11-03T15:24Z","manufacturername":"LUMI","modelid":"lumi.sensor_motion.aq2","name":"Kitchen Motion Sensor","state":{"lastupdated":"2020-11-03T14:32:59.666","presence":false},"swversion":"20170627","type":"ZHAPresence","uniqueid":"00:15:8d:00:02:25:16:ec-01-0406"},"9":{"config":{"alert":"none","battery":47,"on":true,"reachable":true},"ep":1,"etag":"c71f9c41527fc54069917b4778790f74","lastseen":"2020-11-03T06:42Z","manufacturername":"IKEA of Sweden","mode":1,"modelid":"SYMFONISK Sound Controller","name":"SYMFONISK Sound Controller ","state":{"buttonevent":2003,"lastupdated":"2020-10-31T22:20:05.237"},"swversion":"2.1.022","type":"Z`

Smanar, 2020-11-03:
I have make a new test.
You can test the new code, I m checking the manufacture directly (as this one make only the doorlock)

Mdleal, 2020-11-03:
No problem. Let me know when you want to test again. I've heard zigbee2mqtt has the lock working so there might be some hints there. I also have ZHA setup in home assistant using another zigbee controller. I'll see if I can pair it with that.

Smanar, 2020-11-03:
Now, you can , the code is already online

Mdleal, 2020-11-03:
Code pulled and built. I have to wait till I get off work to re-pair the lock. 

Thanks.

Mdleal, 2020-11-03:
Same after resetting the lock and re-pairing. Nothing under lights and device show up as a temp sensor in HA.

Smanar, 2020-11-04:
I m checking the code again, and I don't see how it s possible the code create the sensor.
You are using only 1 deconz instance, you are making the compilation on the same machine you are using deconz

You are doing
```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

You are on the good branch ?
```
cd deconz-rest-plugin
git branch
```
Need to have tuya.

Mdleal, 2020-11-04:
I'll run a build again. 

`root@ubuntu-desktop:/home/mdleal/deconz-rest-plugin# git pull
remote: Enumerating objects: 53, done.
remote: Counting objects: 100% (53/53), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 94 (delta 51), reused 51 (delta 51), pack-reused 41
Unpacking objects: 100% (94/94), 50.40 KiB | 81.00 KiB/s, done.
From https://github.com/Smanar/deconz-rest-plugin
   bbd84e2..487d835  bitron     -> origin/bitron
Already up to date.
root@ubuntu-desktop:/home/mdleal/deconz-rest-plugin# git branch
* tuya
root@ubuntu-desktop:/home/mdleal/deconz-rest-plugin# `

Smanar, 2020-11-04:
Yep it s possible, if you want to do that, I will add some other debug line to trace the code, but I have

```
                    else if (node->nodeDescriptor().manufacturerCode() == VENDOR_KWIKSET)
                    {
                    }
                    else
                    {
                        fpTemperatureSensor.inClusters.push_back(ci->id());
                    }
```

So if the manufacture code = 1092, the ZHATemperature sensor can't be created. There is only 3 lines of code, hard to make mistake.

Can you make a try with deleting the folder and remake 
`git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git`

Mdleal, 2020-11-04:
Success!!!! I deleted the node in deconz and repaired. I now have a lock in HA. Anyway to add other sensors? Battery, alaram,...

Smanar, 2020-11-05:
Ha right I forget battery
So I can make a zhaBattery device, but I can too make a ZHAalamr device with alarm and battery.
But what the alarm part ? Waht this device can return as alarm ?

Mdleal, 2020-11-05:
Alarm shows how the lock was unlocked. manually vs by code. 

[image]

Here is the same lock but Z-Wave.
[image]

Mdleal, 2020-11-05:
Weird. I'm guessing after I read the cluster info the battery popped up in HA along with temp..

[image]

Smanar, 2020-11-06:
Lol, he is coming back ?
Can you find the device JSON of thoses devices ? Or just the uniqueID, I think this is visible in HA, just to know if you have a ZHAbattery with temperature, a ZHAtemperature with battery, or both devices.

IDK what you have in the alarm cluster, but from the information I have the information was from the 0101 cluster.
So I can create a ZHAOpenClose entry, but it mean it will create a door sensor in your third app, I can mimic the on/off state to the locked/unlocked, and add an information field in "state" to return more information ?

Mdleal, 2020-11-08:
devices.json?

`{"1":{"config":{"configured":true,"on":true,"sunriseoffset":30,"sunsetoffset":-30},"etag":"dfe7d9e09c4320b640ec6a1209a6af7d","manufacturername":"Philips","modelid":"PHDL00","name":"Daylight","state":{"dark":true,"daylight":false,"lastupdated":"2020-11-07T23:51:10.665","status":210,"sunrise":"2020-11-07T12:36:54","sunset":"2020-11-07T23:26:10"},"swversion":"1.0","type":"Daylight","uniqueid":"00:21:2e:ff:ff:04:2a:e6-01"},"10":{"config":{"on":true,"reachable":true},"ep":1,"etag":"c3a18cc5322972b147f1ce922d2f6fd7","lastseen":"2020-11-08T00:05Z","manufacturername":"Samjin","modelid":"outlet","name":"outlet","state":{"current":19,"lastupdated":"2020-11-08T00:05:11.983","power":0,"voltage":120},"type":"ZHAPower","uniqueid":"28:6d:97:00:01:06:a5:9c-01-0b04"},"11":{"config":{"on":true,"reachable":true},"ep":1,"etag":"57d4f7aba63219d5ef7327f6c708d28f","lastseen":"2020-11-08T00:05Z","manufacturername":"Samjin","modelid":"outlet","name":"Adapter plug","state":{"current":12,"lastupdated":"2020-11-08T00:05:15.729","power":0,"voltage":124},"type":"ZHAPower","uniqueid":"28:6d:97:00:01:04:a6:3e-01-0b04"},"12":{"config":{"battery":100,"offset":0,"on":true,"reachable":true},"ep":2,"etag":"9e680826e2b0ebb6fa7b19a272a70f2d","lastseen":"2020-11-08T00:05Z","manufacturername":"Kwikset","modelid":"SMARTCODE_CONVERT_GEN1","name":"SMARTCODE_CONVERT_GEN1","state":{"lastupdated":"2020-11-05T17:35:23.784","temperature":2500},"type":"ZHATemperature","uniqueid":"00:24:46:ff:fd:03:23:d3-02-0402"},"3":{"config":{"battery":100,"duration":90,"on":true,"reachable":true,"temperature":2700},"ep":1,"etag":"c2e606f89c55d4c1159f4edc1373ca4c","lastseen":"2020-11-07T23:50Z","manufacturername":"LUMI","modelid":"lumi.sensor_motion.aq2","name":"Kitchen Motion Sensor 2","state":{"lastupdated":"2020-11-07T23:51:52.364","presence":false},"swversion":"20170627","type":"ZHAPresence","uniqueid":"00:15:8d:00:01:e5:50:a6-01-0406"},"4":{"config":{"battery":100,"on":true,"reachable":true,"temperature":2700,"tholddark":12000,"tholdoffset":7000},"ep":1,"etag":"c2e606f89c55d4c1159f4edc1373ca4c","lastseen":"2020-11-07T23:50Z","manufacturername":"LUMI","modelid":"lumi.sensor_motion.aq2","name":"Kitchen Motion Sensor 2","state":{"dark":true,"daylight":false,"lastupdated":"2020-11-07T23:50:22.247","lightlevel":10001,"lux":10},"swversion":"20170627","type":"ZHALightLevel","uniqueid":"00:15:8d:00:01:e5:50:a6-01-0400"},"5":{"config":{"battery":100,"on":true,"pending":[],"reachable":true,"temperature":3300},"ep":1,"etag":"70074b167093fc3d0d7162c5e2ff0773","lastseen":"2020-11-07T23:20Z","manufacturername":"LUMI","modelid":"lumi.sensor_wleak.aq1","name":"DS Hot Water Heater","state":{"lastupdated":"2020-11-07T23:20:30.665","lowbattery":false,"tampered":false,"water":false},"swversion":"20170721","type":"ZHAWater","uniqueid":"00:15:8d:00:02:83:4e:a0-01-0500"},"6":{"config":{"battery":100,"on":true,"pending":[],"reachable":true,"temperature":2700},"ep":1,"etag":"06a22db1e7f01b9706a574fa7197d08b","lastseen":"2020-11-08T00:01Z","manufacturername":"LUMI","modelid":"lumi.sensor_wleak.aq1","name":"Kitchen sink","state":{"lastupdated":"2020-11-08T00:01:10.213","lowbattery":false,"tampered":false,"water":false},"swversion":"20170721","type":"ZHAWater","uniqueid":"00:15:8d:00:02:87:9b:1b-01-0500"},"7":{"config":{"battery":91,"on":true,"reachable":true,"temperature":3300,"tholddark":12000,"tholdoffset":7000},"ep":1,"etag":"54fee18783a20328fbc8c66e6a3aa0d8","lastseen":"2020-11-07T23:50Z","manufacturername":"LUMI","modelid":"lumi.sensor_motion.aq2","name":"Kitchen Motion Sensor","state":{"dark":true,"daylight":false,"lastupdated":"2020-11-07T23:50:23.014","lightlevel":0,"lux":0},"swversion":"20170627","type":"ZHALightLevel","uniqueid":"00:15:8d:00:02:25:16:ec-01-0400"},"8":{"config":{"battery":91,"duration":90,"on":true,"reachable":true,"temperature":3300},"ep":1,"etag":"54fee18783a20328fbc8c66e6a3aa0d8","lastseen":"2020-11-07T23:50Z","manufacturername":"LUMI","modelid":"lumi.sensor_motion.aq2","name":"Kitchen Motion Sensor","state":{"lastupdated":"2020-11-07T23:51:53.165","presence":false},"swversion":"20170627","type":"ZHAPresence","uniqueid":"00:15:8d:00:02:25:16:ec-01-0406"},"9":{"config":{"alert":"none","battery":47,"on":true,"reachable":true},"ep":1,"etag":"dfe7d9e09c4320b640ec6a1209a6af7d","lastseen":"2020-11-07T07:52Z","manufacturername":"IKEA of Sweden","mode":1,"modelid":"SYMFONISK Sound Controller","name":"SYMFONISK Sound Controller ","state":{"buttonevent":2003,"lastupdated":"2020-10-31T22:20:05.237"},"swversion":"2.1.022root@a0d7b954-vscode:/config# `

Smanar, 2020-11-08:
Ok so it s a ZHAtemperature with battery

```
"12":{
   "config":{
      "battery":100,
      "offset":0,
      "on":true,
      "reachable":true
   },
   "ep":2,
   "etag":"9e680826e2b0ebb6fa7b19a272a70f2d",
   "lastseen":"2020-11-08T00:05Z",
   "manufacturername":"Kwikset",
   "modelid":"SMARTCODE_CONVERT_GEN1",
   "name":"SMARTCODE_CONVERT_GEN1",
   "state":{
      "lastupdated":"2020-11-05T17:35:23.784",
      "temperature":2500
   },
   "type":"ZHATemperature",
   "uniqueid":"00:24:46:ff:fd:03:23:d3-02-0402"
}
```

I realy don't understand, where it s from ....
I will add more check on the condition, but BTW you haven't answered, what do you want as sensor for the "state return" ? a ZHAOpenClose  ? you will have true/false for lock state, and a special field in "state" ?

Mdleal, 2020-11-20:
Alarm cluster should return a number. Check out section 7.3.2.1.1 
[](https://raw.githubusercontent.com/Koenkk/zigbee-herdsman/master/docs/Zigbee%20Cluster%20Library%20Specification%20v7.pdf)

[image]

Smanar, 2020-11-20:
Yep but I have a problem the actual entry is a "light" and I can't add special field info on it.
So I need to create a "sensor" entry, if you are agree, I can create a zhaopenclose (with the "light" one), the same than for door sensor, and a field for exemple "lockstate" ?

Mdleal, 2020-11-23:
sounds like a plan.

Smanar, 2020-11-23:
Ok so have started something

```
git clone --branch kiwiset https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

It s realy somethng preliminary
- It will remove the zhatemperature
- it will add a shaopenclose
- the zhaopenclose will mimic the on/off from light entry

The sensor don't support a reboot/reload for the moment, was just for test.

Mdleal, 2020-11-23:
`e_web_plugin.cpp: In member function ‘LightNode* DeRestPluginPrivate::updateLightNode(const deCONZ::NodeEvent&)’:
de_web_plugin.cpp:3114:13: error: expected ‘}’ before ‘else’
 3114 |             else if (ic->id() == BASIC_CLUSTER_ID && (event.clusterId() == BASIC_CLUSTER_ID))
      |             ^~~~
de_web_plugin.cpp:3067:13: note: to match this ‘{’
 3067 |             {
      |             ^
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS -DHAS_SQLITE3 -DGW_SW_VERSION=\"2.06.00\" -DGW_SW_DATE=1606165427 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"88be7a931a86169ff75435b2c4e5e84e188cf311\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtSerialPort -isystem /usr/include/x86_64-linux-gnu/qt5/QtWebSockets -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -Irelease -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o release/de_otau.o de_otau.cpp
make[1]: *** [Makefile.Release:734: release/de_web_plugin.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/mdleal/deconz-rest-plugin'
make: *** [Makefile:42: release] Error 2`

Smanar, 2020-11-24:
My bad, I m not testing my changes, code updated

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Mdleal, 2020-11-24:
No problem. I have updated. I am guessing I'm going to have to re-add the lock in order to see the changes?

Smanar, 2020-11-24:
No I don't think for test.
You can just set phoscon in permitjoin (add new sensor), adn read again basic attribute (cluster 0x0000, button "read") the table is not filled you need to make the device awake in same time.

It will trigger a new inclusion easily, but idk what will happen on sensor already present, you probaby need to delete them (in phoscon, not deconz, for this trick the device need to be included in deconz)

Mdleal, 2020-11-24:
[image]
[image]

Smanar, 2020-11-24:
Ok, not so bad ^^
- the zha temperature is still here, perhpas just because you haven't delete it ?
- the second one is the ZHAopenclose I think
- the battery is probably from the temperature or openclose entry

Can you check the ZHAopenclose for the field "lockstate" ?

Mdleal, 2020-11-24:
,"state":{"lastupdated":"none","lockstate":null,"open":false},"type":"ZHAOpenClose","uniqueid":"00:24:46:ff:fd:03:23:d3-02-0101"}

I do not see a delete under the entity in HA so I think it's still pulling from deconz.

[image]

Smanar, 2020-11-24:
Yep you probably need to delete all the device from dconz or use the API, but will see later

So have checked the doorlock cluster, I have
On attribute 0x0000
```
            <value name="Not fully locked" value="0x00"></value>
            <value name="Locked" value="0x01"></value>
            <value name="Unlocked" value="0x02"></value>
            <value name="Undefined" value="0xff"></value>
```

So for me nothing usefull, the door is locked, ... or not.

And I m reading to the device documentation (realy helpfull, thanx to them), I can see schedule feature, and lot of setting missing in deconz, like language, operating mode, wrong code entry limit, ect ...

You can receive notifications for exemple for all new pin added.

So to return to alarm, it s something new in deconz, no problem easy to add, but I realy don't know how it work ^^.
According to the documentation you need to set the correct mask in the attribute 0x0040, cluster "Door lock" and you will probably receive notifications visible in logs (deconz/help/log) (Not sure because the bind is not made) and the attribute in the cluster 0x0009 will change. If you can confim me that ?

But Now I m not sure for my idea create a ZHAopenClose, for alarm I think a ZHAalarm is better.

Mdleal, 2020-11-24:
Here are all the entities on the z-wave plus version of the lock.
[image]

Smanar, 2020-11-25:
So I have added some configuration (not sure all is working) https://github.com/Smanar/deconz-rest-plugin/commit/a7b350a10d2b9c99c7394d5aff90ec4a8e886706

And added 1 debug line, I think if you enable alarm it will the way used by the device for notification.
But it can be the alarm cluster too, I don't understand if the device make a difference between alarm or notifications, and I m sure it can use the doorlock cluster using CMD_STATUS_CHANGE_NOTIFICATION to send some notifications.

Mdleal, 2020-12-04:
What do I need to provide?

Smanar, 2020-12-04:
For the configuration, just edit the xml file in /usr/share/deCONZ/zcl with https://github.com/Smanar/deconz-rest-plugin/blob/kiwiset/general.xml
You will have more setting in deconz.

For the missing information, just use the door lock, you will see some information in log starting with 'Door lock debug" (or other thing from cluster 0x0101)
To test is just enable the alarm and try to provoke defaut like
- open
- close
- wrong key
- unlock with key
- unlock without key

closed 2021-02-08 by Mimiix

Smanar, 2021-03-08:
Have done a PR for this device, if someone can test it ? https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4540