dresden-elektronik/deconz-rest-plugin #3153: TP-LINK TL-RU-SEN100Z Motion & Light Sensor 2in1

czm97176, 2020-08-16:
- Product name: TP-LINK TL-RU-SEN100Z
- Manufacturer: Can't get it in deCONZ GUI
- Model identifier: Can't get it in deCONZ GUI
- Device type :  Sensor

## Screenshots
[image]
[image]

##Log file from ZigbeeMQTT
[log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5079875/log.txt)
Please note that By 1 time of resetting, it will recevived similar message twice (JSON:20480)
ZigBee人体传感器->传感器->配置

In addition, the message with content TL-RU-SEN100Z_v1, this message is not always appear on every reset
sometime it's may only send out once per 2~3 times reset

Smanar, 2020-08-16:
Ok so no surprise, this device seem to be not zigbee compliant ^^.

He is using the basic cluster with the attribute 20480, but for different message, and I don't see how the gateway reconize them ...

But on discord yesterday you have more cluster displayed (in grey, output cluster) ?

czm97176, 2020-08-16:
Yes, sometime it can reconized as "HA switch ", sometimes it not. 
By Repeating in delete the node and add it back lead different result. it's a very weird product.

btw, i guess we will never understand how it communicated with tp-link gateway without purchasing :(:(

[image]

Smanar, 2020-08-16:
> btw, i guess we will never understand how it communicated with tp-link gateway without purchasing :(:(

I think too.
For the moment I m busy, so I m waiting if someone else have an idea, but if we can find how the gateway reconise the data send to the attribute 20480 (wich one is model id for exemple), we can start to include it in tha API.

But if they do same for other cluster too, it will not be possible without sniffing and decoding.

closed 2020-09-13 by stale[bot]