dresden-elektronik/deconz-rest-plugin #3154: Fimware not connected and MAC address = 0x0000000000

Smanar, 2020-08-16:
Ok, so first it's not a real issue, the problem is solved.

An user have played with a conbee (with zigbee2mqtt), no problem, all was fine, but when he was back to deconz, he have the "firmware not connected".
He tried to update it, without succed (the update was fine, but still not working).

[image]

He have found the problem himself, the MAC adress. So he have forced a MAC adress (different than 0) and all have worked after

## Describe the bug
He have tried to restore a backup, the deconz.conf file contain

`{"apsAck":false,"apsUseExtPanId":"0xdddddddddddddddd","curChannel":11,"deconzVersion":"20577","deviceType":0,"endpoint1":{"deviceId":"0x5","deviceVersion":"0x1","endpoint":"0x1","inClusters":["0x19","0xa"],"index":0,"outClusters":["0x500"],"profileId":"0x104"},"endpoint2":{"deviceId":"0x64","deviceVersion":"0x1","endpoint":"0xf2","inClusters":[],"index":1,"outClusters":["0x21"],"profileId":"0xa1e0"},"extPanId":"0xdddddddddddddddd","macAddress":"0x212effff055626","networkKey":"01030507010b0d0f00020406080a0c0d","nwkAddress":"0x0","nwkUpdateId":1,"otauactive":0,"panId":"0x1a62","securityMode":3,"staticNwkAddress":false,"tcAddress":"0x212effff055626","tcLinkKey":"5a6967426565416c6c69616e63653039"}`

So just with looking TC adress and macAddress we can see the backup wasn't imported.

He haven't tried the "hidden config"

## Steps to reproduce the behavior
I will not try it on my production network ^^, and for him, the bug was random, he haven't do somethng special.
Perhaps we can reproduce it with setting the MAC adress to 0x000000000 ourself.

## Expected behavior
- Avoid to be locked with an unworking device.

## Environment
"deconzVersion":"20577"
Raspberry PI
Conbee 2
Extension cable  : yes

IDK why the backup import haven't worked, the conbee need to be connected for that ?
There is nothing in the code that can check if the MAC adress is 0 and need to be re-generated ?

For information, he have used deconz with the GUI and have forced a random MAC address.

Haven't tried:
- The hidden config
- the command "GCFFlasher_internal -r"

Arquiteto, 2020-10-02:
So apparently same thing happened to me (#3345 ), somewhere along moving from HA Add-on in VM to Docker container. I'd guess it is a problem with restoring a backup.
After I fixed it I tried restoring backup again and MAC changed to 0x0000000000 again. What is curious I tried that with 2 different backup files.

Grobaa, 2023-02-08:
I know it’s been a couple of years, but hope someone can answer. 
I have the same issue now, and when I finally get to the DeCONZ setup, what am I supposed to name the MAC address? Please provide an example.

Smanar, 2023-02-09:
>what am I supposed to name the MAC address

This one is not editable ?
You mean the "extPanId" ?

Smanar, 2023-02-10:
But you have an issue ?
You want to choose your mac adress ? From my memory on deconz you can do that in the TC adress field, but what id the problem with your actual one ?

Grobaa, 2023-02-10:
Yes, when I try to connect the Homebridge DeCONZ plugin, it says that 0x0000000000 is not a RaspBee/ DeCONZ MAC address. So I guess I need to rename it, if possible, because it doesn’t find anything by itself. 
So actually I just want know, what I can name it? Format? Count of numbers/letters? Etc.

Smanar, 2023-02-10:
It's an hexa number with different values, can use all values from 0 -9 and A - F.
For exemple 0x212effff055626

You have a sample of a valid config here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help#in-case-the-network-does-still-not-come-up

But if you look on his backup the mac adress was here (even not displayed in deconz) the problem was more extPanId.

If the conbee have already worked on your side, you can use too https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues

Mimiix, 2023-02-10:
Perhaps on the box you bought it in.

Nevertheless, best would be to open a topic on the forums. Please post a screenshot of your network settings.

I'm also closing this issue as i haven't seen it anymore occuring and ik feel this is a isolated case

closed 2023-02-10 by Mimiix