dresden-elektronik/deconz-rest-plugin #3156: Aqara QBKG21LM one gang showing as 4 devices

Creatish, 2020-08-16:
## Describe the question or issue you are having

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Windows 10 
 - Firmware version: (26580700)
 - deCONZ version: (2.05.79)
 - Device:  ConBee II 
 - Do you use an USB extension cable: yes

Smanar, 2020-08-16:
Hello, still missing the attribute panel (with model id)

Creatish, 2020-08-16:
Updatw with the cluster info

Smanar, 2020-08-17:
Ok, there is something strange, this device is already filterer, but there is something not working

```
                            if (checkMacVendor(node->address(), VENDOR_JENNIC) &&
                                // prevent false positives like Immax IM-Z3.0-DIM which has only two endpoints (0x01)
                                // lumi.ctrl_neutral1 and lumi.ctrl_neutral2 have more 5 endpoints
                                node->simpleDescriptors().size() > 5  &&
                                node->nodeDescriptor().manufacturerCode() == VENDOR_JENNIC && i->endpoint() != 0x02 && i->endpoint() != 0x03)
                            {
                                // TODO better filter for lumi. devices (i->deviceId(), modelid?)
                                // blacklist switch endpoints for lumi.ctrl_neutral1 and lumi.ctrl_neutral2
                                DBG_Printf(DBG_INFO, "Skip load endpoint 0x%02X for 0x%016llX (expect: lumi.ctrl_neutral1 / lumi.ctrl_neutral2)\n", i->endpoint(), node->address().ext());
                            }
```

ebaauw, 2020-08-17:
If this code gets executed before all simple descriptors have been read, it fails.

Smanar, 2020-08-17:
So you mean the fonction void DeRestPluginPrivate::addLightNode(const deCONZ::Node *node) can be called before all simple descriptor was read ?

ebaauw, 2020-08-17:
Bingo.  This is why we need to refactor the pairing of devices, making sure all descriptors and _Basic_ cluster attributes have been read, before creating any resource, `/lights` or `/sensors`.

Smanar, 2020-08-17:
Yeah I know, every week I find a new problem to avoid in api v2 ...

Will try with this fonction for the moment to test.

@Creatish can you try with this code ?
`git clone --branch divers https://github.com/Smanar/deconz-rest-plugin.git`

Or first, do you know how to test code ?

Creatish, 2020-08-17:
no Idea how to test code, what can i doo to help

Smanar, 2020-08-18:
You just need a Unix machine, you have the procedure on github homepage https://github.com/dresden-elektronik/deconz-rest-plugin at "Install deCONZ development package (optional, Linux only)"

Just replace the step 1 by
`git clone --branch divers https://github.com/Smanar/deconz-rest-plugin.git`

Creatish, 2020-08-18:
my host is windows 10, don't have any linux machines

Smanar, 2020-08-19:
Ok so need to wait for someone with the same device and an OS on Unix ^^

Creatish, 2020-08-19:
Would it work in a Virtual machine?

Smanar, 2020-08-20:
Yes, idk wich one VM, but some users use deconz on VM.

closed 2020-09-19 by stale[bot]

russellfortune, 2020-11-03:
I am experiencing the same issue. Pairing this device via. Phoscon web app fails although I can see multiple erroneous switch devices in Home Assistant that seem to intermittently function and swap places. I have the Aqara QBKG21LM hardware and should be capable of testing a development version. What can I do to help?

Creatish, 2020-11-03:
Someone replied to this and I can't see it

I am using this device no problem really, it adds like 3 lights and 1 switch in phoscon ui aswell  as homeassistant, none of the lights function its the switch that is triggering the light, it works although as I am not sure on the speed it should be I am not sure if its slower then it should be stock to xiaomi hub.

Would be nice for all these range to work out of box as they are the best quality no neutral switches you can buy currently.

Smanar, 2020-11-03:
This device still not working ?

Since august I have delete my branch, but your problem is there is too much device created ? All is working but you have too much device created ?

russellfortune, 2020-11-03:
Sorry! It was me that added a comment and deleted it as I wasn’t able to reopen this issue and was intending to open another. 

I ran across this issue as I’m experiencing the same behaviour today - it was very difficult to pair this through Phoscon and in Home Assistant 4x switches appeared which worked cyclically (i.e switch 1 worked intermittently and then switch 2 did and so on)

I’m capable of testing development code and have the device in hand if I can be of help?

Smanar, 2020-11-03:
>4x switches appeared which worked cyclically ?

Seriously ?

russellfortune, 2020-11-03:
Yep. On my initial pairing attempt I got 3x lights and a socket as initially described. I deleted and re-paired and got 4x lights and they worked sporadically amongst each other. It’s only a 1x gang switch so I wasn’t able to make it usable from Home Assistant.

Smanar, 2020-11-03:
> I’m capable of testing development code and have the device in hand if I can be of help?

Yep, probably.

In the code we are checking some information to display only 2 devices
```
                            if (checkMacVendor(node->address(), VENDOR_JENNIC) &&
                                // prevent false positives like Immax IM-Z3.0-DIM which has only two endpoints (0x01)
                                // lumi.ctrl_neutral1 and lumi.ctrl_neutral2 have more 5 endpoints
                                node->simpleDescriptors().size() > 5  &&
                                node->nodeDescriptor().manufacturerCode() == VENDOR_JENNIC && i->endpoint() != 0x02 && i->endpoint() != 0x03)
                            {
                                // TODO better filter for lumi. devices (i->deviceId(), modelid?)
                                // blacklist switch endpoints for lumi.ctrl_neutral1 and lumi.ctrl_neutral2
                                DBG_Printf(DBG_INFO, "Skip load endpoint 0x%02X for 0x%016llX (expect: lumi.ctrl_neutral1 / lumi.ctrl_neutral2)\n", i->endpoint(), node->address().ext());
                            }
```

But it seem we have a problem in the "event order", cf the ebaauwn comment.
So can you edit the code like that

```
                        if (hasServerOnOff)
                        {
                            
                            DBG_Printf(DBG_INFO, "Xiaomi debug 1 : %d", node->simpleDescriptors().size());
                            DBG_Printf(DBG_INFO, "Xiaomi debug 2 :%d ",node->endpoints().size());
                            
                            if (checkMacVendor(node->address(), VENDOR_JENNIC) &&
```

It will display in log the problematic value, on the capture the device have 7 endpoints, but it seem we haven't the same during the process.

Creatish, 2020-11-03:
Yeah It did work intermititaly with me for a while, it ended up settling down and now the ''switch'' in deconz and ha works 100% of the time, What might help is add a phoscon group and add the switch only to it. see if after some times the switch works reliably.

Creatish, 2020-11-03:
Yep mine is working 100%, but do have 3 devices added that don't do anything, can't be for sure if this effects the speed or not, im content with it