dresden-elektronik/deconz-rest-plugin #3157: No longer able to detect or connect to devices

PhoenixMage, 2020-08-17:
I had all my devices, 16 ikea smart bulbs and a vareiety of ikea switches and remotes setup for about a week working perfectly but this morning I am unable to connect or use any of the devices. I am also unable to find the devices after doing a search in Phoscon after reseting both the lights and Phoscon.

I am using a newly purchased RaspBee II, I have attempted to reset it within Phoscon and I have reflashed the deconz firmware again, changing the zigbee channel all without success.

I am beginning to wonder of the RaspBee II has failed somehow, though it is still detected within deconz/phoscon and the firmware flash looks successul.

## Environment

 - Host system: Raspberry Pi 3b
 - Running method: Marthoc Docker container (have also tried native rasbian using the deconz image)
 - Firmware version: 26610700
 - deCONZ version: 2.05.80
 - Device: RaspBee II

### deCONZ Logs
[marthoc/deconz] Starting deCONZ...
[marthoc/deconz] Current deCONZ version: 2.05.80
[marthoc/deconz] Web UI port: 80
[marthoc/deconz] Websockets port: 443
[marthoc/deconz] VNC port: 5900
/usr/bin/xauth:  file /root/.Xauthority does not exist

New 'deconz-5496c54b65-hx7jg:0 (root)' desktop at :0 on machine deconz-5496c54b65-hx7jg

Starting applications specified in /etc/X11/Xvnc-session
Log file is /root/.vnc/deconz-5496c54b65-hx7jg:0.log

Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /root/.vnc/passwd deconz-5496c54b65-hx7jg:0 to connect to the VNC server.

libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/aarch64-linux-gnu/dri:${ORIGIN}/dri:/usr/lib/dri)
libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/aarch64-linux-gnu/dri:${ORIGIN}/dri:/usr/lib/dri)
libpng warning: iCCP: known incorrect sRGB profile
10:10:29:577 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
10:10:29:630 CTRL. 3.16.210:10:30:030 dev /dev/ttyAMA0
10:10:30:031 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
10:10:30:632 parent process /bin/sh
10:10:30:633 gw run mode: normal
10:10:30:633 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
10:10:30:673 DB sqlite version 3.16.2
10:10:30:677 DB PRAGMA page_count: 0
10:10:30:677 DB PRAGMA page_size: 4096
10:10:30:678 DB PRAGMA freelist_count: 0
10:10:30:678 DB file size 0 bytes, free pages 0
10:10:30:678 DB PRAGMA user_version: 0
10:10:30:678 DB upgrade to user_version 1
10:10:31:200 DB write sqlite user_version 1
10:10:31:222 DB PRAGMA user_version: 1
10:10:31:222 DB upgrade to user_version 2
10:10:31:271 DB write sqlite user_version 2
10:10:31:302 DB PRAGMA user_version: 2
10:10:31:302 DB upgrade to user_version 6
10:10:31:382 DB write sqlite user_version 6
10:10:31:408 DB PRAGMA user_version: 6
10:10:31:408 DB cleanup
10:10:31:411 DB create temporary views
10:10:31:430 started websocket server at port 443
10:10:31:435 create default username and password
10:10:31:468 found node plugin: libde_rest_plugin.so - REST API Plugin
10:10:31:478 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
10:10:31:583 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
10:10:31:678 dev /dev/ttyAMA0
10:10:35:022 New websocket 10.244.4.0:50709 (state: 3) 
10:10:36:571 dev /dev/ttyAMA0
10:10:37:591 Announced to internet http://dresden-light.appspot.com/discover
10:10:39:483 created username: (redacted), devicetype: Phoscon#B2560x1307
10:10:41:480 dev /dev/ttyAMA0
10:10:41:627 Device firmware version 0x26610700
10:10:41:675 unlocked max nodes: 200
10:10:41:878 Device protocol version: 0x010B
10:10:41:910 new node - ext: 0x00212effff05ba7a, nwk: 0x0000
10:10:42:042 Current channel 11
10:10:42:086 CTRL ANT_CTRL 0x03
10:10:42:178 Device protocol version: 0x010B
10:10:42:318 Current channel 11
10:10:42:361 CTRL ANT_CTRL 0x03
10:10:42:577 DB UPDATE device_descriptors SET data = x'10400f3511472b0040002b0000', timestamp = 1597623042 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:05:ba:7a') AND endpoint = 0 AND type = 2
10:10:42:577 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'10400f3511472b0040002b0000', 1597623042 FROM devices WHERE mac = '00:21:2e:ff:ff:05:ba:7a'
10:10:42:782 DB UPDATE device_descriptors SET data = x'0104010500010219000a00010005', timestamp = 1597623042 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:05:ba:7a') AND endpoint = 1 AND type = 4
10:10:42:783 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 1, 4, x'0104010500010219000a00010005', 1597623042 FROM devices WHERE mac = '00:21:2e:ff:ff:05:ba:7a'
10:10:42:885 DB UPDATE device_descriptors SET data = x'f2e0a164000100012100', timestamp = 1597623042 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:05:ba:7a') AND endpoint = 242 AND type = 4
10:10:42:886 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 242, 4, x'f2e0a164000100012100', 1597623042 FROM devices WHERE mac = '00:21:2e:ff:ff:05:ba:7a'
10:10:43:029 ZDP discovery done in 366 ms10:10:46:480 dev /dev/ttyAMA0
10:10:46:482 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26580700.bin.GCF
10:10:46:482 GW firmware version: 0x26610700
10:10:49:111 Updated password hash: (redacted)
10:10:49:120 reuse recently created auth username: A1475A8A41, devicetype: Phoscon#B2560x1307
10:10:49:120 apikey username: (redacted), devicetype: Phoscon#B2560x1307 already exists
10:10:50:899 New websocket 10.244.8.1:53103 (state: 3) 
10:10:51:402 Set bridgeid to 00212EFFFF05BA7A
10:10:52:433 saved node state in 1 ms
10:10:52:478 sync() in 44 ms
10:10:56:482 dev /dev/ttyAMA0
10:10:56:483 GW firmware version: 0x26610700
10:11:01:402 send permit join, duration: 59
10:11:06:481 dev /dev/ttyAMA0
10:11:06:484 GW firmware version: 0x26610700
10:11:10:405 LightNode 1: Configuration tool 1 added
10:11:10:406 drop request to zombie (rx = 0)
10:11:12:466 FP indication 0x0104 / 0x0000 (0x00212EFFFF05BA7A / 0x0000)
10:11:12:466                       ...     (0x0000000000000000 / 0x0000)
10:11:12:540 DB UPDATE device_descriptors SET data = x'0104010500010319000a000000010005', timestamp = 1597623072 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:05:ba:7a') AND endpoint = 1 AND type = 4
10:11:12:563 FP indication 0x0104 / 0x0000 (0x00212EFFFF05BA7A / 0x0000)
10:11:12:564                       ...     (0x0000000000000000 / 0x0000)
10:11:16:482 dev /dev/ttyAMA0
10:11:16:484 GW firmware version: 0x26610700
10:11:26:483 dev /dev/ttyAMA0
10:11:26:484 GW firmware version: 0x26610700
10:11:29:447 Current channel 11
10:11:29:469 Device TTL 2703 s flags: 0x7
10:11:32:667 FP indication 0x0104 / 0x0000 (0x00212EFFFF05BA7A / 0x0000)
10:11:32:667                       ...     (0x0000000000000000 / 0x0000)
10:11:32:740 FP indication 0x0104 / 0x0000 (0x00212EFFFF05BA7A / 0x0000)
10:11:32:740                       ...     (0x0000000000000000 / 0x0000)
10:11:36:481 dev /dev/ttyAMA0
10:11:36:482 GW firmware version: 0x26610700
10:11:46:481 dev /dev/ttyAMA0
10:11:46:481 GW firmware version: 0x26610700
10:11:56:483 dev /dev/ttyAMA0
10:11:56:484 GW firmware version: 0x26610700
10:12:00:955 Search sensors done
10:12:02:436 send permit join, duration: 59
10:12:06:489 dev /dev/ttyAMA0
10:12:06:490 GW firmware version: 0x26610700
10:12:16:482 dev /dev/ttyAMA0
10:12:16:483 GW firmware version: 0x26610700
10:12:26:482 dev /dev/ttyAMA0
10:12:26:483 GW firmware version: 0x26610700
10:12:29:448 Current channel 11
10:12:29:470 Device TTL 2643 s flags: 0x7
10:12:36:480 dev /dev/ttyAMA0
10:12:36:481 GW firmware version: 0x26610700

SwoopX, 2020-08-17:
From the debug output above, it looks like you don't have any devices at all in your database. Which debug level (`dbg-info`) was applied, 1 or 2? If not 2, then please provide logs again with level 2, thanks!

PhoenixMage, 2020-08-17:
That is the default debug on the docker image which is 1, I will set it to 2.

At the moment there are no devices in the DB as I reset the gateway and I am trying to rediscover them. I have reset the devices too but to no avail.

[deconz.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5087237/deconz.log)

SwoopX, 2020-08-18:
How do you try to do that? You're not authenticated, your log is flooded with `API error 1, /sensors/new, unauthorized user`

PhoenixMage, 2020-08-18:
That was my Home Assistant trying to authenticate after I reset it. I have deleted the integration from HA and that message has stopped.

PhoenixMage, 2020-08-21:
So weirdly this seems to be radio interference despite working fine for a week prior to everything disconnecting. I have moved the device when I started troubleshooting further today and it is now picking up devices again.

closed 2020-08-21 by PhoenixMage