dresden-elektronik/deconz-rest-plugin #3158: Iko Iko download not available

PrimaryInc, 2020-08-17:
Hi @manup, 
Will Iko Iko ble compiled for download?

Mimiix, 2020-08-17:
@PrimaryInc Hi!

you can find the compiled versions here:
http://deconz.dresden-elektronik.de/

PrimaryInc, 2020-08-17:
@Mimiix, yes correct med version Iko Iko is not there.. Yet.

Mimiix, 2020-08-17:
Is it?

[Raspbian](http://deconz.dresden-elektronik.de/raspbian/beta/deconz-2.05.80-qt5.deb)
[Ubuntu](http://deconz.dresden-elektronik.de/ubuntu/beta/deconz-dev-2.05.80.deb)

Windows isn't there yet. I'll ask around for that one.

PrimaryInc, 2020-08-17:
@Mimiix sorry should have been specific. Yes windows 👍

Mimiix, 2020-08-19:
@PrimaryInc ITS HERE!!!!!!!!
http://deconz.dresden-elektronik.de/win/deCONZ_Setup_Win32_V2_05_80.exe

PrimaryInc, 2020-08-19:
@Mimiix thank you!!

closed 2020-08-19 by PrimaryInc