dresden-elektronik/deconz-rest-plugin #3159: OBLO SPW35ZD0 SPW35Z-D0 Smart Plug - Power Metrics

SeoFood, 2020-08-17:
The on/off works perfect for the device but it doesn't create Power Metrics Sensors.

## Device

- Product name: OBLO SPW35ZD0
- Manufacturer: RT-RK
- Model identifier: SPW35Z-D0
- Device type : Switch

## Screenshots

[image]

[image]

[image]

[image]

SwoopX, 2020-08-17:
Simple metering cluster is missing as well as the basic cluster.

SeoFood, 2020-08-17:
[image]
[image]
[image]

closed 2020-09-15 by manup