dresden-elektronik/deconz-rest-plugin #3161: Tobular Motor Zemismart ZM25TQ Zigbee version

andysaab, 2020-08-17:
Hey,
I got new zigbee roll-motor, but they are not supported by deconz.

Is there any chance to add support for this one?
## Device

- Product name: ZM25TQ
- Manufacturer: _TYST11_wmcdj3aq
- Model identifier: mcdj3aq
- Device type : Roll-motor 

## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2020-08-17:
I thought it was going to be complicated just when I saw the label "tuya" ....
Have you tried the command "on" and "off" in deconz, to see if the device works with thoses command ?

andysaab, 2020-08-17:
Yes, I've tried that before. No effect.

andysaab, 2020-08-17:
[image]

Smanar, 2020-08-17:
Ok so I think there is an hidden cluster tuya somwhere, but first time I see a shutter on this cluster.

Need to find the tuya values to use in their cluster for this kind of device.

andysaab, 2020-08-17:
Um, can I help you with something?

Smanar, 2020-08-18:
Yep, I m still on it but you can try to test the code.

You have the procedure on github page https://github.com/dresden-elektronik/deconz-rest-plugin at "Install deCONZ development package (optional, Linux only)"
Just replace step 1 by 
`git clone --branch tuya2 https://github.com/Smanar/deconz-rest-plugin.git`

and can skip " git checkout -b mybranch HEAD"

ATM, if you re-include the device you will have the field "open" in the JSON and as type "Window covering device" (If I m right)

andysaab, 2020-08-18:
> Yep, I m still on it but you can try to test the code.
>
> You have the procedure on github page https://github.com/dresden-elektronik/deconz-rest-plugin at "Install deCONZ development package [...]

Smanar, 2020-08-18:
Not with HA, but on other Unix OS T think yes.

andysaab, 2020-08-18:
Unfortunately, I don't have any other Unix OS. 
There are no other options?

Smanar, 2020-08-19:
Herrr, not realy.
I will make the code in blind mode, I m on it for other device, but impossible to test it without an Unix OS.

Smanar, 2020-08-25:
@andysaab BTW wich one machine have you used for capture ?

andysaab, 2020-08-25:
> @andysaab BTW wich one machine have you used for capture ?

Hello, Smanar! Sorry for delay. Right now I migrate from Syno docker`s HA into Syno Virtual Machine for using legal image of HA. After that I planning install any UnixOS to VM and try to do yours recomendation above.

Smanar, 2020-08-25:
That you want deconz work on Ubuntu, debian, Raspbian.
But BTW, to take the capture you was already on a Unix OS, and was not Hassio, so what is it ?

To test if your OS is compatible just try to install deconz dev 
`sudo apt install deconz-dev`

With OS updated and APT repository for deCONZ added (from deconz installation)

andysaab, 2020-08-25:
> That you want, deconz work on Ubuntu, debian, Raspbian.
Or, will install RaspbianOS.
> But BTW, to take the capture you was already on a Unix OS, and was not Hassio, so what is it ?
Smanar, previously i used Hassio, but Deconz was in a separate docker container, not add-on.

andysaab, 2020-08-26:
> ATM, if you re-include the device you will have the field "open" in the JSON and as type "Window covering device" (If I m right)
Hi, Smanar! Idid your recomendation. Nothing changed. Is it same situation.
[image]
[image]
[image]

Smanar, 2020-08-26:
Hello, you need to check the json.
Visible in your home automation application, or using the API `http://IP:PORT/api/KEY/lights`
For IP and PORT, same you are using for phoscon, for the key, the same you are using in your application or https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

Phoscon don't support covering (yet)

andysaab, 2020-08-26:
> Visible in your home automation application, or using the API `http://IP:PORT/api/KEY/lights`
> For IP and PORT, same you are using for phoscon, for the key, the same you are using in your application or https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/
Can`t got API key. I`m faced this a first time. Early on other stick (on Windows machine) got key w/o any problems.

[image]

Smanar, 2020-08-26:
If you haven't api key, you can use that

`curl http://127.0.0.1:80/api/ -X POST -d '{"devicetype": "TestKey"}'`

andysaab, 2020-08-26:
> If you haven't api key, you can use that
> 
> `curl http://127.0.0.1:80/api/ -X POST -d '{"devicetype": "TestKey"}'`

[image]

Smanar, 2020-08-26:
Ok so there is not the field I need ...

You have compiled the code using "qmake && make -j2" and copied the file with "sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins"

No error message ?

You have included the device after the modification ?

I m re checking code atm

andysaab, 2020-08-26:
> Ok so there is not the field I need ...
> 
> You have compiled the code using "qmake && make -j2" and copied the file with "sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins"
Yes, did all steps with no errors
> You have included the device after the modification ?
Yes, I disconnect motor from other network and pairing in this network. 
> I m re checking code atm

Smanar, 2020-08-26:
In fact I m waiting a "open" field in "state".
You probably miss it, because the tuya cluster is not visible.

So if you are sure you are right with the procedure, I have modified the code a little, to try it (you will it s faster now)
1- go in folder
2 - `git pull`
3 - `qmake && make -j2`
4 - `sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins`

https://github.com/Smanar/deconz-rest-plugin/commit/4e37cadb211748d01c3ed64d7cb0a17c093e54b8

andysaab, 2020-08-26:
Ok, try it tomorrow with connected device.
Tnx, Smanar

Smanar, 2020-08-26:
Yep sorry, deconz have so much problem with tuya cluster, need to use work around, but your device work on smarthing, so no reason for it don't work here.

Just need more tests.

andysaab, 2020-08-27:
Hi Smanar,
look at screen.
[image]

andysaab, 2020-08-27:
> In fact I m waiting a "open" field in "state".
> You probably miss it, because the tuya cluster is not visible.
Hi Smanar!
Look at picture
[image]

Smanar, 2020-08-27:
Still missing.
You are using only 1 deconz instance ? you are not using one VM for compilation, and another for test ?

I have done a realy agressive modification (already on the fork), it add the "open" field for ALL device with manufacture code = 0, will not be a problem for you (you haven't lot of device).
But after that if we still haven't the field, I realy don't understand.

andysaab, 2020-08-27:
Smanar, I have a two ConbeeII. One of this used in VM and ready for next tests. Now I don't depend on the working system

Smanar, 2020-08-27:
You make me think I need to find another raspberry for compilation, for the same reason ...

But my worry is more software than hardware.
BTW now I m sure of one thing, on some tuya device the tuya cluster is realy not visible, we can do anything we want, we will not see it.

andysaab, 2020-08-27:
It sounds pessimistic.

Smanar, 2020-08-27:
lol, it s not the problem, we can use it, even if deconz don't see it.

The problem is for reconize device, but for you it s not a problem you are the first one with this Device ID

Smanar, 2020-08-27:
Tell me when you will have tested the new code, and if you have the "open" filed !

andysaab, 2020-08-27:
I guess tomorrow evening, Smanar. Now too late.

andysaab, 2020-08-27:
Or you can set several goals, I`ll try to do and shown the result.

andysaab, 2020-08-28:
> Tell me when you will have tested the new code, and if you have the "open" filed !

Hi Smanar!
I am ready for the good time.

Smanar, 2020-08-28:
Hello ^^, do you have the "open" field now ?

andysaab, 2020-08-28:
I`m so sorry. I think I understood you. What should I do?

andysaab, 2020-08-28:
I did this speps again:
1- go in folder
2 - git pull
3 - qmake && make -j2
4 - sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
After that got crash deconz
[image]

andysaab, 2020-08-28:
Now restart deconz app.
I think now we can see that we want.
[image]

Smanar, 2020-08-28:
>sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
This command change the lib file, so I think changing it during deconz is running can cause your crash.

But I haven't find yet why it don't work with "cool mode", but now we are sure the procedure is ok.
Can try with the API

`curl -H 'Content-Type: application/json' -X PUT -d '{"open": true}' http://127.0.0.1:8080/api/F0E614FAFE/lights/2/state`

If it don't work, can you tell me the error message ?

andysaab, 2020-08-28:
[image]

andysaab, 2020-08-28:
OMG! My daughter power off motor. Now fixed.
[image]

Smanar, 2020-08-28:
parameter open not available ?
I m perhaps wrong but your device have 2 as ID and still have "open" in "state" ?

Smanar, 2020-08-28:
I think i have made mistake on name "1" or "l".
Can you make a copy/paste with the JSON ?

andysaab, 2020-08-28:
{"1":{"etag":"f0e9760e1fb607e87536b68110ed2826","hascolor":false,"lastannounced":null,"lastseen":"2020-08-28T16:42Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","state":{"reachable":true},"swversion":"0x264a0700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:04:f2:df-01"},"2":{"etag":"2582a3b307f3e922feda6bf028d87e29","hascolor":false,"lastannounced":null,"lastseen":"2020-08-28T16:42Z","manufacturername":"_TYST11_wmcdj3aq","modelid":"mcdj3aq","name":"Maybe_roll_motor","state":{"alert":"none","on":false,"open":false,"reachable":true},"swversion":"20180727","type":"On/Off light","uniqueid":"bc:33:ac:ff:fe:68:0b:de-01"}}

Smanar, 2020-08-28:
Thx, corrected, you can remake 
2 - git pull
3 - qmake && make -j2
4 - sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins

andysaab, 2020-08-28:
[image]

Smanar, 2020-08-28:
And the motor don't move ?

andysaab, 2020-08-28:
Yes, spinned!

Smanar, 2020-08-28:
Ha ^^, so you can close and open it with "open" and false and true ?

andysaab, 2020-08-28:
> Ha ^^, so you can close and open it with "open" and false and true ?

It is a problem. I couldn't get the API key for this stick before and I can't add it to the HA. I don't know how to push the buttons any other way.

Smanar, 2020-08-28:
I don't understand.
The command using the api are working or not ?
"Push the button" manually or using sofware ?
Why you want to add it already to HA ?

The api key you are using ATM can only work for this deconz instance.

andysaab, 2020-08-28:
I think I clear!
Should change "open": true to "open": false
curl -H 'Content-Type: application/json' -X PUT -d '{"open": false}' http://127.0.0.1:8080/api/F0E614FAFE/lights/2/state
Right?

Smanar, 2020-08-28:
Yep

andysaab, 2020-08-28:
> I don't understand.
> The command using the api are working or not ?
> "Push the button" manually or using software ?
> Why you want to add it already to HA ?
> 
> The api key you are using ATM can only work for this deconz instance.

andysaab, 2020-08-28:
> Yep
Yes, it is rotate to other direction!
Now we need STOP command, I think?

Smanar, 2020-08-28:
Stop and Setlevel (will add the "lift" field is same time)
But going to eat ATM ^^, you will have it at the latest tommorow, will be easy.

andysaab, 2020-08-28:
Thx, Smanar!
You are the best!

Smanar, 2020-08-28:
Ok so on the new code (not tested, so if you have compilation error, stop direclty)
- Stop was already in code in fact
- Have added "lift" field (need re-include)
- Have added fake field to test "sat", totaly useless was just to test future feature
- Have enabled state return.

curl -H 'Content-Type: application/json' -X PUT -d '{"stop": true}' http://127.0.0.1:8080/api/F0E614FAFE/lights/2/state
curl -H 'Content-Type: application/json' -X PUT -d '{"lift": 50}' http://127.0.0.1:8080/api/F0E614FAFE/lights/2/state

andysaab, 2020-08-29:
> Ok so on the new code (not tested, so if you have compilation error, stop direclty)
/QtCore -Irelease -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o release/rest_lights.o rest_lights.cpp
rest_lights.cpp: In member function ‘int DeRestPluginPrivate::setWindowCoveringState(const ApiRequest&, ApiResponse&, TaskItem&, QVariantMap&)’:
rest_lights.cpp:1748:63: error: call of overloaded ‘append(qint8)’ is ambiguous
             lev.append((qint8)((targetLiftZigBee >> 8) & 0xff));
                                                               ^
In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/qstring.h:42:0,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/QString:1,
                 from rest_lights.cpp:11:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qbytearray.h:293:17: note: candidate: QByteArray& QByteArray::append(char)
     QByteArray &append(char c);
                 ^
/usr/include/x86_64-linux-gnu/qt5/QtCore/qbytearray.h:294:17: note: candidate: QByteArray& QByteArray::append(const char*)
     QByteArray &append(const char *s);
                 ^
/usr/include/x86_64-linux-gnu/qt5/QtCore/qbytearray.h:296:17: note: candidate: QByteArray& QByteArray::append(const QByteArray&)
     QByteArray &append(const QByteArray &a);
                 ^
In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/QString:1:0,
                 from rest_lights.cpp:11:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qstring.h:1241:20: note: candidate: QByteArray& QByteArray::append(const QString&)
 inline QByteArray &QByteArray::append(const QString &s)
                    ^
Makefile.Release:908: recipe for target 'release/rest_lights.o' failed
make[1]: *** [release/rest_lights.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/andysaab/deconz-rest-plugin'
Makefile:38: recipe for target 'release' failed
make: *** [release] Error 2

[84 more replies]

closed 2020-10-01 by manup