dresden-elektronik/deconz-rest-plugin #3162: Ikea Fyrtur - Devices connected / Cluster information missing

GoTTi74, 2020-08-17:
Dear community

I need your support. I recently bought 6 Ikea Fyrtur blinds and I'm trying to integrate them into deCONZ (via Raspbee II). After initial issues, I was able to get all 6 devices to join deCONZ. However, after >24h only 4 of the 6 show cluster information and are available in the WebApp (see screenshot)

[image]

Any idea/advice on how to get the other two devices fully added to the environment? Note: I already performed a full/factory reset of the deCONZ application with the same result. I also have tried to "factory-reset" the blinds (pressing both buttons >5 sec.). 

### Environment

-  Host system: (Raspberry Pi)
-  Running method: (Raspbian)
-  Firmware version: 26610700 (2.05.80 / 8/14/2020)
-  deCONZ version: 2.05.80
-  Device: RaspBee II

### deCONZ Logs
Let me know in case I need to provide any logs (I'll figure out how to retrieve them ;) )

### Additional context
I was hoping that I won't need the Tradfri Gateway but read in another forum that it's the only way you can upgrade the firmware on the blinds... so I will make another trip to IKEA later... and hopefully that will help to fully add the two other blinds...

Other devices in the network:
- 3 Philipps Hue bulbs (will add much more once the IKEA blinds are working)
- 4 IKEA range extenders

rockyluke, 2020-08-18:
Hi there!

Since v2.05.80 my **Fyrtur** also stopped working through deCONZ.

## Environment

- Host system: (Raspberry Pi 4)
- Running method: (Raspbian)
- Firmware version: 26610700
- deCONZ version: 2.05.80 / 8/14/2020
- Device: RaspBee II

## deCONZ Logs

Let me know in case I need to provide any logs.

## Additional context

It was working very well with version 2.09.79, just before the 2.05.80 upgrade.

Thanks 🙏

Mimiix, 2020-08-18:
Interesting. @ebaauw  did a change in https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3087 on covering.

Erik, could you take a look on this?

ebaauw, 2020-08-18:
> ebaauw did a change in #3087 on covering.

That was a change in `rest_group.cpp` to support `action.open`.  Nothing to do with FYRTUR support nor pairing.

> However, after >24h only 4 of the 6 show cluster information and are available in the WebApp (see screenshot) 

Looks like deCONZ hasn’t read the descriptors.  Best search for new devices in Phoscon and manually read the simple descriptor from the GUI.  That should populate the right circle, and cause the REST resources to be created.  Alternatively, reset and re-pair the blinds.

> Since v2.05.80 my Fyrtur also stopped working through deCONZ.

I doubt that this is related to the update.  My FYRTUR still becomes incommunicado at times.  Power cycling the blind (remove the battery for 10s) solves this.

GoTTi74, 2020-08-18:
After another 24 hours ~48 in total, the REST resources have finally been created. So, it seems, sometimes it requires more patience than I have 😂

GoTTi74, 2020-08-18:
Quick correction. Although the “right circle” was visible in deCONZ GUI, the two devices still don’t show up in the deCONZ or Phoscom web UI. After following @ebaauw’s suggestion (manually read simple descriptor), the devices are now fully available.

closed 2020-09-20 by stale[bot]