dresden-elektronik/deconz-rest-plugin #3163: New device request - D-Link DCH-B112

shjacobs303, 2020-08-18:
Smart Door / Window Sensor

[image]

[image]

[image]

[image]

[image]

[image]

shjacobs303, 2020-08-18:
[image]

Mimiix, 2020-08-18:
Hi!

I do appreciate the request, as you seem to have the screenshots. But please, could you use the format as per issue template?

Thank you!

SwoopX, 2020-08-25:
I'd need the attributes of the power configuration cluster to be read. Trigger the device by open/close and then directly press the read button.

closed 2020-09-27 by shjacobs303