dresden-elektronik/deconz-rest-plugin #3165: TS0111 - 16A EU Zigbee plug (not switching in Phoscon)

TH1485, 2020-08-18:
## Device

- Product name: 16A EU Zigbee  plug
- Manufacturer: Rehentele
- Model identifier: TS0111
- Device type :  Light (plug)
  
## Screenshots
**Node info:**
[image]

**Basic cluster attributes and endpoint/clusters of the node:**
[image]

## Additional information
- on-off switching from  deCONZ, actually works. Toggle however does not work. 
- on-off Switching with the phoscon app does not work / as well as switching from home assistant.

picture:
[image]
link for sales:
https://it.aliexpress.com/item/4000449939329.html?spm=a2g0s.9042311.0.0.63ff4c4dhSCNdG

Smanar, 2020-08-19:
Hello, wich one button are you using in deconz GUI ?

I m asking because the api use the same command, so I dont' understand why it don't work on both ....

TH1485, 2020-08-19:
> Hello, wich one button are you using in deconz GUI ?
> 
> I m asking because the api use the same command, so I dont' understand why it don't work on both ....

Deconz :. I use exec for on and exec for off. Toggle does not work.

When I switch in phoscon or home assistant it immediately switches back to "on-state" and nothing happens with the plug. 
I noticed today when in press multiple times fast after each other , the plug eventually switches from state some times.

The state in phoscon or home assistant is also wrongly represented as always on. Only at the moment you switch for a short period it represents off.

Smanar, 2020-08-20:
Ha !
So it s perhaps a return state problem ? you can listen the relay ?

When you are using the GUI, deconz don't take care of return (just display a message)
But when you use the API, deconz change the state first, send the command after and take care of return.

TH1485, 2020-08-20:
Hi 

I think you are right, but how to solve this?

Smanar, 2020-08-20:
On the cluster 0006, you have the command, but you have too a table with attribute.

Deconz is looking for the value 0x0000 to set the light state, you can check if you see something not normal in it.

In last solution, need to enable log

TH1485, 2020-08-20:
> On the cluster 0006, you have the command, but you have too a table with attribute.
> 
> Deconz is looking for the value 0x0000 to set the light state, you can check if you see something not normal in it.
> 
> In last solution, need to enable log

0x0000;  correctly respresents the state (after pressing read).  After switching it does not immediately change to the right state, i have to press read a few times, takes roughly 1-2 seconds.

TH1485, 2020-08-30:
Hi 

Hopefully, you can analyse this. The plug in question is:

> manufacturername":"_TYZB01_ougeeqtm","modelid":"TS0111","name":"Woonkamer 

`2020-08-30 10:45:28 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"7","r":"groups","state":{"all_on":true,"any_on":true},"t":"event"}
2020-08-30 10:45:30 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': False}" to "172.30.33.0 /lights/11/state"
2020-08-30 10:45:30 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"3","r":"sensors","state":{"current":3250,"lastupdated":"2020-08-30T08:45:30.464","power":191,"voltage":235},"t":"event","uniqueid":"00:0d:6f:00:14:ba:e0:70-01-0b04"}
2020-08-30 10:45:30 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":null,"on":false,"reachable":true},"t":"event","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"}
2020-08-30 10:45:30 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"7","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"}
2020-08-30 10:45:30 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/11/state/on': False}}]
2020-08-30 10:45:31 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"4","r":"sensors","state":{"consumption":0,"lastupdated":"2020-08-30T08:45:30.585"},"t":"event","uniqueid":"ec:1b:bd:ff:fe:ac:52:f3-01-0702"}
2020-08-30 10:45:31 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":null,"lastseen":"2020-08-30T08:45:30Z","manufacturername":"_TYZB01_ougeeqtm","modelid":"TS0111","name":"Woonkamer Plug","swversion":"0xFFFFFFFF","type":"On/Off light","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"},"e":"changed","id":"11","r":"lights","t":"event","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"}
2020-08-30 10:45:31 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":null,"on":true,"reachable":true},"t":"event","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"}
2020-08-30 10:45:31 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"7","r":"groups","state":{"all_on":true,"any_on":true},"t":"event"}
2020-08-30 10:45:31 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': False}" to "172.30.33.0 /lights/11/state"
2020-08-30 10:45:31 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"5","r":"sensors","state":{"current":0,"lastupdated":"2020-08-30T08:45:31.556","power":0,"voltage":0},"t":"event","uniqueid":"ec:1b:bd:ff:fe:ac:52:f3-01-0b04"}
2020-08-30 10:45:32 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"2","r":"sensors","state":{"consumption":67046,"lastupdated":"2020-08-30T08:45:31.687","power":191},"t":"event","uniqueid":"00:0d:6f:00:14:ba:e0:70-01-0702"}
2020-08-30 10:45:32 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":null,"on":false,"reachable":true},"t":"event","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"}
2020-08-30 10:45:32 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"7","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"}
2020-08-30 10:45:32 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/11/state/on': False}}]
2020-08-30 10:45:32 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":null,"lastseen":"2020-08-30T08:45:32Z","manufacturername":"_TYZB01_ougeeqtm","modelid":"TS0111","name":"Woonkamer Plug","swversion":"0xFFFFFFFF","type":"On/Off light","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"},"e":"changed","id":"11","r":"lights","t":"event","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"}
2020-08-30 10:45:32 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":null,"on":true,"reachable":true},"t":"event","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"}
2020-08-30 10:45:32 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"7","r":"groups","state":{"all_on":true,"any_on":true},"t":"event"}
2020-08-30 10:45:32 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":null,"lastseen":"2020-08-30T08:45:32Z","manufacturername":"_TYZB01_iuepbmpv","modelid":"TS0121","name":"vestibule_licht","swversion":null,"type":"Smart plug","uniqueid":"ec:1b:bd:ff:fe:ac:52:f3-01"},"e":"changed","id":"4","r":"lights","t":"event","uniqueid":"ec:1b:bd:ff:fe:ac:52:f3-01"}
2020-08-30 10:45:33 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"65520","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"}
2020-08-30 10:45:34 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":null,"lastseen":"2020-08-30T08:45:34Z","manufacturername":"eWeLink","modelid":"SA-003-Zigbee","name":"kelder_licht","swversion":null,"type":"On/Off plug-in unit","uniqueid":"00:12:4b:00:1f:52:61:b1-01"},"e":"changed","id":"2","r":"lights","t":"event","uniqueid":"00:12:4b:00:1f:52:61:b1-01"}
2020-08-30 10:45:35 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":null,"lastseen":"2020-08-30T08:45:35Z","manufacturername":"Heiman","modelid":"SmartPlug","name":"wasmachine","swversion":"2018.1.11","type":"Smart plug","uniqueid":"00:0d:6f:00:14:ba:e0:70-01"},"e":"changed","id":"3","r":"lights","t":"event","uniqueid":"00:0d:6f:00:14:ba:e0:70-01"}
2020-08-30 10:45:35 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"65520","r":"groups","state":{"all_on":false,"any_on":true},"t":"event"}
[... cut ...] :00:14:ba:e0:70-01"},"e":"changed","id":"3","r":"lights","t":"event","uniqueid":"00:0d:6f:00:14:ba:e0:70-01"}
2020-08-30 10:46:06 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"65520","r":"groups","state":{"all_on":false,"any_on":true},"t":"event"}
2020-08-30 10:46:06 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"2","r":"sensors","state":{"consumption":67046,"lastupdated":"2020-08-30T08:46:06.691","power":11},"t":"event","uniqueid":"00:0d:6f:00:14:ba:e0:70-01-0702"}`

Smanar, 2020-08-30:
It s only http log, I wanted to see if there were no zigbee errors.

BTW on your log you are trying to set them "on" only 1 time, but just for 1s

> 2020-08-30 10:45:55 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': True}" to "172.30.33.0 /lights/11/state"
> 2020-08-30 10:45:56 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/11/state/on': True}}]
> 2020-08-30 10:45:56 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":null,"lastseen":"2020-08-30T08:45:56Z","manufacturername":"_TYZB01_ougeeqtm","modelid":"TS0111","name":"Woonkamer Plug","swversion":"0xFFFFFFFF","type":"On/Off light","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"},"e":"changed","id":"11","r":"lights","t":"event","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"}
> 2020-08-30 10:45:56 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': False}" to "172.30.33.0 /lights/11/state"
> 2020-08-30 10:45:56 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/11/state/on': False}}]
> 2020-08-30 10:45:56 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":null,"on":false,"reachable":true},"t":"event","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"} 2020-08-30 10:45:56 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"7","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"}

If it s not you that have send the "off" command less than 1s after the "on" command, I think you have a problem in your automation.
BTW deconz is changing the device state before sending the command. So if your device is offline, you can change the visible state in the api even the device don't react. So not normal nothing change in your situation, on a normal bugged situation the visible state will change but the device not.

TH1485, 2020-09-01:
Here the hassio log,  i did the following:

1. switch off in HASSIO (this worked)
2. state went briefly to off and then back to on ( plug stayed off)
3. tried to switch again  multiple times (nothing happened)
4. switched on via hardware button on plug (worked)
5.  switch off in hassio (this worked again) 
6. repeat to step 2.

Hassio server log:

`01T19:07:42Z","manufacturername":"_TYZB01_ougeeqtm","modelid":"TS0111","name":"Woonkamer Plug","swversion":"0xFFFFFFFF","type":"On/Off light","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"},"e":"changed","id":"11","r":"lights","t":"event","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"}
2020-09-01 21:07:43 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"11","r":"lights","state":{"alert":null,"on":false,"reachable":true},"t":"event","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"}
2020-09-01 21:07:43 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"7","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"}
2020-09-01 21:07:44 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":"2020-08-16T17:02:59Z","lastseen":"2020-09-01T19:07:44Z","manufacturername":"Heiman","modelid":"TS0115","name":"On/Off plug-in unit 6","swversion":null,"type":"On/Off plug-in unit","uniqueid":"84:2e:14:ff:fe:fd:ac:45-01"},"e":"changed","id":"6","r":"lights","t":"event","uniqueid":"84:2e:14:ff:fe:fd:ac:45-01"}
2020-09-01 21:07:44 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":"2020-08-16T17:02:59Z","lastseen":"2020-09-01T19:07:44Z","manufacturername":"Heiman","modelid":null,"name":"On/Off plug-in unit 7","swversion":null,"type":"On/Off plug-in unit","uniqueid":"84:2e:14:ff:fe:fd:ac:45-02"},"e":"changed","id":"7","r":"lights","t":"event","uniqueid":"84:2e:14:ff:fe:fd:ac:45-02"}
2020-09-01 21:07:44 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":"2020-08-16T17:02:59Z","lastseen":"2020-09-01T19:07:44Z","manufacturername":"Heiman","modelid":null,"name":"On/Off plug-in unit 8","swversion":null,"type":"On/Off plug-in unit","uniqueid":"84:2e:14:ff:fe:fd:ac:45-03"},"e":"changed","id":"8","r":"lights","t":"event","uniqueid":"84:2e:14:ff:fe:fd:ac:45-03"}
2020-09-01 21:07:44 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":"2020-08-16T17:02:59Z","lastseen":"2020-09-01T19:07:44Z","manufacturername":"Heiman","modelid":null,"name":"On/Off plug-in unit 9","swversion":null,"type":"On/Off plug-in unit","uniqueid":"84:2e:14:ff:fe:fd:ac:45-04"},"e":"changed","id":"9","r":"lights","t":"event","uniqueid":"84:2e:14:ff:fe:fd:ac:45-04"}
2020-09-01 21:07:44 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":"2020-08-16T17:02:59Z","lastseen":"2020-09-01T19:07:44Z","manufacturername":"Heiman","modelid":null,"name":"On/Off plug-in unit 10","swversion":null,"type":"On/Off plug-in unit","uniqueid":"84:2e:14:ff:fe:fd:ac:45-07"},"e":"changed","id":"10","r":"lights","t":"event","uniqueid":"84:2e:14:ff:fe:fd:ac:45-07"}
2020-09-01 21:07:46 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"4","r":"sensors","state":{"consumption":0,"lastupdated":"2020-09-01T19:07:46.056"},"t":"event","uniqueid":"ec:1b:bd:ff:fe:ac:52:f3-01-0702"}
2020-09-01 21:07:47 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"5","r":"sensors","state":{"current":0,"lastupdated":"2020-09-01T19:07:47.056","power":0,"voltage":0},"t":"event","uniqueid":"ec:1b:bd:ff:fe:ac:52:f3-01-0b04"}
2020-09-01 21:07:47 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":null,"lastseen":"2020-09-01T19:07:47Z","manufacturername":"_TYZB01_ougeeqtm","modelid":"TS0111","name":"Woonkamer Plug","swversion":"0xFFFFFFFF","type":"On/Off light","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"},"e":"changed","id":"11","r":"lights","t":"event","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"}
2020-09-01 21:07:48 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":null,"lastseen":"2020-09-01T19:07:48Z","manufacturername":"_TYZB01_iuepbmpv","modelid":"TS0121","name":"vestibule_licht","swversion":null,"type":"Smart plug","uniqueid":"ec:1b:bd:ff:fe:ac:52:f3-01"},"e":"changed","id":"4","r":"lights","t":"event","uniqueid":"ec:1b:bd:ff:fe:ac:52:f3-01"}
2020-09-01 21:07:49 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"65520","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"}
2020-09-01 21:07:49 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":null,"lastseen":"2020-09-01T19:07:49Z","manufacturername":"_TYZB01_ougeeqtm","modelid":"TS0111","name":"Woonkamer Plug","swversion":"0xFFFFFFFF","type":"On/Off light","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"},"e":"changed","id":"11","r":"lights","t":"event","uniqueid":"ec:1b:bd:ff:fe:32:74:a9-01"}
2020-09-01 21:07:49 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":null,"lastseen":"2020-09-01T19:07:49Z","manufacturername":"Heiman","modelid":"SmartPlug","name":"wasmachine","swversion":"2018.1.11","type":"Smart plug","uniqueid":"00:0d:6f:00:14:ba:e0:70-01"},"e":"changed","id":"3","r":"lights","t":"event","uniqueid":"00:0d:6f:00:14:ba:e0:70-01"}
2020-09-01 21:07:49 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"65520","r":"groups","state":{"all_on":false,"any_on":true},"t":"event"}
2020-09-01 21:07:51 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":"2020-08-16T17:02:59Z","lastseen":"2020-09-01T19:07:51Z","manufacturername":"Heiman","modelid":"TS0115","name":"On/Off plug-in unit 6","swversion":null,"type":"On/Off plug-in unit","uniqueid":"84:2 [... cut ...]
21:09:11:907 don't close database yet, keep open for 900 seconds
21:09:15:271 0xEC1BBDFFFEAC52F3 error APSDE-DATA.confirm: 0xE9 on task
21:09:16:347 0xEC1BBDFFFEAC52F3 error APSDE-DATA.confirm: 0xE9 on task
21:09:17:273 don't close database yet, keep open for 900 seconds
21:09:20:071 0x842E14FFFEFDAC45 error APSDE-DATA.confirm: 0xE9 on task
21:09:22:767 don't close database yet, keep open for 900 seconds
21:09:23:995 0x842E14FFFEFDAC45 error APSDE-DATA.confirm: 0xE1 on task
21:09:25:134 0x842E14FFFEFDAC45 error APSDE-DATA.confirm: 0xE1 on task`

Smanar, 2020-09-01:
I realy don't understand, it seem deconz don't use report attribute (goog it s not working) but use pooling

```
21:07:27:210 0xEC1BBDFFFE3274A9 force poll (2)
21:07:27:224 ZCL attribute report 0xEC1BBDFFFE3274A9 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
```

But why the values is not correctly updated ?
Perhaps deconz need to wait 3/3 s before the pool, but he do that every 10s, so even it miss one, the next one need to correct the sate ...

Smanar, 2020-09-01:
BTW in the fonction void DeRestPluginPrivate::handleZclAttributeReportIndication() I can see the fonction handleZclAttributeReportIndicationXiaomiSpecial() for Xiaomi device, but where is managed the other device ? From my memory philips too use pooling, but I don't find the code.

TH1485, 2020-09-01:
> BTW in the fonction void DeRestPluginPrivate::handleZclAttributeReportIndication() I can see the fonction handleZclAttributeReportIndicationXiaomiSpecial() for Xiaomi device, but where is managed the others devices ? From my memory philips too use pooling, but I don't find the code.

Sorry can't help here. 
How do I set the log level to 2, sorry quite new to this

Smanar, 2020-09-02:
I think with changing setting

`-e DEBUG_INFO=2`

Mimiix, 2020-09-03:
@TH1485 That message is because you can't specify the reporting rate. 

@Smanar The Blitzwolf BHP13 gives the same error.

TH1485, 2020-09-03:
> I think with changing setting
> 
> `-e DEBUG_INFO=2`

I did the debug for hassio you had as option 2 , cannot change this there.

Smanar, 2020-09-03:
On the api code

```
    if (DBG_IsEnabled(DBG_INFO))
    {
        DBG_Printf(DBG_INFO, "ZCL attribute report 0x%016llX for cluster: 0x%04X, ep: 0x%02X, frame control: 0x%02X, mfcode: 0x%04X \n", ind.srcAddress().ext(), ind.clusterId(), ind.srcEndpoint(), zclFrame.frameControl(), zclFrame.manufacturerCode());
    }

    if (DBG_IsEnabled(DBG_INFO_L2))
    {
        DBG_Printf(DBG_INFO_L2, "\tpayload: %s\n", qPrintable(zclFrame.payload().toHex()));
    }
```

If debug_info = 1 you have the first line
>ZCL attribute report

But if debug_info = 2 you have all the data.

Mimiix, 2020-09-03:
@Mimiix Yes. That's fine. The BHP13 reports every 10 seconds.

Smanar, 2020-09-03:
So the BHP13 is working ? you are speaking about an error ?

Mimiix, 2020-09-03:
Yes.

The error he gets on the screen: failed UNSUP_GENERAL_COMMAND

That's what i get on that one too. but it just reports every 10 sec.

closed 2020-10-04 by stale[bot]