dresden-elektronik/deconz-rest-plugin #3166: Gledopto GL-C-009

michi1g, 2020-08-19:
I paired my Gledopto GL-C-009 controller with deconz but it only shows me one device and I can't change from CW to WW. Both are on all the time and I can only dim both at a time. Is there a workaround?
Screenshots:
[image]

[image]

ebaauw, 2020-08-19:
According to my (probably outdated) info, the GL_C_009 is a dimmer only?  How many channels does it drive?  What are they labelled?  Could you post a photo of the controller?

The node in deCONZ seems consistent with the device being a dimmer only (a _Dimmable Light_ on endpoint 0x0B), except for the _Color Control_ cluster.

> Both are on all the time and I can only dim both at a time.

Both what?  The WW and CW channels?

>  it only shows me one device

Is this a 1-ID or a 2-ID version?  What does the `/lights` resource look like?  Could you try and re-read the descriptors from the GUI, just to be sure.

The 2-ID versions typically expose a _Dimmable Light_ on endpoint 0x0A and an _Extended Color Light_, or _Color Light_ on endpoint 0x0B.  The _Dimmable Light_ endpoint would not expose _Color Control_.  The 1-ID versions only expose an _Extended Color Light_ or _Color Light_ endpoint on 0x0B.

michi1g, 2020-08-19:
Yeah, it's a dimmer with a WW and CW Channel. (https://de.gearbest.com/diy-parts-and-tools/pp_009893773954.html?lkid=20248294)
With both I mean WW an CW channel.
On my devices page I got one light identified as GL-C-009 with manufacturer Gledopto

ebaauw, 2020-08-19:
The link shows a dimmer with only one white channel labelled `W` (with two connections, similar to how the `V+` has two connections).

closed 2020-09-19 by stale[bot]