dresden-elektronik/deconz-rest-plugin #3167: VIMAR ZIGBEE devices 19592, 19593, 19594

samuele2723, 2020-08-19:
Hello, i am asking the support for VIMAR connected devices radiofrequency.

There are 3 kind of devices manufactured by VIMAR:

Deviator 2 way - XX592

https://www.vimar.com/en/int/catalog/product/index/code/14592.0

Shutter command - XX594

https://www.vimar.com/en/int/catalog/product/index/code/19594.0

Actuator controller - - XX593

https://www.vimar.com/en/int/catalog/product/index/code/14593

XX can be 14, 19 or 20 depending on the series but inside they are all the same device.

They support zigbee 3.0 And third party hubs

  Required screenshots:
  - Endpoints and clusters of the node
  - Node Info panel
  - Basic Cluster attributes in the Cluster Info panel.

  In the Cluster Info panel press "read" button to retreive the values. Please note that at least "Manufacturer Name" and "Model Identifier" must be populated with data (therefore, must not be empty), otherwise that information will not be usable. For battery powered devices, after pressing read it is required to wake-up the device by pressing a button or any other means of interaction.
-->

Mimiix, 2020-08-19:
Hi @samuele2723!

 I am closing this issue, as all screenshots are missing and there is 0 information provided. Please stick to the template and provide screenshots of the required clusters.

Please open a new device request with the information.

Thanks!

closed 2020-08-19 by Mimiix

testorp, 2020-12-08:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

testorp, 2020-12-08:
hope you can help, because when i pair the switch it appears as a light in Phoscon

[image]

Smanar, 2020-12-08:
Wired switches are light
Battery switches are sensor.

testorp, 2020-12-08:
@Smanar so what are switches?
Where can I find this information pls

Thx
Paolo