dresden-elektronik/deconz-rest-plugin #3168: Carbon Monoxide Detector

panenka, 2020-08-19:
- Product name: Carbon Monoxide Detector SCA01ZB
- Manufacturer: feibit
- Model identifier: FNB56-COS06FB1.7.
- Device type : Please remove all unrelated device types. 
  - Sensor
  - Siren

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

panenka, 2020-09-13:
When this device will be added to the database, please?

Mimiix, 2020-09-14:
@panenka At next relase probably. the 15th.

closed 2020-09-15 by manup

panenka, 2020-09-15:
Not understand. Where is problem? New version has been generated, but this device has not been added to V2_05_81_stable.

Mimiix, 2020-09-15:
@panenka Nope, but will be in the beta of .82. Release should be later today ;)

jordibsk10, 2020-10-04:
Has been released already? If not, is there any issue?
If it has been release, thank you very much and don't forget to change the status to "Merged" ;)

panenka, 2020-10-04:
Still waiting for release, not only final version, but for release new version for HA addon. After that tests could begin.

SwoopX, 2020-10-04:
@jordibsk10 
[image]