dresden-elektronik/deconz-rest-plugin #3169: Gas Leak Detector & Repeater

panenka, 2020-08-19:
- Product name: Natural gas coal gas LPG combustible gas leakage sensor detecor SGA01ZB and Repeater
- Manufacturer: feibit
- Model identifier: FNB56-GAS05FB1.4
- Device type :
  - Sensor
  - Siren
  - Other: Repeater

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

panenka, 2020-09-13:
When this device will be added to the database, please?

Mimiix, 2020-09-14:
At next release: the 15th

closed 2020-09-15 by manup

panenka, 2020-09-15:
Not understand. Where is problem? New version has been generated, but this device has not been added to V2_05_81_stable.

Mimiix, 2020-09-15:
@panenka Nope, but will be in the beta of .82. Release should be later today ;)

panenka, 2020-09-15:
I have stable version. When it will be?

Mimiix, 2020-09-15:
the stable? i expect 1st week of next month. Or at least 15th of next month.

panenka, 2020-09-15:
It is very long time in development cycle. Two month after which I can make first check if it is working or not.

Mimiix, 2020-09-15:
Nope. You can test it in Beta. We need to make sure it works properly first.

15th is the Beta. > That beta goes into stable at the 1st (around that). 

You were just after the last beta. You are free to compile yourself :)

panenka, 2020-09-15:
I don't know how to move to beta and back to stable. I don't know how safe it is. I would like to be end user and not developer in this product. Beta is for testing only. I could help to add devices, I have no problem with it, but I don't want to use beta product.
I have these devices from Aliexpress and I and seller both are waiting a month if the device is working or not. I must stop this waiting and must accept it, because this is not problem of the seller but of deconz and of long development cycle. You said, that support will be in new version at 15th a few days ago, but you didn't said that this will be next month, not current, a more than month later.

SwoopX, 2020-09-15:
I'll give you some further insights later.

panenka, 2020-09-15:
I understand that this is my decision and my freedom to use Home Assistant and deconz and not to use Tuya. This is not something bad against you or against deconz, I would like to be constructive to make deconz better.

SwoopX, 2020-09-16:
@panenka 

> I don't know how to move to beta and back to stable.

[On Phoscon webpage](https://phoscon.de/en/conbee2/install#raspbian), It is described how to choose/switch the branches. Anyway, before you do anything, it is highly recommended to make a backup in Phoscon first so you're prepared should anything go wrong. It might also be a good idea to first uninstall a previous version (google helps in doubt). Please take note that this applies to "bare" operating systems only. So in case you're using HA, you're either dependent on their releases or on docker.

> I would like to be end user and not developer in this product. Beta is for testing only. I could help to add devices, I have no problem with it, but I don't want to use beta product.

I'm afraid that's to a certain extend part of the game if you want to have a new device supported. We as devs do this firstly in our leisure time and secondly, don't have the devices at hand in 90% of the cases. That being said, we depend on feedback and testing to iron out the quirks. Another possibility is of course finding a dev and donating the device, certainly the best way to gain best possible device support.
Chosing a "premium" vendor is mostly the route which people are trying to evade, as this is the most costly option. However, the vendors often provide very detailed technical documentation, which in turn eliminates a considerable amount of trial and error as well as assumptions. On the other hand, you pretty much know exactly about what you're dealing with.

> You said, that support will be in new version at 15th a few days ago, but you didn't said that this will be next month, not current, a more than month later.

Support for the devices you raised requests for have been included in the beta version released yesterday as indicated.

Lastly, it always depends on the way and preference of oneself how to run deconz. If you choose to use HA, which can be the "most complex" constellation (because in worst case, you're dependent on the HA folks to release an updated version), then there's not much we can do about it. Once a PR for a device has been release for merge, you're free to instantly use and test it. If you don't want that, it's your decision, simple as that.

In the end, we're happy if you wanna be constructive and make deconz better, help us to help you and be a part of it. Remember, we're kinda have the pain, but you got the gain.