dresden-elektronik/deconz-rest-plugin #3170: Docker - Deconz fail to connect

sirsiegfrieds, 2020-08-19:
## Describe the question or issue you are having

Hey there, i'm having problem using deconz on a docker.
I have followed the information for adding this at my "compose", but it look's like have problem in logs.
And i can't had nothing anymore.

I'm a docker beginner, and i'm trying my best for understand it :).
Sorry for my bad english, i'am a french guy
## Screenshots

## Environment

 - Host system: (Vm linux runing docker)
 - Running method:  Marthoc Docker container
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device:  ConBee II 
 - Do you use an USB extension cable: no -- only relevant for ConBee II

### deCONZ Logs

[marthoc/deconz] Starting deCONZ...
[marthoc/deconz] Current deCONZ version: 2.05.80
[marthoc/deconz] Web UI port: 80
[marthoc/deconz] Websockets port: 443
[marthoc/deconz] VNC Disabled
libpng warning: iCCP: known incorrect sRGB profile
16:13:41:176 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
16:13:41:183 CTRL. 3.16.216:13:41:205 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
16:13:41:250 parent process /bin/sh
16:13:41:251 gw run mode: docker
16:13:41:251 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
16:13:41:251 DB sqlite version 3.16.2
16:13:41:251 DB PRAGMA page_count: 30
16:13:41:251 DB PRAGMA page_size: 4096
16:13:41:251 DB PRAGMA freelist_count: 0
16:13:41:251 DB file size 122880 bytes, free pages 0
16:13:41:251 DB PRAGMA user_version: 6
16:13:41:251 DB cleanup
16:13:41:252 DB create temporary views
16:13:41:254 started websocket server at port 443
16:13:41:255 GW firmware update not supported on x86 linux headless
16:13:41:255 found node plugin: libde_rest_plugin.so - REST API Plugin
16:13:41:256 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
16:13:41:260 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
16:13:42:383 New websocket 192.168.xxxx:49197 (state: 3)
16:13:44:494 New websocket 192.168.xx.xx:53730 (state: 3)
16:13:46:886 Announced to internet http://dresden-light.appspot.com/discover
16:13:49:249 Websocket disconnected 192.168.xx.xx:49197 (state: 0)
16:13:50:318 New websocket 192.168.xx.xx:49208 (state: 3)
16:13:51:135 device disconnected reason: 4, index: 0
16:13:52:113 wait reconnect 15 seconds
16:13:53:113 wait reconnect 14 seconds
16:13:54:114 wait reconnect 13 seconds
16:13:55:114 wait reconnect 12 seconds
16:13:56:113 wait reconnect 11 seconds
16:13:57:114 wait reconnect 10 seconds
16:13:58:114 wait reconnect 9 seconds
16:13:59:113 wait reconnect 8 seconds
16:14:00:114 wait reconnect 7 seconds
16:14:01:114 wait reconnect 6 seconds
16:14:01:114 start reconnect to network
16:14:02:113 wait reconnect 5 seconds
16:14:03:113 wait reconnect 4 seconds
16:14:04:114 wait reconnect 3 seconds
16:14:05:113 wait reconnect 2 seconds
16:14:06:114 failed to reconnect to network try=1
16:14:06:114 wait reconnect 1 seconds
16:14:06:135 device disconnected reason: 4, index: 0
16:14:07:114 wait reconnect 15 seconds
16:14:08:114 wait reconnect 14 seconds
16:14:09:113 wait reconnect 13 seconds
16:14:10:114 wait reconnect 12 seconds
16:14:11:113 failed to reconnect to network try=2
16:14:11:113 wait reconnect 11 seconds
16:14:12:114 wait reconnect 10 seconds
16:14:13:113 wait reconnect 9 seconds
16:14:14:114 wait reconnect 8 seconds
16:14:15:114 wait reconnect 7 seconds
16:14:16:114 failed to reconnect to network try=3
16:14:16:114 wait reconnect 6 seconds
16:14:17:113 wait reconnect 5 seconds
16:14:18:113 wait reconnect 4 seconds
16:14:19:114 wait reconnect 3 seconds
16:14:20:113 wait reconnect 2 seconds
16:14:21:113 failed to reconnect to network try=4
16:14:21:113 wait reconnect 1 seconds
16:14:21:140 device disconnected reason: 4, index: 0
16:14:22:113 wait reconnect 15 seconds
16:14:23:113 wait reconnect 14 seconds
16:14:24:113 wait reconnect 13 seconds
16:14:25:113 wait reconnect 12 seconds
16:14:26:113 failed to reconnect to network try=5
16:14:26:113 wait reconnect 11 seconds
16:14:27:114 wait reconnect 10 seconds
16:14:28:113 wait reconnect 9 seconds
16:14:29:113 wait reconnect 8 seconds
16:14:30:113 wait reconnect 7 seconds
16:14:31:113 failed to reconnect to network try=6
16:14:31:113 wait reconnect 6 seconds
16:14:32:114 wait reconnect 5 seconds
16:14:33:114 wait reconnect 4 seconds
16:14:34:113 wait reconnect 3 seconds
16:14:35:114 wait reconnect 2 seconds
16:14:36:114 failed to reconnect to network try=7
16:14:36:114 wait reconnect 1 seconds
16:14:36:130 device disconnected reason: 4, index: 0
16:14:37:113 wait reconnect 15 seconds
16:14:38:114 wait reconnect 14 seconds
16:14:39:114 wait reconnect 13 seconds
16:14:40:114 wait reconnect 12 seconds
16:14:41:113 failed to reconnect to network try=8
16:14:41:113 wait reconnect 11 seconds
16:14:42:114 wait reconnect 10 seconds
16:14:43:113 wait reconnect 9 seconds
16:14:44:113 wait reconnect 8 seconds
16:14:45:113 wait reconnect 7 seconds
16:14:46:114 failed to reconnect to network try=9
16:14:46:114 wait reconnect 6 seconds
16:14:47:114 wait reconnect 5 seconds
16:14:48:113 wait reconnect 4 seconds
16:14:49:113 wait reconnect 3 seconds
16:14:50:114 wait reconnect 2 seconds
16:14:51:113 failed to reconnect to network try=10

## Additional context

I can accès to My Home Assistant well at : 
myip:8123
and to phoscon webui at : 
myip ( without port) how it's possible ?

peterguy04, 2020-08-19:
try adding 

```- DECONZ_DEVICE=/dev/ttyUSB1```  
 to your compose under environment, changing the path to your conbee

Smanar, 2020-08-19:
If the problem is USB disconnection loop, you need to update manualy the firwmare > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2493

sirsiegfrieds, 2020-08-19:
Thanks for help

All my device are ACM*

/dev/ttyACM1 
/dev/ttyACM0 
/dev/ttyACM2

But i try this : 
`  deconz:
    image: marthoc/deconz
    container_name: deconz
    network_mode: host
    restart: always
    volumes:
      - /opt/deconz:/root/.local/share/dresden-elektronik/deCONZ
    devices:
      - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2130375-if00
    environment:
      - DECONZ_WEB_PORT=80
      - DECONZ_WS_PORT=443
      - DEBUG_INFO=1
      - DEBUG_APS=0
      - DEBUG_ZCL=0
      - DEBUG_ZDP=0
      - DEBUG_OTAU=0
      - DECONZ_DEVICE=/dev/ttyUSB1`

But all my device are ACM*

Thanks for help : 
 Smanar, I've done : GCFFlasher_internal -l 
and it return commande introuvable

peterguy04, 2020-08-19:
I have a feeling you might not be able to use /serial/by-id/ path for -device...dont quote me on it but try tty* path on that aswell

sirsiegfrieds, 2020-08-19:
I have cp the wrong thing sorry for device i have this : ( that was a another configuration try )

devices:
- /dev/ttyACM1

i have a question does i have to create the folder manually ? 
- /opt/deconz:/root/.local/share/dresden-elektronik/deCONZ

sirsiegfrieds, 2020-08-19:
Aw nice, i have found my error.

Forgot to specify this /dev/ttyACM1 in the home assitant service ...
My bad :)

Smanar, 2020-08-19:
@peterguy04 it s normal he have "Command not found" when he use "GCFFlasher_internal -l" on docker ?

peterguy04, 2020-08-19:
@Smanar if you run it on host OS you get not found ye. But if i open a console in my deconz container it find GCFFlasher but has version : Glibc 2.28 not found

sirsiegfrieds, 2020-08-19:
It's normal to have the conbee on ACM1 and ACM2 ?

[ 6994.366508] cdc_acm 3-2:1.0: ttyACM2: USB ACM device
[12570.638283] usb 3-2: USB disconnect, device number 9
[12570.638749] usb 4-1: USB disconnect, device number 10
[12572.044378] usb 4-2: reset full-speed USB device number 6 using uhci_hcd
[12572.384343] usb 3-2: new full-speed USB device number 10 using uhci_hcd
[12572.843035] usb 3-2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[12572.843037] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[12572.843038] usb 3-2: Product: ConBee II
[12572.843039] usb 3-2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[12572.843039] usb 3-2: SerialNumber: DE2130375
[12572.846568] cdc_acm 3-2:1.0: ttyACM1: USB ACM device
[12573.308325] usb 4-1: new full-speed USB device number 11 using uhci_hcd
[12573.768401] usb 4-1: New USB device found, idVendor=0658, idProduct=0200, bcdDevice= 0.00
[12573.768403] usb 4-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[12573.774636] cdc_acm 4-1:1.0: ttyACM2: USB ACM device

sirsiegfrieds, 2020-08-19:
All work fine :).

Mimiix, 2020-08-20:
@sirsiegfrieds Nice!

Can we close this?

sirsiegfrieds, 2020-08-20:
yes, 
i don't understand why the symlink are not allowed in deconz,
but work fine with /dev/ttyACM1
Any alternative for this ?

Thanks for help :)

closed 2020-08-20 by sirsiegfrieds