dresden-elektronik/deconz-rest-plugin #3187: VIMAR window covering control IoT device 14594.0 (or 19594.0 or 20594.0)

samuele2723, 2020-08-23:
## Device

- Product name: Vimar - 14594.0 (or 19594.0 or 20594.0) - Rolling shutter IoT connected mechanism
- Manufacturer: Vimar
- Model identifier: Window_Covering_v1.0
- Device type : 
  - Other: Window Shutter control
## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]

NOTE: The device is not updating automatically the states changes when interacted physically. Means i OPEN the shutter and Deconz is not automatically notified and updated, i need to manually press "read" to generate the state update.

NOTE2: The device seems coming by factory with kind of calibration in order to know how long time takes to move for 1% of shutter opened. Nevertheless further calibration should be done when connected to the motor

SwoopX, 2020-08-28:
With regard to reporting: have you already tried fully open -> fully closed -> fully open? Also, that must be done on each power cycle (in case that applies for you).

samuele2723, 2020-08-31:
Hello SwoopX, 

I have a update on this, i bought a 30 eur motor on Amazon, connected and now i get payload reporting and state update on lift percentage. I am not sure why this didn’t happen before, but now seems working properly (there are 4-5 sec delay after end of action before payload generation).

Now, only big question remaining is about the 0-100% calibration. How is possible to set that? From my understanding should be possible to change time and speed of the motor connected. Are those parameters configurable in deconz?

closed 2020-09-10 by manup