dresden-elektronik/deconz-rest-plugin #3190: Smart things Multipurpose sensor - incorrect temp being shown

JamieP83, 2020-08-24:
## Describe the bug
My samsung smartthings multi purpose sensor, keeps reporting a -46 temperature. its also not showing a battery level?
i have deleted the device and readded.
i have reset the device and readded
no change

## Steps to reproduce the behavior
just added the sensor and it keeps showing the incorrect temp.
i have 10+ other devices, all of those are ok

## Expected behavior
correct temp

## Screenshots
[image]

[image]

## Environment

 - Host system: Raspberry Pi4 
 - Running method:Home Assistent deCONZ Add-on /
 - Firmware version: 26580700
 - deCONZ version: 2.0.5.79 / 5/22/2020
 - Device:  ConBee II 
 - Do you use an USB extension cable: yes 

[image]

## Additional context

```
    "113": {
        "config": {
            "battery": null,
            "on": true,
            "pending": [],
            "reachable": true
        },
        "ep": 1,
        "etag": "c8048b9c7faae5a2b3f9eed51706e6ca",
        "lastseen": "2020-08-24T18:10:40.680",
        "manufacturername": "Samjin",
        "modelid": "multi",
        "name": "Dining Room Door ST DZ",
        "state": {
            "lastupdated": "2020-08-24T18:10:40.680",
            "open": false
        },
        "type": "ZHAOpenClose",
        "uniqueid": "28:6d:97:00:01:04:86:73-01-0500"
```

JamieP83, 2020-08-25:
i just replaced the battery and thats fixed the unknown battery issue, but temp is still off

[image]

Mimiix, 2020-08-26:
@ebaauw As far as i can find, you integrated [this one](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2769) Are you able to give some info on this?

JamieP83, 2020-10-04:
> @ebaauw As far as i can find, you integrated [this one](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2769) Are you able to give some info on this?

sorry! i missed this one
yes, thats the right model! sorry again

JamieP83, 2020-11-01:
Hey
This is still an issue

Smanar, 2020-11-02:
If 10 are working and not this one, I don't think the problem was from code.
You can see a difference beetwen the working and the unworking one in deconz ?

Can you take a look in the cluster 0402 and sho the attribute list (need probable press the "read" button and making the device wake up) ?

JamieP83, 2020-11-18:
sorry, how do i do that?
i cant see a way to show the info for 0402

[image]

JamieP83, 2020-11-18:
sorrry, think i have it now:

This is the bad result from the dining room door sensor

[image]
[image]

this is a result from a "working" device
[image]

JamieP83, 2020-11-18:
sorry, i didnt mean/want to close this either

JamieP83, 2020-11-18:
it seems tolerance is not set for the device thats playing up?

[image]

closed 2020-11-18 by JamieP83

SwoopX, 2020-11-18:
Temperature measurement of the device in question seems to report erroneous data.

JamieP83, 2020-11-18:
agreed - but how to fix that?

SwoopX, 2020-11-18:
Erm, return the faulty and get a new one?

JamieP83, 2020-11-18:
> Erm, return the faulty and get a new one?

sorry, with you now! will order a new one and bin this then