dresden-elektronik/deconz-rest-plugin #3193: Raspbee cannot join network

ma-ca, 2020-08-25:
The Raspbee was up and running for more than a year. No recent update or reboot. The Raspbee suddenly lost the network connection and since then I cannot join the network anymore. I already did reboot the Raspberry without success.

```
firmware version: 0x26330500
deCONZ version: 2.05.69
```

It turns out that I cannot change the network settings anymore. When I change the Security Mode to _Standard - Network Key from TC_ (which I believe is right)

[image]

and press Save, then the setting switches back to _No Security_. Also when I deselect the channels 24, 25, 26 or change the Network Key, then the new setting is reverted again after pressing Save, and the channels 24, 25, 26 are selected again.

[image]

(Yes, I did stop deCONZ before the next steps)

```
$ sudo GCFFlasher -l
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10
RaspBee: /dev/ttyAMA0 (default)

       1  FTDI device found

 device | vendor | product | serial  | description
--------|--------|---------|---------|----------------------
   0    | 0x0403 | 0x6001  | AI03D972 | FT232R USB UART

$ sudo GCFFlasher -r
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10
no device specified use RaspBee (/dev/ttyAMA0)
reset target SUCCESS

$ sudo GCFFlasher -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCF
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10
no device specified use RaspBee (/dev/ttyAMA0)
using firmware file: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCF
reset target

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01

flashing 125684 bytes: |=============================|
verify: ....
SUCCESS
```

Smanar, 2020-08-25:
To change setting in deconz in some situation you need to rejoin a network (setting are apply at next rejoin), so if you are locked in "not in network" can be a reason.

For me the command result at "sudo GCFFlasher -l" is not good at all.
Can you try first to update your deconz version (the flasher you are using is an old one too)

And try the command with "-t 60" > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

From my memory there is a LED on the raspbee that can give information about the device ?

Mimiix, 2020-08-26:
@ma-ca I am missing template input. 

Nevertheless: I'd recommend updating deconz, as your version is old. Maybe a fresh start would do?

@Smanar  the device seems flashed at the end. I think it might be that the old deconz version can't speak to the new firmware.

SwoopX, 2020-08-26:
@ma-ca very strange behaviour, indeed. For what it's worth, standard settings for security is "high" and you also need to check the predefined PAN ID, which is like the SSID in wifi.

Regardless, probably a case for @manup to look into.

ma-ca, 2020-08-26:
Solved.

### Long story short: 

1. Stop all router devices (power off). 
2. Set predefined PAN ID in network settings
3. Join Network
4. Start all router devices again

Result: All devices were able to connect without the need to resetting and re-pairing.

[image]

### Long story:

Yesterday the Raspbee was completely unable to join the network. The red LED on the Raspbee was blinking fast. I did leave the Rasbee and deCONZ running during the night.

However, this morning the red LED was constantly on (not blinkng anymore) and the Raspbee was _In Network_. But all devices did not respond. Also the PAN ID had changed. Leaving and joining the network did always generate a new PAN ID. Also checking the box _Predefined PAN ID_ and entering the old PAN ID was overwritten again after joining the network.

Finally, the predefined PAN ID did remain unchanged only after powering off all router devices. The Raspbee was able to form a new network with the old (existing) PAN ID. Then all devices were able to join the known network PAN ID.

Important: A backup of the `.local/share/dresden-elektronik/deCONZ` directory. The old PAN ID is stored in the `zll.db` in the `zbconf` table.

Note: The Raspbee red LED is constantly on when connected to a network. The green LED blinks during network activity.

I have no idea why the Raspbee suddenly dropped off the network and was unable to reconnect again. The old deCONZ software and firmware did run very stable for almost a year (since September 2019).

Mimiix, 2020-08-26:
Great to see it solved. Would be hard to say without any logs . If you encounter it again, let us know.

Smanar, 2020-08-26:
Ha nice, but I have a question too, what is PAN ID ?

MattWestb, 2020-08-26:
Zigbee 1 have only PAN-ID (16 bit) and Zigbee 2 (pro) have PAN-ID and Extended PAN-ID (64 bit) and both suld being unique in the radio surroundings or its being network conflict and jamming.

Smanar, 2020-08-26:
nice, thx ^^

closed 2020-09-29 by stale[bot]