dresden-elektronik/deconz-rest-plugin #3196: Conbee II randomly disconnects after update

tyr84, 2020-08-25:
## Describe the bug

  Controller randomly leaves the network. This may take one of two forms:
1. VNC app shows that the controller is not in the network (see pic):

 When i press join, it joins back in, and all is back to normal.

2. VNC app tells me that Conbee II is disconnected. Reconnecting through the VNC app doesn't work. When i restart the hass.io deconz addon it reconnects and all is normal.

I have not been able to figure out if this behaviour is triggered by something. deconz-addon log shows nothing special. The intervall between theese events can be everything from minutes to days (random).

What I know that this started happening roughly a month ago after an update, but I don't have a record of what version I ran before the update

## Screenshots

  [image]

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Home Assistant deCONZ Add-on
 - Firmware version: 26580700
 - deCONZ version: 2.05.79
 - Device: ConBee II 
 - I've tried using extension cable, and also connecting it to a 2.0 usb hub, not noticable difference

### deCONZ Logs

If you can provide me with the relevant flags, I can enable logging and see if I can capture something

## Additional context

Mimiix, 2020-08-26:
Hi @tyr84 !

I've noticed this before in which the addon stopped [working](https://github.com/home-assistant/hassio-addons/issues/1546) . There is no more logging to enable: As far as i know all logging is already there on the addon logs. 

As you are using the HA add-on, we can't be much of help here as we don't maintain that. If you issue isn't the same as the one i mentioned, i suggest asking your question there. 

Thanks!

Mimiix, 2020-08-26:
In addition: I just got headsup from @Smanar that it could be something else.  

Are you able to provide logging of the addon output (on the times this occurs?)

Smanar, 2020-08-26:
There is a way to make a test only with deconz part, without using the HA plugin ?

Or can you make a try on another machine to be sure the problem is not hardware ?

tyr84, 2020-08-26:
I support that this is not the same as [#1546](https://github.com/home-assistant/hassio-addons/issues/1546) since the addon does not seem affected. It keeps running with no significant output to the logs. 

I don't have an easily available setup to isolate deconz, but I will investigate the possibility of pulling the complete addon log from docker (got to be stored somewhere). 

Which debug flags are of interest for this kind of issue? and what level?

Smanar, 2020-08-26:
Good question ^^.

At minimum --dbg-error=2 --dbg-info=2
--dbg-http=0 , http totaly useless for me if the problem is not from HA
--dbg-zdp=0 --dbg-zcl=0, i dont think they are usefull
 --db-aps=1, there is some aps error usefull and idk if they are displayed with dbg-error

tyr84, 2020-08-26:
debug flags worked with the hass addon :-) So now I just need to wait for the next bug

tyr84, 2020-08-31:
So, I finally managed to log a reasonable interval (10min) where it happens, it is however 12k lines of log, and I'm unsure what I'm looking for. The last lines of the log describes the situation where I detect failed response, and the first lines are some seconds before Home Assistant last detected a light beeing toggled.

[deconz-bug(truncated).log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5152764/deconz-bug.truncated.log)

Smanar, 2020-09-01:
You have some timeout, but I don't think it s important.

> 2020-08-31T19:37:19.289638000Z 21:37:19:202 APS-DATA.request id: 142, addrmode: 0x02, addr: 0x8f14, profile: 0x0104, cluster: 0x0300, ep: 0x01 -> 0x01 queue: 92 len: 5 tx.options 0x00
> 2020-08-31T19:37:22.591788000Z 21:37:20:910 APS-DATA.confirm id: 142, status: 0xA3 ILLEGAL_REQUEST
> 2020-08-31T19:37:19.197851000Z 21:37:19:068 APS-DATA.request id: 132, addrmode: 0x02, addr: 0x83a8, profile: 0x0104, cluster: 0x0300, ep: 0x01 -> 0x01 queue: 91 len: 5 tx.options 0x00
> 2020-08-31T19:37:22.591615000Z 21:37:20:906 APS-DATA.confirm id: 132, status: 0xA3 ILLEGAL_REQUEST

> 2020-08-31T19:37:17.207825000Z 21:37:17:145 APS-DATA.request id: 153, addrmode: 0x02, addr: 0x8f14, profile: 0x0104, cluster: 0x0300, ep: 0x01 -> 0x01 queue: 55 len: 5 tx.options 0x00
> 2020-08-31T19:37:17.372903000Z 21:37:17:299 APS-DATA.request id: 153, status: BUSY (counter: 11)
> 2020-08-31T19:37:17.373029000Z 21:37:17:299 APS-DATA.request id: 153, set state: 0x05
> 2020-08-31T19:37:17.373175000Z 21:37:17:299 emit artificial APSDE-DATA.confirm id: 153
> 2020-08-31T19:37:17.373288000Z 21:37:17:299 APS-DATA.request id: 153, failed-status: APP_BUSY (0x02)
> 2020-08-31T19:37:22.593337000Z 21:37:21:174 aps request id: 153 failed, erase from queue

> 2020-08-31T19:39:44.020118000Z 21:39:41:980 API error 901, /lights/41/state/on, Internal error, 951
> 2020-08-31T19:39:44.020253000Z 21:39:41:990 API error 901, /lights/44/state/on, Internal error, 951

951 mean ERR_BRIDGE_BUSY
It happen when you are (trying to) sending too much command in the same time limit.

Can you try to disable some scene/group/automation for some time to compare ? Perhaps you have something that "spam the network" with command.

SwoopX, 2020-09-01:
There's even more in the logs. More than 300 entries mentioning fails, at the beginning some `0xD0` and quite a number of send delays. My first impression was also that too much is happening at the same time.

How large is your network (lights, sensors, etc.)?

tyr84, 2020-09-01:
79 nodes in total
51 lights, 22 sensors + switches 

I've now turned of [flux](https://www.home-assistant.io/integrations/flux/) which was sending commands to adjust colortemp every 10 minutes. The answer to whether this works will possibly take a couple of days.

closed 2020-10-25 by stale[bot]