dresden-elektronik/deconz-rest-plugin #3197: bootloop of the device after upgrade of device with deconz plugin of jeedom

cedricmercy, 2020-08-25:
Hello , 
I'm a new jeedom user and jeedom uses deconz plugin to interface with the conbee 2 usb stick.
My jeedom Virtual Machine is under a VirtualBox and the plugin proposed me to upgrade the firmware of the stick from version 
0x264a0700 to the last version which is 0x26580700. The upgrade did apparently well work, no error message, but since I've rebooted my platform, the stick is in a bootloop in /var/log/kern.log:

Aug 25 18:56:12 esx1 kernel: [102745.426226] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Aug 25 18:56:12 esx1 kernel: [102745.426229] usb 3-1: Product: ConBee II
Aug 25 18:56:12 esx1 kernel: [102745.426232] usb 3-1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Aug 25 18:56:12 esx1 kernel: [102745.426235] usb 3-1: SerialNumber: DE2219637
Aug 25 18:56:12 esx1 kernel: [102745.427784] cdc_acm 3-1:1.0: ttyACM0: USB ACM device
Aug 25 18:56:15 esx1 kernel: [102748.811938] usb 3-1: USB disconnect, device number 108
Aug 25 18:56:16 esx1 kernel: [102749.200461] aufs aufs_fill_super:918:mount[635500]: no arg
Aug 25 18:56:16 esx1 kernel: [102749.210476] overlayfs: missing 'lowerdir'
Aug 25 18:56:31 esx1 kernel: [102764.454611] aufs aufs_fill_super:918:mount[635572]: no arg
Aug 25 18:56:31 esx1 kernel: [102764.465238] overlayfs: missing 'lowerdir'
Aug 25 18:56:32 esx1 kernel: [102765.204260] usb 3-1: new full-speed USB device number 109 using xhci_hcd
Aug 25 18:56:32 esx1 kernel: [102765.358412] usb 3-1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Aug 25 18:56:32 esx1 kernel: [102765.358417] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Aug 25 18:56:32 esx1 kernel: [102765.358420] usb 3-1: Product: ConBee II
Aug 25 18:56:32 esx1 kernel: [102765.358422] usb 3-1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Aug 25 18:56:32 esx1 kernel: [102765.358424] usb 3-1: SerialNumber: DE2219637
Aug 25 18:56:32 esx1 kernel: [102765.360021] cdc_acm 3-1:1.0: ttyACM0: USB ACM device
Aug 25 18:56:35 esx1 kernel: [102768.748253] usb 3-1: USB disconnect, device number 109
Aug 25 18:56:46 esx1 kernel: [102779.777864] aufs aufs_fill_super:918:mount[635675]: no arg
Aug 25 18:56:46 esx1 kernel: [102779.789120] overlayfs: missing 'lowerdir'
Aug 25 18:57:00 esx1 kernel: [102793.135929] usb 3-1: new full-speed USB device number 110 using xhci_hcd
Aug 25 18:57:00 esx1 kernel: [102793.293532] usb 3-1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
so I tried to see if I could downgrade the firmware but I don't win to do it with the line command to flash the firmware :

# GCFFlasher_internal -d 0 /dev/ttyACM0 -t 1000 -f deCONZ_ConBeeII_0x264a0700.bin.GCF

sometimes the command to display the device appears, sometimes not :
root@esx1:/home/cedric# GCFFlasher_internal -l
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
root@esx1:/home/cedric# GCFFlasher_internal -l
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  | DE2219637  | ConBee II 
root@esx1:/home/cedric# GCFFlasher_internal -l
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
root@esx1:/home/cedric#

I've well stopped the ModemManager , deconz and deconz-gui on my host ( of course I don't work on the VirtualMachine to try to recover the stick ) , but I'm not able to flash a new time the device to fix the issue.
Do you have an Idea of what I could do to fix the stick ?

Many thanks if you could help me.

Cédric.

Mimiix, 2020-08-26:
Hi @cedricmercy 

First off, i'd like to ask you to stick to the format. You did provide some information, but not in the way we asked.

To answer your question: [Please follow this guide](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually)

I'd recommend using a windows machine to fix it :)

cedricmercy, 2020-08-26:
Hello Dennis D, 

Many thanks , sorry to didn't follow the format, I'm a little bit in Panic because I'm really affraid I did brick the device. :-( 

I followed the guide you mentioned on my Linux host. 

I tried to fix it on a windows system too, but the result is the same on my windows 7 Computer. 
I'm not able to see the device sometimes with the list command of the gcfflasher tool 

I've tried to change the parameter -t 60 to -t 1000 but no change. 

My operating system is the last ubuntu 20.04, is it possible that another service ( other than ModemManager could avoid to flash the device ? ) 

thank you. 

Cédric. 

----- Mail original -----

Mimiix, 2020-08-26:
Hi!

On the linux host, is that a ESXI (as i can tell?)

cedricmercy, 2020-08-26:
No, I use VirtualBox version 6.1.6, and I configured the key to the Virtual Machine with adding a filter with the name of the device in the configuration of the Virtual Machine in the USB parameter Box.

Mimiix, 2020-08-26:
To be clear: Are you using Virtual box to **update/flash the conbee?**.

cedricmercy, 2020-08-26:
> To be clear: Are you using the OS within Virtual box to update/flash the conbee? .

The first time , yes, because it was in Jeedom with the plugin in Web Interface, but now, I try to fix it directly from the main host , not the virtual Machine. 
I can't connect the key to the Virtual Machine anymore because VirtualBox hasn't enough time to update the list of Devices during the 2 seconds between the connect and disconnection of the key. 

Thank you for your time 

Cédric 

----- Mail original -----

Mimiix, 2020-08-26:
@cedricmercy 

The host, is that ubuntu? 

The thing that happened here is that the firmware was corrupted due to Jeedom.

To fix that, you need to follow the guide i mentioned earlier. Please make sure to use Sudo and make sure to have deconz STOPPED on the host aswell. Otherwise, the stick is occupied.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Smanar, 2020-08-26:
@cedricmercy you miss some parameter in your command, for exemple "-t 60" to increase timeout.
And like have said Mimixx  make the procedure on REAL machine.

cedricmercy, 2020-08-26:
Hello all,

I've found my mistake by using the debug mode of the tool, I made a confusion with the device name and the serial name, so I added a 0 before the device name, now I've win to flash a new time the firmware and it works :-)

/usr/bin/GCFFlasher_internal.bin -d 0 /dev/ttyACM0 -t 60 -f deCONZ_ConBeeII_0x264a0700.bin.GCF -x 3

16:24:10:837 using firmware file: deCONZ_ConBeeII_0x264a0700.bin.GCF 
16:24:10:872 device /dev/ttyACM0 not found
16:24:10:872 retry, elapsed 5 seconds
16:24:10:977 parse commands: /usr/bin/GCFFlasher_internal.bin -d /dev/ttyACM0 -t 60 -f deCONZ_ConBeeII_0x264a0700.bin.GCF -x 3

16:24:10:977 using firmware file: deCONZ_ConBeeII_0x264a0700.bin.GCF 
16:24:11:010 device /dev/ttyACM0 not found
16:24:11:010 retry, elapsed 5 seconds
16:24:11:109 parse commands: /usr/bin/GCFFlasher_internal.bin -d /dev/ttyACM0 -t 60 -f deCONZ_ConBeeII_0x264a0700.bin.GCF -x 3

16:24:11:109 using firmware file: deCONZ_ConBeeII_0x264a0700.bin.GCF 
16:24:11:134 ls dev: /dev/ttyACM0 (0x1CF1/0x0030) sn: DE2219637
Reboot device /dev/ttyACM0 (ConBee II)
16:24:11:140 query bootloader v1 ID after 5 ms
16:24:11:151 RX 60 bytes ASCII 
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
, 08:48:53
 after 15 ms
16:24:11:151 bootloader start after 16 ms
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
16:24:11:152 GCF_ResetDeviceDone
16:24:11:153 bootloader v1 update firmware
flashing 158985 bytes: |==============================|
verify: .
SUCCESS
Wait 10 seconds until application starts
root@esx1:/home/cedric# vi /var/log/kern.log
Many thanks for your help, and sorry that I didn't feel correctly the template .

Regards.
Cédric.

closed 2020-08-26 by cedricmercy

Mimiix, 2020-08-26:
@cedricmercy Happy to hear that!!

Have a goodnight :)