dresden-elektronik/deconz-rest-plugin #3198: devices unavailable in HA but available in phoscon web

Thoky81, 2020-08-25:
I am switching from Raspbee on Rpi to Conbee2. I have setup docker image from marthoc on Synology with ports XXXX:80 and XXXX:443. Everything is ok and I can connect to phoscon web, VNC and even add Deconz integration in Hassio. All devices are online and working on Phoscon web. The problem is that all my devices are in "grey" Unavailable state in HA. If anybody have an idea I will appreciate it. Thanks

Mimiix, 2020-08-26:
Hi @Thoky81 , 

The only thing i can recommend is checking the logs and see from there. However, we can't help with anything else as we don't maintain the HA Integration. 

You should post your question here: https://github.com/home-assistant/core/issues/

closed 2020-08-26 by Mimiix

SwoopX, 2020-08-26:
To add to this: we can only influence the deCONZ REST API and apparently, it is working ok as you described all is good in Phoscon. Phoscon ist just another REST API client and from a functionality perspective, data is pushed and picked up.

In that sense, it seems HA is not getting that data or misinterpreting, which is not in our hands.

Thoky81, 2020-08-26:
Hi and thank you both for reply. I have post this issue in HA community forum so hopefully somebody will have an idea.
https://community.home-assistant.io/t/devices-unavailable-in-ha-but-available-in-phoscon-web/222085?u=thoky