dresden-elektronik/deconz-rest-plugin #3199: Announcement: Please stick to the issue-formats!

Mimiix, 2020-08-26:
Hi everyone!

Just a quick heads-up here. 

Since we've introduced the issue template, issues get solved way faster than before. 

The last weeks we've noticed a decline in the usage in which users simply decided to ignore the issue-template. **I'd like to ask you to please use them as intended as they make our lifes way easier.** 

Tracking, helping and moderating is something that we do in our own time so please help us to help you! We really do not want to start closing issues when they do not comply to a template, just to save time for others. 

If you have feedback on them, please let us know as we can always improve 😄!

Thank you! 

Kind regards,
Dennis

DIYtechie, 2020-09-26:
Where is the template? A link in the top post here would be helpful :)

bencrazy, 2021-03-02:
Hi, I'd like to submit a potential issue but I'm having a hard time finding said template...

bencrazy, 2021-03-02:
Hi, I'd like to submit a potential issue but I'm having a hard time finding said template. Any guidance appreciated.