dresden-elektronik/deconz-rest-plugin #3200: Switch Editor doesn't show switch

Surfer2010, 2020-08-26:
## Describe the bug

I have a Switch (Osram Lightyfy 4 Button Switch) in the "switches" menu, but no available switch in the switch editor in my Room

## Expected behavior
In my "room" i go to SwitchEditor to add a switch that is known to the system

## Environment
Newest Docker Image (2.05.80 / 14.8.2020)

 - Host system: intel server
 - Running method: Marthoc Docker container on a debian server
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.05.80)
 - Device: ConBee II
 - Do you use an USB extension cable: yes

Mimiix, 2020-08-26:
@Surfer2010 Can you show us the deCONZ gui and map?

How do they show there? Can you provide a screen of the Basic cluster of the device?

Surfer2010, 2020-08-26:
[screenshots_pdf.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5128812/screenshots_pdf.pdf)
[image]

Mimiix, 2020-08-26:
Hi!

No i mean the deCONZ map. What you showed is Phoscon  :) 

You need to use VNC to use that.

Surfer2010, 2020-08-26:
> Hi!
> 
> No i mean the deCONZ map. What you showed is Phoscon :)
> 
> You need to use VNC to use that.
> 
> What you could try, is to re-pair the device to the system and see what happens there.

I cannot connect to VNC using VNC Viewer using the correct port i started the container with 
` -e DECONZ_VNC_MODE=1 -e DECONZ_VNC_PORT=5987 -e DECONZ_VNC_PASSWORD=...`

I did try to re-pair the switch ... no changes

Mimiix, 2020-08-26:
Does the switch work with the rest api?

Surfer2010, 2020-08-26:
I don't know what you mean "work with the rest api" ??? how/where would i check that?
i can ssh into the dockerMarthoc container, i can access the new and old phoscon web interface.

VNC still does not work, i restarted the container using the default port 5900 still no difference. No connection with port added to IP nor without :-(
Hope you i can get help on this

Mimiix, 2020-08-26:
First: I'd suggest reading this guide trough to get a basic understanding of how deCONZ works:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-for-Dummies

On the RestAPI, documentation is found here:  
https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

For the VNC part, check this page: 
https://github.com/marthoc/docker-deconz

The reason i want you to check the rest api, is that the device might not  be added in Phoscon yet. To make sure of that, we need to check it.

If it needs support added, i transfer this issue to Phoscon's repo.

Surfer2010, 2020-08-26:
Ok i discovered the Connection with REST API but the result is this:
`{
        "id": "00212EFFFF050496",
        "internalipaddress": "172.18.0.6",      <----- internal Docker IP ?!
        "macaddress": "00212EFFFF050496",
        "internalport": 8880,
        "name": "MyZigbee",
        "publicipaddress": "83.135.xxx.1x4"         <------ external Internet IP
    }`

because the server has the IP Adress 192.168.xxx.48
the Port 8880 is correct for the container

I started the container with:
"docker run -d --network dockernet --name=deconz --restart=always -e DECONZ_WEB_PORT=8880 -p 8880:8880 -e DECONZ_WS_PORT=8443 -p 8443:8443 -e TZ=Europe/Berlin -e DECONZ_VNC_MODE=1 -e DECONZ_VNC_PORT=5900 -e DECONZ_VNC_PASSWORD=xxxxx -v /etc/localtime:/etc/localtime:ro -v /sharedfolders/DockerData/ConBee:/root/.local/share/dresden-elektronik/deCONZ --device=/dev/ttyACM0 -e DECONZ_DEVICE=/dev/ttyACM0 marthoc/deconz"

--network dockernet   <--- could this cause the problem? i did this because it should talk to the other containers (iobroker/...) not sure if neccessary?

But i can access the poscon page by the adress 192.168.xxx.48:8880

Mimiix, 2020-08-26:
Wait, you are using iobroker?

Does the switch show up there?

Surfer2010, 2020-08-26:
> Wait, you are using iobroker?
> 
> Does the switch show up there?

yes

Surfer2010, 2020-08-26:
Reconnected the Switch

[image]

but still not available at the phoscon switch editor

Mimiix, 2020-08-26:
No, but that mightbe because it's not added in Phoscon.

Surfer2010, 2020-08-26:
> No, but that mightbe because it's not added in Phoscon.

so how do i continue from here?

Surfer2010, 2020-08-26:
API Access with --> /sensors   gives me:
`{
    "1": {
        "config": {
            "configured": true,
            "on": true,
            "sunriseoffset": 30,
            "sunsetoffset": -30
        },
        "etag": "72e2ee061045ad8c2df19a1d05c63011",
        "manufacturername": "Philips",
        "modelid": "PHDL00",
        "name": "Daylight",
        "state": {
            "dark": false,
            "daylight": true,
            "lastupdated": "2020-08-26T13:39:31.713",
            "status": 170,
            "sunrise": "2020-08-26T04:25:00",
            "sunset": "2020-08-26T18:08:51"
        },
        "swversion": "1.0",
        "type": "Daylight",
        "uniqueid": "00:21:2e:ff:ff:05:04:96-01"
    },
    "2": {
        "config": {
            "battery": null,
            "group": "6",
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "3c037e8aeb8420c900cd7a54d8d5c89c",
        "lastseen": "2020-08-26T13:59Z",
        "manufacturername": "OSRAM",
        "mode": 1,
        "modelid": "Switch 4x EU-LIGHTIFY",
        "name": "Lightify Dimmschalter",
        "state": {
            "buttonevent": 1002,
            "lastupdated": "2020-08-26T13:59:07.367"
        },
        "swversion": "e.1.0.0M",
        "type": "ZHASwitch",
        "uniqueid": "00:0d:6f:00:0f:e4:88:a4-01-0006"
    }
}`