dresden-elektronik/deconz-rest-plugin #3201: Raspbee2: unable to connect devices

cosmo84, 2020-08-27:
Hi all,

I bought a new Raspbee2 and want to migrate all devices to it. until now everything was working with a Raspbee1. 
I made a fresh install of the Pi, installed deconz and the latest firmware And followed the manual step by step ( incl. wiringpi Update). So far so good, but I can‘t connect anything. No Xiaomi sensor, no OSRAM lights, no Hue light. Of course all of them are resetted before trying to join. 

I also noticed that in the app the device is recognized as Raspbee and not Raspbee2, maybe this is correct, just to mention it. 

Any hint what I might be doing wrong?

Thanks a lot,

Stephan

Mimiix, 2020-08-27:
Hi! 

Please let me remind you on [this issue](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3199) .

The issue template you [should](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new?assignees=&labels=User+Question&template=user-question.md&title=) have used, asked for logs to be included. This is the very first step in solving the issue. The might be stuff going on we don't know.

Are you able to provide those?

Thanks!

cosmo84, 2020-08-27:
I moved my question to the template and uploaded the logs.

Thanks,

Stephan

Smanar, 2020-08-27:
Hello
>19:09:26:560 channel is 11 but should be 20, start channel change

Can you set the channel again ?

cosmo84, 2020-08-27:
I changed the channel to 20 in the app. Looking at your hint I found out that the app shows 20 after the change, but then looking into deconz, going into the network config and clicking 'read' it switches back to 11. This is reproducable.

[debug2.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5137971/debug2.log)
Anyway, I then changed back to 11 and that seems to work, at least it keeps stable and the app and deconz show the same channel.

Unfortunately it did not change anything concerning the big problem. I still can't add devices. I upload another log with channel 11 and multiple times to connect a sensor.

Smanar, 2020-08-27:
If you have the GUi, can you compare your setting with this one pls https://user-images.githubusercontent.com/1177810/44988558-66835c80-af8b-11e8-9cc1-751f60c2b439.png

cosmo84, 2020-08-27:
The only difference I can see is the NWK Update ID which is 7 in my case. 

[image]

Mimiix, 2020-08-27:
@cosmo84 Are you using HA or Docker? There is a Websocket opened on a 172 ip indicating Docker.

cosmo84, 2020-08-27:
No, I don't use Docker or HA. Deconz runs on a Pi, fresh installed with Raspbian. Nothing else is installed so far.

Mimiix, 2020-08-27:
On the screen you showed, you need to change the channel. 

Can you provide a log when you are searching for lights and trying to add something?

cosmo84, 2020-08-27:
Both logs were created while searching for sensors via the App and trying to pair a Xiaomi water leak sensor.

Mimiix, 2020-08-28:
That is not shown in the log as far as i can tell

cosmo84, 2020-08-28:
Strange. I am using deconz since years and I know how to pair devices, and I tried several from different vendors (Osram, Xiaomi, Hue). In the meanwhile I also tested the SD image from DE, but same issue here. Can it be checked if the Raspbee2 has a hardware issue?

Mimiix, 2020-08-28:
I've send a PM to @manup to look into this for you @cosmo84 . I have no answers left.

cosmo84, 2020-08-28:
Thanks a lot.

Smanar, 2020-08-28:
NWK Update ID = 7 ? So not realy a fresh install with new device.
Have tried to take a look in the hidden config > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues

cosmo84, 2020-08-28:
I have no idea why there is a 7, but I can confirm that I made fresh installations, formatting the SD and flashing it with Raspbian and also the predefined image.

Smanar, 2020-08-28:
And nothing strange in the hidden config ?

cosmo84, 2020-08-28:
In fact that can be the reason. As said above, I had the issue with channel changes in the app, which were not persistent and still the old channel in deconz. For that I was clicking some time to reproduce it :-)

Anyway, no explanation why I can't join devices I guess...

manup, 2020-08-31:
Can you please create a log file with even more debug switches enabled to get a broader picture.

`deCONZ --dbg-info=1 --dbg-aps=1 --dbg-error=1 --http-port=8080`

Preferably with the latest version: http://deconz.dresden-elektronik.de/raspbian/alpha/deconz-2.05.80-raspbian-jessie-alpha~git.90b522.54d442_armhf.deb

To exclude interference as reason:
* Is the Raspberry Pi near any WiFi router or Bluetooth streaming device?
* Are there any SSD or USB 3.0 devices attached to the Raspberry Pi?

cosmo84, 2020-08-31:
I have invested some more time in the issue yesterday. 
I had a SanDisk USB3.0 stick connected and wanted to exclude that. Because I played around a lot, I also made another fresh install, without anything connected, bluetooth and WiFi deactivated. Still the same issue, no devices join. Since my home was now offline for days I made another fresh install with the Raspbee1 connected. After one hour I had all devices connected and homebridge running. 

I really suppose that the Raspbee2 might have a small damage. I was already concerned when I saw that Amazon put it without any isolation into a thin envelope. I will send it back as defect. 

This is not a good end and I want to thank everybody for contributing, but having my home back online was my higher priority since yesterday. 

Last word: If this would be a general issue with Raspbee2, I guess more people would complain, so one more argument for a HW defect. Let's close this for now. Thanks.

Mimiix, 2020-08-31:
@cosmo84 Great to hear from you on this and sharing your story. Let's see if it was a hardware defect.

If you can: I recommend getting a Conbee 2 instead of raspbee.  If you get another Raspbee and have the same issues: Please ask me to re-open this one.

closed 2020-08-31 by Mimiix