dresden-elektronik/deconz-rest-plugin #3204: Device name : TS0601 _TZE200_wunufsil

Grigorek, 2020-08-27:
## Device

- Product name: Tuya Smart Home 2 way
- Manufacturer: _TZE200_wunufsil
- Model identifier: TS0601
- Device type : 
  - Switch

## Screenshots
[image]
[image]
[image]

Smanar, 2020-08-27:
I think we already have it > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2879

Just need to edit code

Do you know how to compile code to test ? BTW wich one deconz version have you ?

Grigorek, 2020-08-27:
I have version 2.05.78, and if u provide some tutorial how to compile it i think i'll handle it. 
This is 2 gang version and I have also 1 gang version of it which have also time cluster
[image]
[image]
[image]
[image]

Smanar, 2020-08-28:
You just need an UNIX OS, the procedure is at https://github.com/dresden-elektronik/deconz-rest-plugin at "Install deCONZ development package (optional, Linux only)"

Just replace step 1 by 
git clone --branch tuya2 https://github.com/Smanar/deconz-rest-plugin.git

Can skip "git checkout -b mybranch HEAD"

After thatn you can re-include the device.
Now the bad new, ATM deconz create 3 entries for all devices, so for 1 gang 2 or 3 , you will have 3 entries, you can delete the useless one later.

Grigorek, 2020-08-30:
Ok, I have managed to build it an replace file but now I can't do anything in phoscon app, neither search for lights or even display gateway info (there are many 404 errors in console) and I don't know if its intended. I have connected via VNC and this is what I see right now 
[image]

Smanar, 2020-08-30:
Ok so there is something bad, all your device are grayed, can be normal for end device but less for router (can change after some minuts)

None of your device are name, idk if it was same before ?
You are using the same deconz instance than previously ?

where are from the 404 error ?

Grigorek, 2020-08-30:
My devices had name before and I'm using the same instance.
404 errors are from api
[image]
I can't see any device in phoscon, and my gateway page is just blank now

Smanar, 2020-08-30:
I know how to have error 403, but error 404 .... Have you tried to delete the cache ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1727

Grigorek, 2020-08-30:
I have recreated container and everything was ok, then copied file and same again - erros.
Container logs says 
19:25:19:638 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin                                                                           
19:25:19:661 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin 
So there is no rest plugin found so I have check if file is there and there 
[image]
So what could I did wrong?

Smanar, 2020-08-30:
You are using container ? whare have you find your deconz version ?
And yes it can explain the problem, but I idk why the procedure fail, all seem fine, surely a subtility with container.

Grigorek, 2020-08-30:
Yea got it from here https://github.com/marthoc/docker-deconz.
Actually I have compiled it on different container because I dont have all necessary tools there.

Grigorek, 2020-08-30:
Hmm, I have updated deconz to 2.05.80 and actually added succesfully one of my switches. It's added in deconz as 3 smart plugs and I can remotely turn on/off one of them.

Smanar, 2020-08-31:
The compilation make the file "hardware dependent" (if we can dare say that for a virtual machine), so if you haven't exactly the same configuration it can explain the problem.

>t's added in deconz as 3 smart plugs and I can remotely turn on/off one of them

Yes was the last state we have stopped it for previous version ^^.

Grigorek, 2020-08-31:
I will just close this issue for now and wait for next version, because they are recognized so far and I think everything should be ok later.

closed 2020-08-31 by Grigorek

SvenPe, 2021-08-19:
I recently got two Tuya 2-way switches that seems to be the same as (or at least similar to) the one mentioned here. Deconz, however, only lets me control the left switch. The right one is not showing up, no matter what I do. Deconz is showing it as Fairy Lights and giving me several effects to choose from. None of them show any response on the switch though and reversely the switch does not have any effect on them. 

I have attached the screenshots for comparison. The only difference is the newer Application Version (66 instead of 64) though, I believe.

@Smanar Do you still have the changes you made in the past to support the switch in your repo? The branch you mentioned here before doesn't exist anymore and I couldn't find any reference to this in your repo myself. 

[image]
[image]
[image]
[image]
[image]

Smanar, 2021-08-19:
Hello,
Bad luck for you, there is a problem, ATM I can't add your device because it can break others one.
To make short It's not possible yet to reconize exaclty your device, so it can break for exemple TRV device.

If you want I can explain how to make it working on your side, if you don't use docker, but I don't think I can make an official PR.

SvenPe, 2021-08-19:
Hey Smanar, 
thanks for the super fast reply. Not entirely the answer I was hoping for, but if it's not too much trouble for you I'd be happy to know how to make it work for me. I'm not using docker, so I hope I'm able to change whatever it is I have to change.

Smanar, 2021-08-19:
Ok so it s easy.
Here you have the procedure to compile the code https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing

Then on de-web_plugin.cpp file, around the line 2162 you will see this code

```
    //Make 2 fakes device for tuya switches
    if (node->nodeDescriptor().manufacturerCode() == VENDOR_EMBER && !node->simpleDescriptors().empty())
    {
        const deCONZ::SimpleDescriptor *sd = &node->simpleDescriptors()[0];
        bool hasColorCluster = false;

        if (sd && (sd->deviceId() == DEV_ID_SMART_PLUG) && (node->simpleDescriptors().size() < 2) &&
        (((node->address().ext() & 0xffffff0000000000ULL ) == silabs3MacPrefix) ||
         ((node->address().ext() & 0xffffff0000000000ULL ) == silabs4MacPrefix))
        )
        {
```

Just add your mac adress

```
        if (sd && (sd->deviceId() == DEV_ID_SMART_PLUG) && (node->simpleDescriptors().size() < 2) &&
        (((node->address().ext() & 0xffffff0000000000ULL ) == silabs3MacPrefix) ||
         ((node->address().ext() & 0xffffff0000000000ULL ) == silabs8MacPrefix) ||
         ((node->address().ext() & 0xffffff0000000000ULL ) == silabs4MacPrefix))
        )
```

SvenPe, 2021-08-19:
Thank you so much! I had just about figured that out when you posted it. The information with the MAC addresses and some searching did the trick. 

It's working now! Cheers, enjoy the coffee!

Smanar, 2021-08-20:
Ha, it was you ^^, thx a lot :)

BTW, I have perhaps a good news, from the release note on the next version https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.12.6 the DDF files system is realy close to be used, and this feature will help to support "custom" device

Arjan1993, 2022-12-17:
Hii,

I have te same issue as above. I`ve got a 3 way tuya switch witch is reconised by Deconz, but not by Phoscon. It shows the same 0X code as above. Is there already another way to get the switches work? Or is this stil the best way?

Smanar, 2022-12-17:
No there is a beter way.
Can use DDF, just check if your manufacture name is not already in PR list (there is so much of them locked), if not make a device request.

DDF are just text fil, easy to add https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

Arjan1993, 2022-12-18:
> No there is a better way. Can use DDF, just check if your manufacture name is not already in PR list (there is so much of them locked), if not make a device request.
> 
> DDF are just text fil, easy to add https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

Thanks for the reply. After some research it became clear to me how it works and I finally managed to get the switches in Phoscon and domoticz.

Thanks a lot :) You made my day.