dresden-elektronik/deconz-rest-plugin #3205: Aqara wall switch QBKG12LM group toggle problem

BenDavidson90, 2020-08-28:
Hi, i got four Aqara wall switch 2 rockers with neutral (QBKG12LM) and I successfully added them in Phoscon.
When added, it creates two lights.

- For two switches, I added the two lights in one only group (two buttons, one group):
[image]
[image]
- For one switch, I added the two lights in two groups (two buttons, one group per button):
[image]
[image]
- And for the last switch, I added one light alone in one group and the other light in an other group with an other light (from a simple rocker switch):
[image]
[image]

### The problem
The problem is that I can control all lights by clicking individually on it HOWEVER in all of the above cases if I switch an entire group the light corresponding to the right button will not toggle (but it's state will).
So, if I link an other switch to toggle the entire group, it will only toggle the left light.

For example:
- In the first of above cases (when the two buttons are in the same group), if I switch on the group, both lights are considered ON by Phoscon but only the left light is really ON ("Lumière Salle de bain" here).
[image]
- Same in the second case (where the right light is alone in one group), here Phoscon thinks that the light in ON but in real life it is not:
[image]

This problem is only with the right button of each switches, no problem with the left button...

Thank's in advance for your help.

ebaauw, 2020-08-28:
Not sure which switch this is, but if memory serves, the `lumi.ctrl_ln` switches only carry a _Groups_ cluster on the first light endpoint.  They simply don't support groups for the second light.  You might want to check what the switch looks like in the GUI, and whether it has a server (blue) _Groups_ cluster on each light endpoint.

BenDavidson90, 2020-08-28:
> Not sure which switch this is, but if memory serves, the `lumi.ctrl_ln` switches only carry a _Groups_ cluster on the first light endpoint. They simply don't support groups for the second light. You [...]

It is the `lumi.ctrl_ln2.aq1`.
Not sure on how to verify what you said, is it here ?
[image]

ebaauw, 2020-08-28:
Yes, you see the _Groups_ cluster is missing from endpoint 0x02.

BenDavidson90, 2020-08-28:
Oh ok. So nothing to do... Thanks for your help, I understand better now.

closed 2020-08-28 by BenDavidson90