dresden-elektronik/deconz-rest-plugin #3206: Aqara wireless switch D1

giuliandenicola1, 2020-08-28:
## Device

- Product name: Aqara Wireless switch D1
- Manufacturer: Xiaomi
- Model identifier: WXKG07LM
- Device type: 
  - Switch

## Screenshots

[image]

Smanar, 2020-08-29:
Hello can you give the capture from deconz ?
Because there is so much model that look like this one and with the same name.

Else it seem this device is already working > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3066

giuliandenicola1, 2020-08-29:
> Hello can you give the capture from deconz ?
You mean this?

[image]
[image]

Smanar, 2020-08-29:
Yep, but for phoscon we can do nothing with the API, need to ask that to Phoscon dev.

Mimiix, 2020-08-29:
@giuliandenicola1 I am giving you the benefit of the doubt here and move this to the Phoscon App repo.

In the future: Please make sure to fill the complete template. 
 

@Smanar If you encounter issues like this in the future (Meant for phoscon repo) , feel free to tag me in the issue so i can move them! Thanks ❤️