dresden-elektronik/deconz-rest-plugin #3209: Disable Leds ststus leds for raspbee and conbee

sorryusernameisalreadytaken, 2020-08-29:
## Feature request type
Extend the REST-API with capabilities to set the status LEDs on or off. 
I'm always frustrated when the status LEDs brightenup my bedroom :D

## Description
I try to find a way to disable all my leds on my raspberry-pi 4. On rpi4 istself this is not a problem but I did not find a way to do the same with the raspbee-1 module.

Just a boolean flag for both LEDs would be enough.

Is there a chance or would it be possible to add this with the next firmware and also with the rest-api?

## Considered alternatives
I dont want to destroy or remove the leds on my raspbee module.

## Additional context
Nice would be a time-based setting for that or the chance to handle this by some lux-sensors. The status LEDs does make sense and should be available if they did jot disturb the user.

Hypfer, 2020-09-15:
Since those _might_ be hard-wired and that's a niche request as niche as you can get, I'd suggest using a 1-3 layers of electrical tape to cover the LEDs. It's 100% revertable, very effective and also very cheap.

sorryusernameisalreadytaken, 2020-09-22:
> 
> 
> Since those _might_ be hard-wired and that's a niche request as niche as you can get, I'd suggest using a 1-3 layers of electrical tape to cover the LEDs. It's 100% revertable, very effective and also very cheap.

I understand your quick solution hint but instead of any self-made zigbee stick this a comercial stick. This could be benefit for dresden elektronik ingenieurtechnik GmbH and there products.

The other german company AVM is a good example for this. All there new updated devices had a shutoff-led setting.

Hypfer, 2020-09-22:
I'm not saying that it's a bad Idea. It's just most likely very low priority and won't happen anytime soon I assume.

Dresden Elektronik is pretty much drowning in work since months now due to the insane increase in popularity of their products.