dresden-elektronik/deconz-rest-plugin #3211: Whitelisting Bitron Video 902010/24A Smoke-Detector

kptkip, 2020-08-31:
## Feature request type
Enabling Smoke-Detector from the Bitron Video Family

## Description
While 902010 is already whitelisted the newer model 902010/24A is not whitelisted yet. Signaling smoke detection works perfectly but `alarm=select` or `alarm=lselect` doesn't work.

Thanks for implementation

ebaauw, 2020-08-31:
> While 902010 is already whitelisted the newer model 902010/24A is not whitelisted yet.

It is, and it isn't.  The current implementation is very inconsistent.  Half of the whitelist entries check for `.startsWith(QLatin1String("902010/24"))`, the other half for `== QLatin1String("902010/24")`.  I don't know why, but I think all entries should use the `.startsWith()` implementation.

> Signaling smoke detection works perfectly but alarm=select or alarm=lselect doesn't work.

I think you mean PUTting a body of `{"alert": "select"}` to the light `state` to activate remotely the siren?  The Bitron uses non-standard values for the _Options_ parameter of the _Start Warning_ command.  And that's exactly where the `==` whitelisting is used.

kptkip, 2020-08-31:
> I think you mean PUTting a body of {"alert": "select"} to the light state to activate remotely the siren?

You're completely right. It's about activating the siren from my smarthome-system to group the smoke-detectors. 

I use the feature via FHEM and there the syntax looks like `set DEVICENAME alert select`

closed 2020-11-14 by manup

kptkip, 2020-12-21:
Are there any new information about the siren activating feature for the 902010/24A smoke detector?

Thanks for any response.

Mimiix, 2020-12-21:
@kptkip How'd you mean?

kptkip, 2020-12-21:
With the whitelisting topic (see above) the siren activation on 902010/24A devices is possible.

Just tested the feature and it works since the last release.