dresden-elektronik/deconz-rest-plugin #3214: ConBee II USB device loop reconnection after x days running stable

MrksHfmn, 2020-09-01:
## Describe the bug
after conbee runs stable for a few days and is connected to all devices, at some point the stick starts to reconnect in a 4-second rhythm. accordingly, i can no longer communicate with the devices. i have to shut down the container, unplug the stick for 10 seconds, plug it in again and start up the container.

## Environment

 - Host system:  PC / NAS
 - Running method: ArchLinux with marthoc/deconz in Docker
 - Firmware version: 26580700
 - deCONZ version: 2.05.80 / 8/14/2020
 - Device: ConBee II
 - Do you use an USB extension cable: no

###  Logs
Sep 01 07:14:49 HOME-SERVER kernel: cdc_acm 1-2:1.0: ttyACM0: USB ACM device
Sep 01 07:15:20 HOME-SERVER kernel: usb 1-2: USB disconnect, device number 30
Sep 01 07:15:20 HOME-SERVER kernel: usb 1-2: new full-speed USB device number 31 using xhci_hcd
Sep 01 07:15:20 HOME-SERVER kernel: usb 1-2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 01 07:15:20 HOME-SERVER kernel: usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 01 07:15:20 HOME-SERVER kernel: usb 1-2: Product: ConBee II
Sep 01 07:15:20 HOME-SERVER kernel: usb 1-2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 01 07:15:20 HOME-SERVER kernel: usb 1-2: SerialNumber: DE2147881
Sep 01 07:15:20 HOME-SERVER kernel: cdc_acm 1-2:1.0: ttyACM0: USB ACM device
Sep 01 07:15:23 HOME-SERVER kernel: usb 1-2: USB disconnect, device number 31
Sep 01 07:15:24 HOME-SERVER kernel: usb 1-2: new full-speed USB device number 32 using xhci_hcd
Sep 01 07:15:24 HOME-SERVER kernel: usb 1-2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 01 07:15:24 HOME-SERVER kernel: usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 01 07:15:24 HOME-SERVER kernel: usb 1-2: Product: ConBee II
Sep 01 07:15:24 HOME-SERVER kernel: usb 1-2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 01 07:15:24 HOME-SERVER kernel: usb 1-2: SerialNumber: DE2147881
Sep 01 07:15:24 HOME-SERVER kernel: cdc_acm 1-2:1.0: ttyACM0: USB ACM device
Sep 01 07:15:59 HOME-SERVER kernel: usb 1-2: USB disconnect, device number 32
Sep 01 07:16:24 HOME-SERVER kernel: usb 1-1: new full-speed USB device number 33 using xhci_hcd
Sep 01 07:16:24 HOME-SERVER kernel: usb 1-1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 01 07:16:24 HOME-SERVER kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 01 07:16:24 HOME-SERVER kernel: usb 1-1: Product: ConBee II
Sep 01 07:16:24 HOME-SERVER kernel: usb 1-1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 01 07:16:24 HOME-SERVER kernel: usb 1-1: SerialNumber: DE2147881
Sep 01 07:16:24 HOME-SERVER kernel: cdc_acm 1-1:1.0: ttyACM0: USB ACM device
Sep 01 07:16:28 HOME-SERVER kernel: usb 1-1: USB disconnect, device number 33
Sep 01 07:16:28 HOME-SERVER kernel: usb 1-1: new full-speed USB device number 34 using xhci_hcd
Sep 01 07:16:28 HOME-SERVER kernel: usb 1-1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 01 07:16:28 HOME-SERVER kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 01 07:16:28 HOME-SERVER kernel: usb 1-1: Product: ConBee II
Sep 01 07:16:28 HOME-SERVER kernel: usb 1-1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 01 07:16:28 HOME-SERVER kernel: usb 1-1: SerialNumber: DE2147881

Kazfr, 2020-09-01:
Good morning,

This is the exact same for me just now....
The raspberry pi work since two days and this morning:

`Sep  1 07:57:29 usbs kernel: [   94.366771] usb 1-1.1.2.1.1: reset full-speed USB device number 9 using dwc_otg
Sep  1 07:57:29 usbs kernel: [   94.498605] usb 1-1.1.2.1.1: device firmware changed
Sep  1 07:57:29 usbs kernel: [   94.499511] usb 1-1.1.2.1.1: USB disconnect, device number 9
Sep  1 07:57:29 usbs kernel: [   94.596760] usb 1-1.1.2.1.1: new full-speed USB device number 13 using dwc_otg
Sep  1 07:57:29 usbs kernel: [   94.731735] usb 1-1.1.2.1.1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep  1 07:57:29 usbs kernel: [   94.731754] usb 1-1.1.2.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep  1 07:57:29 usbs kernel: [   94.731766] usb 1-1.1.2.1.1: Product: ConBee II
Sep  1 07:57:29 usbs kernel: [   94.731779] usb 1-1.1.2.1.1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep  1 07:57:29 usbs kernel: [   94.731790] usb 1-1.1.2.1.1: SerialNumber: DE2189638
Sep  1 07:57:34 usbs kernel: [   99.096769] usb 1-1.1.2.1.1: reset full-speed USB device number 13 using dwc_otg
Sep  1 07:57:34 usbs kernel: [   99.228621] usb 1-1.1.2.1.1: device firmware changed
Sep  1 07:57:34 usbs kernel: [   99.229406] usb 1-1.1.2.1.1: USB disconnect, device number 13
Sep  1 07:57:34 usbs kernel: [   99.326763] usb 1-1.1.2.1.1: new full-speed USB device number 14 using dwc_otg
Sep  1 07:57:34 usbs kernel: [   99.461229] usb 1-1.1.2.1.1: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep  1 07:57:34 usbs kernel: [   99.461246] usb 1-1.1.2.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep  1 07:57:34 usbs kernel: [   99.461258] usb 1-1.1.2.1.1: Product: ConBee II
Sep  1 07:57:34 usbs kernel: [   99.461271] usb 1-1.1.2.1.1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep  1 07:57:34 usbs kernel: [   99.461282] usb 1-1.1.2.1.1: SerialNumber: DE2189638`

After rebooting the device, all is working again...

Mimiix, 2020-09-01:
@MrksHfmn Did you perhaps click the update firmware button?

Looks like a firmware that is broken ish.

MrksHfmn, 2020-09-01:
> @MrksHfmn Did you perhaps click the update firmware button?
> 
> Looks like a firmware that is broken ish.

Thanks for your reply. No. Is it possible to "reflash" the current firmware?

Mimiix, 2020-09-01:
Yep!
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

MrksHfmn, 2020-09-01:
> Yep!
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Thanks! I re-flashed the device. Let's wait and see ...

```
GCFFlasher_internal -d /dev/ttyACM0 -t 60 -f deCONZ_ConBeeII_0x26580700.bin.GCF
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
deCONZ firmware version 26580700
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 160930 bytes: |==============================|
verify: .
SUCCESS
Wait 10 seconds until application starts
```

Mimiix, 2020-09-01:
@MrksHfmn Lets see what happens :)

@kazfr If you have issue's : I'd like to ask you to open a own User question issue :)

Mimiix, 2020-09-03:
@MrksHfmn I just saw the word "ArchLinux", I've read about this earlier on arch. However, please note that arch is not supported and is used at own expertise with deconz.

Kazfr, 2020-09-03:
Thank you @MrksHfmn for sharing this fix.

SwoopX, 2020-09-04:
@MrksHfmn Thanks for sharing your steps. I suspected this to be a potential solution but that one time I had this situation, it didn't work out (was no docker though). Is that a procedure you still need to apply every now and then? Might be worth do dedicate a wiki article to it.

MrksHfmn, 2020-09-05:
> 
> 
> @MrksHfmn Thanks for sharing your steps. I suspected this to be a potential solution but that one time I had this situation, it didn't work out (was no docker though). Is that a procedure you still need to apply every now and then? Might be worth do dedicate a wiki article to it.

the solution is not perfect, but it runs stable since a few days. anyway, it's because of the cdc_acm kernel module that you have to unload and reload. as soon as the kernel module is reloaded and the container/deconz is started, the conbee stick is mounted stable. if i reboot the computer and **don't** start the container/deconz i'm in an endless reconnect loop every 10 seconds

SwoopX, 2020-09-05:
I assume that could be handled by a script. I've seen one while researching potential solutions. Happy to share the link if I can dig it later in case you're interested.

MrksHfmn, 2020-09-25:
knocked on wood ... the problem was solved with the last deCONZ firmware update. maybe there were some kernel adjustments in linux 5.8 too. after the update i can't find any bootloops anymore. the range of the stick is also much wider than before ... i close the issue. many thanks to all

closed 2020-09-25 by MrksHfmn