dresden-elektronik/deconz-rest-plugin #3215: Release cycle: When are .79 and .80 released to stable?

mountainsandcode, 2020-09-01:
## Describe the question or issue you are having

Have a quick question regarding the release current release cycle - I tried looking in the Wiki - apologies if I missed something.

I’m currently running 2.05.78 on Raspbian, having installed it using apt-get and the deconz repository. I noticed that the two most recent releases were originally released as beta, but have not since been released as stable. Is that intentional? Are there already plans when the next stable will be released?

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: (26xxyy00) Latest
 - deCONZ version: 2.05.78
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II) Conbee II
 - Do you use an USB extension cable: (yes / no) Yes

### deCONZ Logs

## Additional context

manup, 2020-09-01:
The next stable release is coming within the next days, there are still a few issues which need to be fixed.

Beta releases are scheduled on every 15th of the month. For stable releases there isn't a fixed schedule yet, the near term goal is a predictable cycle to bring them within 2–3 weeks after a beta release with mostly bug fixes from the beta.

mountainsandcode, 2020-09-01:
Super, thanks!

closed 2020-09-01 by mountainsandcode