dresden-elektronik/deconz-rest-plugin #3219: indirection requires pointer operand on rest_configuration.cpp

Rania-M, 2020-09-01:
Hi,
after downloading the deconz-dev package and running the project on linux, i had this error :

**deconz-rest-plugin-master/rest_configuration.cpp:88: error: indirection requires pointer operand ('unsigned long long' invalid)
        const QDateTime d = QDateTime::fromMSecsSinceEpoch(GW_SW_DATE * 1000LLU);**

I think the problem comes from the 1000LLU variable name, it seems though that it is not used anywhere else in the code, so i may just have to change its name, but i was wondering if anyone else had this problem and if i'm missing something important
Thank you

ebaauw, 2020-09-01:
That’s not a variable, but an integer literal: 1000 as long long unsigned integer.  The `*` is the binary multiply operator, not the unary pointer dereference.  Probably `GW_SW_DATE` isn’t `#define`d properly.

Rania-M, 2020-09-01:
@ebaauw thank you , yes i was changing my comment i just realized that its is an integer ! , as for GW_SW_DATE, i just downloaded the plugin as it is and the variable seems to be defined, i think the IDE has mistaken the * operand for a pointer (as i did), i'm gonna check the encoding, thank you !

Rania-M, 2020-09-01:
Okay so the problem comes from the LLU notation, i suppose it should be pre defined somewhere right !?

ebaauw, 2020-09-01:
Not sure, but I think it might actually be passed to the compiler through `-D`.  The `qmake` creates a `Makefile` with some of these from, I think, `de_web.pro`.  The IDE probably doesn’t understand ths.

Rania-M, 2020-09-01:
I am using it in QtCreator , and the de_web.pro is actually .pri as i am using it as a package inside a bigger project, maybe this causes this problem

ebaauw, 2020-09-01:
Back at my computer:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/474ac118d5b051c71b3b45698d3a694b2493ff2b/de_web.pro#L79

> I am using it in QtCreator , and the de_web.pro is actually .pri as i am using it as a package inside a bigger project, maybe this causes this problem

I'm afraid that's beyond me.

Rania-M, 2020-09-02:
thank you, i erased the LLU notation for now, i will deal with it after, now i have issues with sqlite, all the methods seem to be unrecognized, is there something to add to the project installation ? i do have sqlite3 installed but i still got errors like : 

**deconz-rest-plugin-master/database.cpp:225: error: undefined reference to `sqlite3_prepare_v2'**

thank you for your help

Rania-M, 2020-09-02:
Answering my question, in a Linux environnement,  pkg-config package needs to be installed as Linux use it to find sqlite3, this may help someone in the same situation !

closed 2020-09-02 by Rania-M