dresden-elektronik/deconz-rest-plugin #3220: Zigbee 3 Gang Remote TS0043

andysaab, 2020-09-01:
## Device

- Product name: Zigbee 3 Gang Remote
[image]
[image]

- Manufacturer: _TZ3000_bi6lpsew.
- Model identifier: TS0043
- Device type : Please remove all unrelated device types. 
    - Remote

## Screenshots

  Required screenshots:
  - Endpoints and clusters of the node
[image]

  - Node Info panel
[image]

  - Basic Cluster attributes in the Cluster Info panel.
[image]

  Screenshots of other clusters.
  
[image]
[image]

It is added to HA as three switches, but have`t any action pressing any button.
I bought two remote and add second one to zigbee2mqtt.
It is recognized as a one click button with three actions: right_click, middle_click and left_click. Thats all, no any long clicks or other actions.
Also send via MQTT value of battery level and battery voltage.
[image]
[image]

Smanar, 2020-09-01:
Can you try this version pls ? https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3218

To compile it, change the step 1 by
`git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git`

andysaab, 2020-09-01:
Smanar, it's a different opposite tuya2 branch from ZM25TQ's topic? Can we use same branch?
It's more quite for me. But if it impossible I need powered other VM.

Smanar, 2020-09-01:
Yep ATM "tuya2" is the same than "tuya".
But I can't say if the 3 endpoint you have are real or fake. From the code it works only for model = DEV_ID_SMART_PLUG and you have on / off switch.
Range ip is not good you have silabs7MacPrefix
And VENDOR_EMBER is good.

So I don't think it s the deconz code that make the fake device, it s for that I prefer you use the last tuya change before chnaging code.

andysaab, 2020-09-02:
Some updates for this remote:
[image]
[image]
[image]
[image]

andysaab, 2020-09-02:
> Can you try this version pls ? #3218
> 
> To compile it, change the step 1 by
> `git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git`

[image]

Smanar, 2020-09-02:
Ok so sorry I haven't see it s a battery device, not the same than previoulsy, so forget that I have said.
From z2m it seem this device use the cluster 0006, but with special command.

- attribute reporting is working ?
- the attribute 0x0000 in the cluster 0x0006 change or not if you use the device ?

I have alreadu coorected this bug, but the code move a lot, sorry

I have added your device in whitelist for test, but I realy don't understand what happen, this device is end device and not yet in whitelist so strange you already have it in phoscon ....

BTW the command is 
`git clone --branch tuya2 https://github.com/Smanar/deconz-rest-plugin.git`

Tuya is the offcial, tuya2 the modified one.

andysaab, 2020-09-02:
Smanar, I have a problems.
What I did:
1. Delete folder deconz-rest-plugin
2. git clone --branch tuya2 https://github.com/Smanar/deconz-rest-plugin.git
3. Go to deconz-rest-plugin folder
4. qmake && make -j2
Got a lot of errors

Smanar, 2020-09-02:
ok give me 30 mn, to compile it.

Smanar, 2020-09-02:
Done, and BTW I have found why your device is visible in phoscon, and why it s a light.

It s because VENDOR_EMBER in whitelist ....

andysaab, 2020-09-02:
> * attribute reporting is working ?
Not clear, please clarify.
> * the attribute 0x0000 in the cluster 0x0006 change or not if you use the device ?
Nothing changes when I click on the buttons, double click or hold. Always "false" value.

Smanar, 2020-09-02:
> 
> 2. Manually set up attribute reporting
> Then you need to set up attribute reporting. In most cases, there are default settings already defined and should be placed into action either after a deconz restart or power-cycling the device. However, if values do not get updated upon the taken action or not at all, attributes reporting has presumably not been set up properly.
> 
> Select windows covering cluster of your device, double click on 0x0008 (current position).
> On the lower end of the dialog, enter the values
> 
> Min. report interval: 1
> Max. report interval: 300
> Reportable change: 1
> 
> then press write config

But I think you will have an error message, and as the device is battery one, you need to wake it up during the command.

andysaab, 2020-09-02:
Now compile without any errors.
What do we do next? I`m ready for tests.

Smanar, 2020-09-02:
Oups sorry, was eating.

- Retry to include the device.
Problem, it will be managed as router, I need to disable that tommorow
- Take a look in entries created in the API, in "lights" but I think the "sensors" will be more usefull.
- Take a look if something is changing in the json, the field "state/on" ou 'state/buttenevent"

More usefull, try to run deconz with debug enabled
`deCONZ --dbg-error=2 --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=0 --dbg-http=0`
And use a button, I m not sure the actual code is enought to get raw value in cluster 0006, byt can make a try.

The devices will be invisible in phoscon (Yeah I know ATM you can see 3 devices, but they will disapear on the next modification).
So you need to check directly the API. You can use an home automation application or just a browser.

https://dresden-elektronik.github.io/deconz-rest-doc/

To get an API key, with curl
`curl http://127.0.0.1:80/api/ -X POST -d '{"devicetype": "TestKey"}'`

The url to have sensors list 
`http://IP:PORT/api/API_KEY/sensors`

IP and PORT are the same used for phoscon.

andysaab, 2020-09-02:
> * Retry to include the device.
>   Problem, it will be managed as router, I need to disable that tommorow
Added as endpoint
[image]

> * Take a look in entries created in the API, in "lights" but I think the "sensors" will be more usefull.
In "lights" - nothing on page.
In "sensors"
[image]

> * Take a look if something is changing in the json, the field "state/on" ou 'state/buttenevent"
Nothing
> More usefull, try to run deconz with debug enabled
> `deCONZ --dbg-error=2 --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=0 --dbg-http=0`
> And use a button, I m not sure the actual code is enought to get raw value in cluster 0006, byt can make a try.
After pressing any buttons this message appears
[image]

> The devices will be invisible in phoscon (Yeah I know ATM you can see 3 devices, but they will disapear on the next modification).
Yes, it invisible.

Smanar, 2020-09-03:
I mean in phoscon, deconz is always right.
But on your last try, it have failed, the sensor you can see is the integrated one. so you have no sensor at all in the API.

If your device is already in deconz you can use this trick
- set phoscon in permit join (add new sensor)
- go on cluster 0x0000
- on the cluster panl press the "read" button (you need to wake up the device probably in same time)

It wil update the basic attribute and trigger a new inclusion in same time.

I have updated the code to prevent the inclusion in light section, so the device will be visible only in the API.

To update the code just use
```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```
and reload deconz.

andysaab, 2020-09-03:
Hi Smanar!
I did all yours recommendation but no changes - lights section is empty. Sensor section shown Daylight as previously.
[image]

Maybe I do something wrong?

Smanar, 2020-09-03:
Even using the trick ?
> If your device is already in deconz you can use this trick
> 
>     set phoscon in permit join (add new sensor)
>     go on cluster 0x0000
>     on the cluster panl press the "read" button (you need to wake up the device probably in same time)

andysaab, 2020-09-03:
> Are you sure for the pairing method ? on your second try it haven't worked, but the code was not modified yet.
No, no any tricks. Always delete devices and pair it again.

andysaab, 2020-09-03:
Smanar, I don`t know what vent wrong but after " curl http://127.0.0.1:8080/api/ -X POST -d '{"devicetype": "TestKey"}' " I got new API key, try to use it and faced this message, same for lights and sensors section:
[image]

andysaab, 2020-09-03:
After that started deconz as deCONZ --dbg-error=2 --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=0 --dbg-http=0
Got this
[image]

Smanar, 2020-09-03:
Yep, everytime you use this command, it generate a new api key, but the previous was not deleted.
For your bug , perhaps a typo ?
And for the "configuration tool" it s the conbee, so missed again ^^.

But yes it s a good idea to run deconz in debug mode during inclusion but to have less lines can use 
`deCONZ --dbg-error=2 --dbg-info=2 --dbg-zdp=0 --dbg-zcl=0 --db-aps=0 --dbg-http=0`

For the moment just need to find why the inclusion is not working. Your device is still in deconz ? the trick is not working ? you have the LED blinking in deconz in the node title ?

andysaab, 2020-09-03:
> Yep, everytime you use this command, it generate a new api key, but the previous was not deleted.
> For your bug , perhaps a typo ?
Think no, always use copy&paste and check twice.
> And for the "configuration tool" it s the conbee, so missed again ^^.
[image]

> But yes it s a good idea to run deconz in debug mode during inclusion but to have less lines can use
> `deCONZ --dbg-error=2 --dbg-info=2 --dbg-zdp=0 --dbg-zcl=0 --db-aps=0 --dbg-http=0`
Done
> For the moment just need to find why the inclusion is not working. Your device is still in deconz ? the trick is not working ? you have the LED blinking in deconz in the node title ?
Yes, device in deconz and LED blinking - you see it on picture above

andysaab, 2020-09-04:
Smanar,
I did a fresh installationof VM, make all previous recomendation - got the same result
[image]

Smanar, 2020-09-04:
I m checking the code again but all seem fine.
For inclusion, you are setting Phoscon in permet join (and not deconz) with "add new sensor", then you include the device, and you have a message in phoscon ?

andysaab, 2020-09-07:
Hi Smanar!
So if we are finished with motor let`s continue this request?

Smanar, 2020-09-07:
Sure, but for the moment I realy don't see problem.
Can you try to make an inclusion but with debug enabled in same time ?

andysaab, 2020-09-07:
> Sure, but for the moment I realy don't see problem.
> Can you try to make an inclusion but with debug enabled in same time ?
Smanar, done.
I`m ready for tests. Now test`s instance have only one remote connected to deconz.
In deconz I see stick and remote. But remote not appears in lights and sensors section.

Smanar, 2020-09-07:
It s for that I want to take a look in log during the inclusion.
The device is on whitelist
>{ VENDOR_EMBER, "TS0043", silabs7MacPrefix }, // Tuya triple gang wireless

And have an usefull cluster, the 0x0006, so there is no reason for no device was created.

andysaab, 2020-09-07:
Where is this log?
In Terminal messages goes non-stop.

Smanar, 2020-09-07:
Wait a little, I think I have found the problem on an other issue.

andysaab, 2020-09-07:
No problem. Still wait.

ebaauw, 2020-09-08:
> The device is on whitelist

That’s the whitelist for sensor resources.

> so there is no reason for no device was created

It doesn’t have _Receiver On When Idle_.  You’ll need to whitelist it at that check to create the light resources.

Smanar, 2020-09-08:
> > The device is on whitelist
> 
> That’s the whitelist for sensor resources.
> 
> > so there is no reason for no device was created
> 
> It doesn’t have _Receiver On When Idle_. You’ll need to whitelist it at that check to create the light resources.

But it s a battery device, so its better to manage it at sensor nope ?

ebaauw, 2020-09-08:
Sorry, I assumed this was a wired switch, because of the server (!) _On/Off_ clusters listed [above](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3220#issuecomment-685975147).

Do I understand that it is a wireless switch, acting as a sensor, using attribute reporting on the _On/Off_ to signal button presses?

This is very different from “normal” wireless switches, acting as controller, using a client _On/Off_ cluster to control (a group of) lights.

In that case you probably need to whitelist it while handling the server _Basic_ cluster.  This is done for the Xioami Aqara door sensor, which also uses this cluster, to indicate whether there’s contact. There’s no straightforward mapping of the cluster to the sensor type, because of this weird use of the server _On/Off_ cluster.

ebaauw, 2020-09-08:
Here: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/474ac118d5b051c71b3b45698d3a694b2493ff2b/de_web_plugin.cpp#L4462

andysaab, 2020-09-08:
Hi guys!
Sorry to interrupt but do I need any further action?

Smanar, 2020-09-08:
> Do I understand that it is a wireless switch, acting as a sensor, using attribute reporting on the On/Off to signal button presses?
> This is very different from “normal” wireless switches, acting as controller, using a client On/Off cluster to control (a group of) lights.

IDK yet ^^, it s tuya, they are able to use another invisible  cluster for return, but I hope yes
and thx, was the part I have missed

@andysaab you can make a try with the new code pls ?

andysaab, 2020-09-08:
Yes, I can. 
Please let me know what I need to do.

Smanar, 2020-09-08:
Now ^^.
You will have problem at restart, but the inclusion can work now, with the new whilelist.

Smanar, 2020-09-08:
yep

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

andysaab, 2020-09-09:
Hi Smanar!
All I did. Start Add new sensors, reset remote and Phoscon reported Sensor ready but not appears in Phoscon.
[image]

[image]
[image]

Smanar, 2020-09-09:
They will be invisible in phoscon, but it s better we can see it in the API.
Entries are correct, buttinevent is null atm, can you use the device to check if the value change ?

andysaab, 2020-09-09:
> They will be invisible in phoscon, but it s better we can see it in the API.
> Entries are correct, buttinevent is null atm, can you use the device to check if the value change ?
Check button event during pressing keys - no changes.

Smanar, 2020-09-09:
Why I m not suprised ....
Ok so it seem the device use the cluster 0x0006 but with raw command, so I have updated the code, to display some informations.

So same than previously
```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

But run deconz with
`deCONZ --dbg-error=2 --dbg-info=2`

Every time you will press a button you will have a message 
>ZCL attribute report 0x%016llX for cluster: 0x%04X, ep: 0x%02X, frame control: 0x%02X, mfcode: 0x%04X

I just need this message and a "payload" just after.

andysaab, 2020-09-09:
Sorry for a lot of info - too hard to catch running log :(

1. First button:
 One click:
0:04:31:665 APS-DATA.indication from child 0xFD6F
00:04:31:665 verify 0xbc33acfffe6be774 is child node after 39 s
00:04:31:666 no button map for: TS0043 ep: 0x01 cl: 0x0006 cmd: 0xFD pl[0]: 000
00:04:31:671 verify neighbor status: APP_SUCCESS (0x00) 
00:04:31:751 MAC Poll 0x02 0xFD6F
00:04:31:853 MAC Poll 0x02 0xFD6F
00:04:31:956 MAC Poll 0x02 0xFD6F
00:04:32:323 sql exec SELECT conf FROM zbconf ORDER BY rowid desc limit 1
00:04:32:323 Idle timer triggered
00:04:32:674 APS-DATA.indication from child 0xFD6F
00:04:32:674 no button map for: TS0043 ep: 0x01 cl: 0x0006 cmd: 0xFD pl[0]: 000
00:04:32:760 MAC Poll 0x02 0xFD6F
00:04:32:862 MAC Poll 0x02 0xFD6F
00:04:32:964 MAC Poll 0x02 0xFD6F
00:04:33:650 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
00:04:33:773 MAC Poll 0x02 0xFD6F
00:04:33:822 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
00:04:36:529 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000

Double click:
00:06:11:567 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
00:06:13:506 APS-DATA.indication from child 0xFD6F
00:06:13:506 verify 0xbc33acfffe6be774 is child node after 101 s
00:06:13:506 no button map for: TS0043 ep: 0x01 cl: 0x0006 cmd: 0xFD pl[0]: 001
00:06:13:511 verify neighbor status: APP_SUCCESS (0x00) 
00:06:13:592 MAC Poll 0x02 0xFD6F
00:06:13:694 MAC Poll 0x02 0xFD6F
00:06:13:796 MAC Poll 0x02 0xFD6F
00:06:13:824 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
00:06:14:451 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
00:06:14:517 APS-DATA.indication from child 0xFD6F
00:06:14:517 no button map for: TS0043 ep: 0x01 cl: 0x0006 cmd: 0xFD pl[0]: 001
00:06:14:601 MAC Poll 0x02 0xFD6F
00:06:14:702 MAC Poll 0x02 0xFD6F
00:06:14:805 MAC Poll 0x02 0xFD6F
00:06:15:613 MAC Poll 0x02 0xFD6F
00:06:17:318 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
00:06:20:209 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
00:06:23:089 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
00:06:23:823 Daylight now: nightStart, status: 230, daylight: 0, dark: 1

Hold button:
0:07:09:158 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
00:07:09:322 Idle timer triggered
00:07:09:533 APS-DATA.indication from child 0xFD6F
00:07:09:534 verify 0xbc33acfffe6be774 is child node after 56 s
00:07:09:534 no button map for: TS0043 ep: 0x01 cl: 0x0006 cmd: 0xFD pl[0]: 002
00:07:09:535 verify neighbor status: APP_SUCCESS (0x00) 
00:07:09:624 MAC Poll 0x02 0xFD6F
00:07:09:723 MAC Poll 0x02 0xFD6F
00:07:09:826 MAC Poll 0x02 0xFD6F
00:07:10:544 APS-DATA.indication from child 0xFD6F
00:07:10:545 no button map for: TS0043 ep: 0x01 cl: 0x0006 cmd: 0xFD pl[0]: 002
00:07:10:632 MAC Poll 0x02 0xFD6F
00:07:10:735 MAC Poll 0x02 0xFD6F
00:07:10:837 MAC Poll 0x02 0xFD6F
00:07:11:646 MAC Poll 0x02 0xFD6F
00:07:12:053 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000

2. Second button
One click:
00:09:02:626 MAC Poll 0x02 0xFD6F
00:09:02:727 MAC Poll 0x02 0xFD6F
00:09:02:827 MAC Poll 0x02 0xFD6F
00:09:03:545 APS-DATA.indication from child 0xFD6F
00:09:03:546 no button map for: TS0043 ep: 0x02 cl: 0x0006 cmd: 0xFD pl[0]: 000
[... 79 lines cut ...]
00:12:01:505 APS-DATA.indication from child 0xFD6F
00:12:01:505 no button map for: TS0043 ep: 0x03 cl: 0x0006 cmd: 0xFD pl[0]: 001
00:12:01:592 MAC Poll 0x02 0xFD6F
00:12:01:696 MAC Poll 0x02 0xFD6F
00:12:01:796 MAC Poll 0x02 0xFD6F
00:12:02:605 MAC Poll 0x02 0xFD6F
00:12:02:928 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000

Hold button:
0:12:34:609 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
00:12:37:488 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
00:12:37:906 APS-DATA.indication from child 0xFD6F
00:12:37:906 verify 0xbc33acfffe6be774 is child node after 37 s
00:12:37:906 no button map for: TS0043 ep: 0x03 cl: 0x0006 cmd: 0xFD pl[0]: 002
00:12:37:911 verify neighbor status: APP_SUCCESS (0x00) 
00:12:37:992 MAC Poll 0x02 0xFD6F
00:12:38:093 MAC Poll 0x02 0xFD6F
00:12:38:196 MAC Poll 0x02 0xFD6F
00:12:38:916 APS-DATA.indication from child 0xFD6F
00:12:38:916 no button map for: TS0043 ep: 0x03 cl: 0x0006 cmd: 0xFD pl[0]: 002
00:12:38:001 MAC Poll 0x02 0xFD6F
00:12:39:104 MAC Poll 0x02 0xFD6F
00:12:39:206 MAC Poll 0x02 0xFD6F
00:12:40:013 MAC Poll 0x02 0xFD6F
00:12:40:367 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000

Smanar, 2020-09-09:
Nice perfect, I have all, thx
You will have the final code tommorow

Smanar, 2020-09-10:
Done, the code is online.

IDK yet if the 3 entries will work, perhaps you will have aonly the first one than change for the 3.
And BTW do you want to regroup the 3 entries in one ?

andysaab, 2020-09-10:
> Done, the code is online.
Thx, Smanar. I'll check it out evening.
> 
> IDK yet if the 3 entries will work, perhaps you will have only the first one than change for the 3 (the buttonevent field)
> And BTW do you want to regroup the 3 entries in one ?
Hmm.... What will it look like in deconz/Phoscon and HA? One switch with three states like a Xiaomi wireless button?
In general I don't mind grouping as long as it works later in HA.

andysaab, 2020-09-10:
Hi Smanar!
Some info with new code.
[image]

Log for first button (for second and third buttons is a same):

First button one click:

19:06:34:323 Idle timer triggered
19:06:34:600 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:06:37:470 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:06:39:823 Daylight now: sunsetEnd, status: 200, daylight: 0, dark: 0
19:06:40:361 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:06:43:242 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:06:46:120 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:06:46:455 APS-DATA.indication from child 0xB86F
19:06:46:455 verify 0xbc33acfffe6be774 is child node after 19 s
19:06:46:457 button 1002 B1 short
19:06:46:461 verify neighbor status: APP_SUCCESS (0x00) 
19:06:46:462 Websocket 192.168.1.96:52872 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-09-10T16:06:46.457"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
19:06:46:464 discard sensor state push for 2: state/lastupdated (already pushed)
19:06:46:540 MAC Poll 0x02 0xB86F
19:06:46:642 MAC Poll 0x02 0xB86F
19:06:46:743 MAC Poll 0x02 0xB86F
19:06:47:465 APS-DATA.indication from child 0xB86F
19:06:47:467 button 1002 B1 short
19:06:47:472 Websocket 192.168.1.96:52872 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-09-10T16:06:47.466"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
19:06:47:474 discard sensor state push for 2: state/lastupdated (already pushed)
19:06:47:552 MAC Poll 0x02 0xB86F
19:06:47:653 MAC Poll 0x02 0xB86F
19:06:47:756 MAC Poll 0x02 0xB86F
19:06:48:562 MAC Poll 0x02 0xB86F
19:06:48:000 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:06:49:822 Daylight now: sunsetEnd, status: 200, daylight: 0, dark: 0

First button double click:

19:07:09:823 Daylight now: sunsetEnd, status: 200, daylight: 0, dark: 0
19:07:12:035 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:07:14:919 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:07:17:802 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:07:19:822 Daylight now: sunsetEnd, status: 200, daylight: 0, dark: 0
19:07:20:680 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:07:20:682 Websocket 192.168.1.96:52872 send message: {"attr":{"lastannounced":null,"lastseen":"2020-09-10T16:07Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","swversion":"0x264a0700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:04:f2:df-01"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:21:2e:ff:ff:04:f2:df-01"} (ret = 335)
19:07:20:006 APS-DATA.indication from child 0xB86F
19:07:20:006 verify 0xbc33acfffe6be774 is child node after 34 s
19:07:21:007 button 1002 B1 short
19:07:21:012 Websocket 192.168.1.96:52872 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-09-10T16:07:21.007"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
19:07:21:014 discard sensor state push for 2: state/lastupdated (already pushed)
19:07:21:015 verify neighbor status: APP_SUCCESS (0x00) 
19:07:21:088 MAC Poll 0x02 0xB86F
19:07:21:190 MAC Poll 0x02 0xB86F
19:07:21:290 MAC Poll 0x02 0xB86F
19:07:22:009 APS-DATA.indication from child 0xB86F
19:07:22:011 button 1002 B1 short
19:07:22:018 Websocket 192.168.1.96:52872 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-09-10T16:07:22.011"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
19:07:22:020 discard sensor state push for 2: state/lastupdated (already pushed)
19:07:22:113 MAC Poll 0x02 0xB86F
19:07:22:197 MAC Poll 0x02 0xB86F
19:07:22:300 MAC Poll 0x02 0xB86F
19:07:23:107 MAC Poll 0x02 0xB86F
19:07:23:563 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000

First button Hold:

19:07:49:822 Daylight now: sunsetEnd, status: 200, daylight: 0, dark: 0
19:07:52:362 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:07:55:240 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:07:58:110 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:07:59:822 Daylight now: sunsetEnd, status: 200, daylight: 0, dark: 0
[... 1 lines cut ...]
19:08:03:872 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:08:04:323 Idle timer triggered
19:08:05:060 APS-DATA.indication from child 0xB86F
19:08:05:060 verify 0xbc33acfffe6be774 is child node after 44 s
19:08:05:061 button 1002 B1 short
19:08:05:066 verify neighbor status: APP_SUCCESS (0x00) 
19:08:05:067 Websocket 192.168.1.96:52872 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-09-10T16:08:05.061"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
19:08:05:070 discard sensor state push for 2: state/lastupdated (already pushed)
19:08:05:150 MAC Poll 0x02 0xB86F
[... cut ...]
19:08:07:171 MAC Poll 0x02 0xB86F
19:08:09:642 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:08:09:822 Daylight now: sunsetEnd, status: 200, daylight: 0, dark: 0
19:08:12:521 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:08:15:400 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
19:08:18:283 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000

Smanar, 2020-09-10:
My bad I compare only mode, ep, cluster, cmd  and not param.

Code updated

andysaab, 2020-09-10:
> My bad I compare only mode, ep, cluster, cmd and not param.
> 
> Code updated
andysaab@Deconz-test:~/deconz-rest-plugin$ qmake && make -j2make -f Makefile.Release
make[1]: Entering directory '/home/andysaab/deconz-rest-plugin'
g++ -c -m64 -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=c++0x -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS -DHAS_SQLITE3 -DGW_SW_VERSION=\"2.05.80\" -DGW_SW_DATE=1599757373 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"22d3c35dda7020f6b399b865810f44c17923d0d9\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26350500 -DGW_MIN_R21_FW_VERSION=0x26580700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtSerialPort -isystem /usr/include/x86_64-linux-gnu/qt5/QtWebSockets -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -Irelease -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -o release/de_web_plugin.o de_web_plugin.cpp
de_web_plugin.cpp: In member function ‘void DeRestPluginPrivate::checkSensorButtonEvent(Sensor*, const deCONZ::ApsDataIndication&, const deCONZ::ZclFrame&)’:
de_web_plugin.cpp:3854:17: error: expected ‘}’ before ‘else’
                 else if (ind.clusterId() == ONOFF_CLUSTER_ID && sensor->manufac
                 ^
de_web_plugin.cpp:3858:25: error: ‘attrId’ was not declared in this scope
                     if (attrId == 0x0000 && dataType == 0x10 && // onoff attrib
                         ^
de_web_plugin.cpp:3858:45: error: ‘dataType’ was not declared in this scope
                     if (attrId == 0x0000 && dataType == 0x10 && // onoff attrib
                                             ^
de_web_plugin.cpp:3859:49: error: ‘pl3’ was not declared in this scope
                         buttonMap->zclParam0 == pl3)
                                                 ^
de_web_plugin.cpp:3901:25: error: ‘attrId’ was not declared in this scope
                     if (attrId == 0x0055 && dataType == 0x21 && // Xiaomi non-s
                         ^
de_web_plugin.cpp:3901:45: error: ‘dataType’ was not declared in this scope
                     if (attrId == 0x0055 && dataType == 0x21 && // Xiaomi non-s
                                             ^
de_web_plugin.cpp:3902:49: error: ‘pl3’ was not declared in this scope
                         buttonMap->zclParam0 == pl3)
                                                 ^
de_web_plugin.cpp: At global scope:
de_web_plugin.cpp:4298:5: error: expected unqualified-id before ‘if’
     if (sensor->modelId().startsWith(QLatin1String("ZBT-CCTSwitch-D0001"))) // 
     ^
de_web_plugin.cpp:4303:5: error: expected unqualified-id before ‘if’
     if (checkReporting && sensor->node() &&
     ^
de_web_plugin.cpp:65:13: warning: ‘MaxGroupTasks’ defined but not used [-Wunused-variable]
 static uint MaxGroupTasks = 4;
             ^
Makefile.Release:626: recipe for target 'release/de_web_plugin.o' failed
make[1]: *** [release/de_web_plugin.o] Error 1
make[1]: Leaving directory '/home/andysaab/deconz-rest-plugin'
Makefile:38: recipe for target 'release' failed
make: *** [release] Error 2

Smanar, 2020-09-10:
Corrected and compiled, to it work this time.

andysaab, 2020-09-10:
> Corrected and compiled, so it work this time.
Now seems working!!!
I think need change button events for Double click and Hold, for rising value.

Log for first button (other the same):
First button One click:

00:23:36:300 APS-DATA.indication from child 0xB86F
00:23:36:300 verify 0xbc33acfffe6be774 is child node after 37 s
00:23:36:300 button 1002 B1 short
00:23:36:301 Websocket 192.168.1.96:36776 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-09-10T21:23:36.300"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
00:23:36:301 Websocket 192.168.1.96:36818 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-09-10T21:23:36.300"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
00:23:36:303 discard sensor state push for 2: state/lastupdated (already pushed)
00:23:36:305 verify neighbor status: APP_SUCCESS (0x00) 
00:23:36:330 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
00:23:36:386 MAC Poll 0x02 0xB86F
00:23:36:487 MAC Poll 0x02 0xB86F
00:23:36:584 MAC Poll 0x02 0xB86F
00:23:37:306 APS-DATA.indication from child 0xB86F
00:23:37:306 BUTTON 1002 B1 SHORT
00:23:37:307 Websocket 192.168.1.96:36776 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-09-10T21:23:37.306"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
00:23:37:308 Websocket 192.168.1.96:36818 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-09-10T21:23:37.306"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
00:23:37:309 discard sensor state push for 2: state/lastupdated (already pushed)
00:23:37:397 MAC Poll 0x02 0xB86F
00:23:37:495 MAC Poll 0x02 0xB86F
00:23:38:403 MAC Poll 0x02 0xB86F
00:23:39:210 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
00:23:39:823 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
00:23:39:832 Master: read param with arg 0x19
00:23:39:844 Current channel 25
00:23:39:849 CTRL got nwk update id 0
00:23:39:859 Device TTL 2442 s flags: 0x7
00:23:42:088 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000

First button Double click:

00:24:38:711 APS-DATA.indication from child 0xB86F
00:24:38:711 verify 0xbc33acfffe6be774 is child node after 62 s
00:24:38:711 button 1004 B1 double
00:24:38:712 Websocket 192.168.1.96:36776 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1004,"lastupdated":"2020-09-10T21:24:38.711"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
00:24:38:712 Websocket 192.168.1.96:36818 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1004,"lastupdated":"2020-09-10T21:24:38.711"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
00:24:38:714 discard sensor state push for 2: state/lastupdated (already pushed)
00:24:38:714 verify neighbor status: APP_SUCCESS (0x00) 
00:24:38:798 MAC Poll 0x02 0xB86F
00:24:38:901 MAC Poll 0x02 0xB86F
00:24:38:004 MAC Poll 0x02 0xB86F
00:24:39:680 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
00:24:39:718 APS-DATA.indication from child 0xB86F
00:24:39:718 BUTTON 1004 b1 DOUBLE
00:24:39:720 Websocket 192.168.1.96:36776 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1004,"lastupdated":"2020-09-10T21:24:39.718"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
00:24:39:720 Websocket 192.168.1.96:36818 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1004,"lastupdated":"2020-09-10T21:24:39.718"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
00:24:39:722 discard sensor state push for 2: state/lastupdated (already pushed)
00:24:39:810 MAC Poll 0x02 0xB86F
00:24:39:823 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
00:24:39:833 Master: read param with arg 0x19
00:24:39:844 Current channel 25
00:24:39:849 CTRL got nwk update id 0
00:24:39:858 Device TTL 2382 s flags: 0x7
00:24:39:913 MAC Poll 0x02 0xB86F
00:24:40:013 MAC Poll 0x02 0xB86F
00:24:40:819 MAC Poll 0x02 0xB86F

First button Hold:

00:25:09:807 APS-DATA.indication from child 0xB86F
00:25:09:807 verify 0xbc33acfffe6be774 is child node after 31 s
00:25:09:807 button 1003 B1 long
00:25:09:808 Websocket 192.168.1.96:36776 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1003,"lastupdated":"2020-09-10T21:25:09.806"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
00:25:09:808 Websocket 192.168.1.96:36818 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1003,"lastupdated":"2020-09-10T21:25:09.806"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
00:25:09:810 discard sensor state push for 2: state/lastupdated (already pushed)
00:25:09:812 verify neighbor status: APP_SUCCESS (0x00) 
00:25:09:826 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
00:25:09:893 MAC Poll 0x02 0xB86F
00:25:09:996 MAC Poll 0x02 0xB86F
00:25:10:097 MAC Poll 0x02 0xB86F
00:25:10:817 APS-DATA.indication from child 0xB86F
00:25:10:817 BUTTON 1003 b1 LONG
[... cut ...]
00:25:10:818 Websocket 192.168.1.96:36818 send message: {"e":"changed","id":"2","r":"sensors","state":{"buttonevent":1003,"lastupdated":"2020-09-10T21:25:10.816"},"t":"event","uniqueid":"bc:33:ac:ff:fe:6b:e7:74-01-0006"} (ret = 164)
00:25:10:819 discard sensor state push for 2: state/lastupdated (already pushed)
00:25:10:904 MAC Poll 0x02 0xB86F
00:25:10:005 MAC Poll 0x02 0xB86F
00:25:11:108 MAC Poll 0x02 0xB86F
00:25:11:367 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
00:25:11:915 MAC Poll 0x02 0xB86F

Smanar, 2020-09-12:
Its  not good ?

You have 1002 1003 and 1004 ?

andysaab, 2020-09-12:
> Ha ^^.
> Is not me that choose value.
> It s something normalized by deconz https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#wireless-switches

Ок, clear.
No problem, let’s keep it that way. Really the state Hold for this remote mean Release after hold. So I agree with you. Thank you, Smanar!

closed 2020-10-01 by manup

MattL0, 2022-01-14:
I have the one button version. Is it supposed to work with deconz ? it is seen as a light + a switch.  But receive no events from on nodered exept the first one describing the device... I tried vnc but i can't zoom, i use to be able to zoom. 
[image]
[image]
[image]

MattL0, 2022-01-14:
[image]

MattL0, 2022-01-14:
This is the json from jowihue plugin from HomeSeer
[image]

MattL0, 2022-01-14:
[image]

Smanar, 2022-01-15:
Strange it was reconized in the API, this device is not in the whitelist

"modelids": ["_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0"],

[2 more replies]