dresden-elektronik/deconz-rest-plugin #3222: Can't delete device that is not in phoscon through VNC

fribse, 2020-09-01:
## Describe the bug
I wish to add my Kadrilj blind to deconz. I already have the repeater and the controller connected. The cover is showing up in the VNC interface, but not in the deconz.
If I delete it throught the VNC interface, and then restart deconz it reappears by itself, so I can not readd it.

## Steps to reproduce the behavior
  1. Delete a device that is only showing in VNC
  2. Restart Deconz
  3. Open VNC, the device is still there

This makes it not possible to readd the cover.

## Expected behavior
If I delete it in VNC, it should disappear.

## Screenshots

## Environment
 - Host system: Intel NUC with PVE 6.x
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26580700)
 - deCONZ version: (2.05.79)
 - Device: ConBee II
 - Do you use an USB extension cable: yes, USB3

### deCONZ Logs

## Additional context

ebaauw, 2020-09-01:
Deleting a node from the map in the GUI is just that.  The device remains joined to the network, and te GUI will re-display the node when it sees the next message from it.  Also the REST resources for the device are still there.  See #2377.

You need to delete the resources through the API, while specifying a body of `{"reset": true}`.  This might not work for battery-powered devices; in that case, factory-reset the device to force it to forget the network.

fribse, 2020-09-02:
I see, ok, I'll try to get that done. As it's a HA addon, I've started reading the REST API docs, I'm a bit unsure on how to start the 'webapp' to get the apikey?

fribse, 2020-09-02:
I see, ok, I'll work my way through the API documentation, install a REST client in my browser and see if I can obtain the apikey to reattempt a delete of the blind. I really hope it works, it would be silly if not 😄

fribse, 2020-09-02:
Ok, I got the API key, and I'm trying to figure out the <id> that the docs are referring to.
Is that the long hexadecimal number?
And it's just sending a 
GET http://IP:port/api/key/lights/D0CF5EFFFEEA2305
?
When doing that I just get
[{"error":{"address":"/lights/D0CF5EFFFEEA2305","description":"resource, /lights/D0CF5EFFFEEA2305, not available","type":3}}]

fribse, 2020-09-02:
Ok, if I just do
GET http://IP:port/api/key/
I get

`
"47":{"config":{"on":true,"reachable":true},"ep":1,"etag":"7628599f608067dae935738e24bba20a","lastseen":"2020-09-02T19:29:24.722","manufacturername":"IKEA of Sweden","modelid":"KADRILJ roller blind","name":"KADRILJ roller blind ","state":{"battery":0,"lastupdated":"2020-09-02T19:27:27.781"},"swversion":"2.2.009","type":"ZHABattery","uniqueid":"d0:cf:5e:ff:fe:ea:23:05-01-0001"}`

So I guess it's not registered as a light, since if I add lights, it doesn't show up in the list?
How do I then delete it, and what's the ID, is it the 47 or the uniqueid without : and ending date?

ebaauw, 2020-09-02:
There should be a `/lights` resource in addition to the `ZHABattery` `/sensors` resource.

I always use the resource ID (`47` for the `ZHABattery` above) instead of the mac address for the resource.  I don’t know why the mac address isn’t working.

fribse, 2020-09-03:
There isn't a light ressource with the same MAC address, and I guess that's the problem, for some reason, only the battery is there, and so when it attaches, it is just recognized as the existing ressource, a battery, and will not allow it to also add the cover itself?
So I guess deleting the battery sensor could help here?

Mimiix, 2020-09-03:
@fribse In response of the question i asked Manuel, he confirmed this "button" will be changed to the expected behaviour.

ebaauw, 2020-09-03:
If you search for new devices from Phoscon and then read the _Simple Descriptor(s)_ and/or _Basic_ cluster attributes in the GUI, the `/lights` resource should be created.  That is, assuming the blind is still in the network.  If not, reset the blind using the physical buttons (see manual) and re-pair it.

fribse, 2020-09-03:
I got everything deleted with the API, I found a few odd things here and there.
After that, as soon as I paired it it showed up as a window cover!
Now it's just not updating the position of the cover?
Which means that the 'down' and 'stop' buttons in HA are active, but not the 'up' button.
The slider showing how much it is open is not updating, but moving the slider moves the cover.

Mimiix, 2020-09-03:
@fribse use the open_cover ,close_cover services and the set_cover service.  The sliders are buggy.

ebaauw, 2020-09-03:
Check that attribute reporting for _Current Position Lift Percentage_ has been configured, using the _Attribute Editor_ in the GUI.  If not create a binding from the _Window Covering_ cluster to the gateway.  If it is, but still doesn’t report, power cycle the blind (remove the battery for 10s).

fribse, 2020-09-04:
@ebaauw Ok, I found the Attribue Editor, did a read a few times, and it did show the Current Position Lift Percentage there, so I powercycled the cover, and now it updates correctly.
Oddly now the cover doesn't show up in the list of devices from phoscon in HA now, but it is in the complete device list, weird, but that's a whole other problem I guess.
Thankyou for helping out @Mimiix and @ebaauw from the original problem of a cover that couldn't be removed to getting it working, I'm feeling much more comfortable with adding the other blinds now, and finally getting rid of the IKEA hub.

fribse, 2020-09-08:
Ok, position problem with the second blind added, if I press the 'read' for the WIndows Covering Cluster' it updates the position.
I can set the lift percentage and execute on that as well.
But the deconz integration to HA does not update the percentage until I press read for the cluster?
Also the naming is odd. Looking in Phoscon App, the 'light' is called 'Bedroom Blind'.
But looking in HA the name is different:
[image]
The name in phoscon and HA should be the same?
I've tried removing the blind, and the battery, and that just left an empty 'Fyrtur' without a blind and battery status, and then when readding it, it went into the same 'Fyrtur'?
I really wouldn't like to remove the deconz integration and add it back, as I fear that would mess up a lot of names, but is that the only way to clear this?

ebaauw, 2020-09-08:
Cannot help you with HA.

I think the issue is with the blind not honouring the attribute reporting settings, either because the API didn’t set them up correctly, or because the blind just doesn‘t feel like it.  In both cases, power cycling the blind (just remove the battery for 10s) usually remedies the situation.  If not, double-check the attribute reporting settings in the GUI.

fribse, 2020-09-10:
Hmm, so frustrating. I've tried to remove the blind (by using Phoscon), and readding it, no change.
I've done the 'read' of attributes, and it updates it nicely, and when I press read, the position in attributes and HA is updated, but it doesn't do it by itself.
I've tried removing the battery for 10s, but that didn't change anything.
Where do I see 'attribute reporting settings'?

ebaauw, 2020-09-10:
Double-click on the _Current Position Lift Percentage_ to popup the _Attribute Editor_ window.  See the _User Manual_ under _Help_ for details.

fribse, 2020-09-11:
Found it, thankyou.
The blind is connected to the nearby repeater, so that part is ok.
[image]
But it seems to be sort of offline, it doesn't react on commands from the button.

If I press read for the attributes, the blind flashes blue in VNC and updates the values.

But if I open the editor and presses read, it fails, and the same happens when I try to write.
[image]
[image]

ebaauw, 2020-09-11:
And you can still read the attribute?  Odd.  Maybe the binding needs to be created first.  In the _Bind dropbox_, drag the _Window Covering_ cluster to the _Input_ and endpoint 0x01 of the coordinator to the _Outout_ dropbox.  Then press _Bind_.

If you cannot read the attribute either, the blind has become unreachable.  Do you see any errors in the deCONZ output?

fribse, 2020-09-11:
I just want to be sure so it's this one:
[image]
and this one:
[image]
Which makes it look like this:
[image]
?

fribse, 2020-09-11:
I removed and repaired it today.
I can now write values in the attribute editor for the fyrtur blind. It still doesn't update the position, but I can give it a position and it will go there.
And now the kadrilj blind disconnected, and doesn't react, without me doing anything, not even restarting deconz/phoscon, geeez, it looks like it simply isn't stable.

ebaauw, 2020-09-11:
> Which makes it look like this:

That looking good.

> it looks like it simply isn't stable.

No it’s not; I lose my Fyrtur on average once every week.  Power cycling the blind usually fixes this, if not, open the network and power down suspicious routers, before power cycling the blind.

fribse, 2020-09-12:
They seem to fall asleep over night. I guess deconz needs some special handling of these devices for them to be usable.
This morning none of the buttons worked, until I had pressed them a couple of times, then they sprung into action.
I'm a bit unsure of the WAF here, or if I should just move them back to the IKEA hub. They were more stable there, albeit that the IKEA hub is not that stable in itself.

ebaauw, 2020-09-12:
I think there’s a number of issues:
1. On larger networks, the gateway device seems to lose the route to the blind.  The blind still reports its state, but (unicast) commands by the gateway don’t reach the blind.  I’m not sure if this is because the blind’s parent disowned it and the blind doesn’t notice, or because of generic routing issues.  I moved my blind to another network without additional routers to find out.  The workaround here is to power cycle the blind.
2. The blind firmware sometimes stops reporting the current position.  Could be by design, when the blind no longer receives ACK, due to 1).  Could also be a bug.  Power cycling the blind remedies this.
3. The controller sends a group command using the _All devices_ broadcast address.  Normally, group commands are sent to the _Receiver on when idle_ broadcast address.  I suspect some parents don’t handle the _All devices_ broadcasts well.  I fear the RaspBee/ConBee might actually be one of those devices.  It takes forever for the blind to react to the controller on my router-less network.
4. deCONZ nor the API can issue group commands to the _All devices_ broadcast address.  I’ve asked @manup to add support, as I think this requires changes in the core and/or firmware.  This means that currently it’s not possible to use group commands as workaround for the routing issues.
5. The blind deviates from the Zigbee standard.  It stops moving when it receives an _Open_ or _Close_ command while opening or closing.  This is probably to accommodate the open/close remote.  But is does break the idempotency of these commands.  deCONZ typically resends a unicast command when it doesn’t receive an ACK in time, causing the blind to stop.  As workaround use `state.lift`, which sends a _Go to lift percentage_.

fribse, 2020-09-12:
That is a lot of good info, you have way more insights into the zigbee protocol than I have, but the points seems to fit the problems.
It's interesting that these problems doesn't show in the IKEA hub. As I have primarily IKEA lights in the zigbee network, which should act as routers, then they at least internally should handle the disownement I guess.
I did notice the 5) and was wondering if that was per design, or just my network acting odd, but good to know that it's by design, albeit wrong design :-)
I'll close this for now, thankyou very much for your help.

closed 2020-09-12 by fribse