dresden-elektronik/deconz-rest-plugin #3225: Tuya Window covering DT82LEMA-1.2N

czm97176, 2020-09-02:
## Device

- Product name: DT82LEMA-1.2N
- Manufacturer: _TYST11_xu1rkty3
- Model identifier: u1rkty3
- Device type : Please remove all unrelated device types. 
  - Other: Window Convering

## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]

[image]

czm97176, 2020-09-02:
Update: By adding this model into 'owvfni3' on zigbee2mqtt. i am able to play with this motor. Please take a look.

However, i don't have rail on hand so i don't know what is means of 'position' or 'calibration` and seems the state is always showing 'running' even i pressed stop. i know it might zigbee2mqtt issue, so please ignore if you feel don't applicable to deconz. 

https://github.com/Koenkk/zigbee-herdsman-converters/issues/1159

Smanar, 2020-09-02:
I m ATM on tuya covering > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3161
I have just added your device, so you can test the code (open/close/stop is working) but there is a problem yet for device detection so you need to edit the file de_web_plugin.cpp to force detection

```
        //Disabled for this deconz version
        //if (node->nodeDescriptor().manufacturerCode() == VENDOR_NONE)
        //{
        //    lightNode.addItem(DataTypeBool, RStateOpen);
        //    lightNode.addItem(DataTypeUInt8, RStateLift);
        //    lightNode.addItem(DataTypeUInt8, RStateBri);
        //}
```

just remove all "//"

czm97176, 2020-09-03:
> 
> 
> I m ATM on tuya covering > #3161
> I have just added your device on the code, so you can test it (open/close/stop is working) but there is a problem yet for device detection so you need to edit the file de_web_plugin.cpp to force detection
> 
> ```
>         //Disabled for this deconz version
>         //if (node->nodeDescriptor().manufacturerCode() == VENDOR_NONE)
>         //{
>         //    lightNode.addItem(DataTypeBool, RStateOpen);
>         //    lightNode.addItem(DataTypeUInt8, RStateLift);
>         //    lightNode.addItem(DataTypeUInt8, RStateBri);
>         //}
> ```
> 
> just remove all "//"
> 
> For setLevel, the code is ready but untested yet.

Hi Smanar,

thank you for your help! i am really appreciated for your great help. 

As i am assuming the changes you made that are not released to docker at this moment, so i tried to duplicate the docker and build the "master" version on top of it. i have replaced the libde_rest_plugin.so following by ` qmake && make -j2` but nothing change. it's still showing as `unknown device`.

 when i see your commit change, it's seems there is no coding mentioned u1rkty3 (my device code). Would you please confirm u1rkty3 is already added to the master branch?

czm97176, 2020-09-03:
Sorry, i just figure out that i should use 

git clone --branch tuya2 https://github.com/Smanar/deconz-rest-plugin.git

let me rebuild everything and try.

Thanks

czm97176, 2020-09-03:
Update:

i did the same in branch tuya2 and replace the libde_rest_plugin.so. 
No luck. it doesn't appear in my home assistant.
i tried following things

Remove / Re-add in deCONZ
Re-read basic attribute in deCONZ
Add as light in Phoscon APP
Reboot
Restart deCONZ & Home Assistant

May i know how do i expose this device to Home Assistant ? Thanks

[image]

Mimiix, 2020-09-03:
@czm97176 It should be in the rest api.

czm97176, 2020-09-03:
> @czm97176 It should be in the rest api.

Hi Mimiix, Sorry i don't get it. 
i would like to expose this motor to home assistant. Would you please give me more hints?  Thank you

btw, @Smanar , the issue you mentioned https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3161 is a Tobular Motor and mine one is rail track motor, do they the same in program level?

czm97176, 2020-09-03:
Update

@Mimiix , got your point , just figure out how to look into the api. However, the devices doesn't appear at the api.

[image]

Smanar, 2020-09-03:
Oups, sorry my bad, haven't see your device is battery one, the code work only for powered one, I will update the code tonight.

Mimiix, 2020-09-03:
@Smanar Next time tag me, because you need to delete the revisions. Also: always add a note in the edit ;)

Smanar, 2020-09-03:
Ok so long story, there is a problem n deconz ATM, it don't ask for model id before creating a lightnode.
So need to reconize the device only with few information, and it's not possible for tuya, so need to improve the code, but we can force the code to make the motor react (good for spirit having something working first ^^)

BTW I have updated the code on the fork to test your device first, so the new code to enable, to force the detection is on de_web_plugin.cpp line 1796

```
        //if (node->nodeDescriptor().manufacturerCode() == VENDOR_NONE)
        //{
        //    hasServerOnOff = true;
        //}
```

With this code you will have your device in light section in phoscon (and surely visible on phoscon), like before just remove the "//"

czm97176, 2020-09-04:
@Smanar Thanks! it's my bad i didn't highlight that is a battery one and it's a rail track type. 

i have tested your latest changes ! it works and appear as lightnode and i am able to start the motor in home assistant but there are two things i am aware of ... not sure it's expected during the test. Please take a look

1. a Dummy light node is created when i am adding the motor, it seems belongs to my light strip. i tried to delete this dummy one, it will delete the actual one together.

[image]

2. The motor is showing as on/off light in Home Assistant, i am only able to kick on the motor , but not stop it. i expect there should have 3 buttons or actions for a curtain motor ? - open / close / stop ? 

[image]

Thank you so much! 
Gordon

Smanar, 2020-09-04:
I don't think it s a dummy node, one unique id is finishing by 0B and the other by 0D, I think your device realy have 2 endpoint but have "basic" cluster only on the 0B one.

What does mean "kick" the motor ?

Can you expend the state field ? it s the more important one, you need to have inside "lift" "open" or more.

czm97176, 2020-09-06:
HI Smanar,

i just tried to revert back to the normal branch - it will only show 1 light instead of two in phoscon app 
i am not sure why it has a dummy light for this branch . 
Attached is the jsons file of the API in this two branch
please take a loook

**Tuya2 Branch**

[tuya2_branch.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5179353/tuya2_branch.json.txt)

**Normal Branch**

[Normal_Branch.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5179344/Normal_Branch.json.txt)

Regarding the window conver, there is no "open" "close" "stop" in HA. 

[image]

[image]

However, in the zigbee2mqtt , it will allow me to do this and treat it as "cover" object not switch 
is it related to the API ? i am not sure how to make HA threat it as cover like zigbee2mqtt does.

[image]

Smanar, 2020-09-06:
Take a look in deconz to check the device "00:12:4b:00:21:13:f8:4f" to count the endpoint.

BTW for the moment works fine, the device is not detected on old code but you have all needed field on the tuya2.

>"etag":"b1e09f7de02fdd4c399e7acafad47ac6","hascolor":false,"lastannounced":"2020-09-06T08:57:01Z","lastseen":"2020-09-06T09:23Z","manufacturername":"_TYST11_xu1rkty3","modelid":"u1rkty3","name":"窗簾","powerup":7,"state":{"alert":"none","bri":0,"lift":0,"on":false,"open":false,"reachable":true},"swversion":null,"type":"On/Off light","uniqueid":"ec:1b:bd:ff:fe:68:80:15-01"}

For the moment, to test the device you need to use directly the API
```
curl -H 'Content-Type: application/json' -X PUT -d '{"open": true}' http://IP:PORT/api/API_KEY/lights/2/state
curl -H 'Content-Type: application/json' -X PUT -d '{"open": false}' http://IP:PORT/api/API_KEY/lights/2/state
```

Set level is workinh too now

I can try to force the "type" but not sure others devs will be happy with that.

czm97176, 2020-09-06:
Hi @Smanar , 

i just updated your code to the last commit version - 160a3d7.  The cover devices is missed in phoscon, the api as well.
but it's still visible in deCONZ. i tried to reset / re-join / delete node  so many time but no luck.
Would you please check ? Please let me know if i need to provide any information

[image]

[6F18D4EFD6.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5179639/6F18D4EFD6.json.txt)

Gordon

Smanar, 2020-09-06:
Ok, sorry was my fault, I m on another covering in same time, and have removed all hacks, so there is missing the one for your device now (remember the line you have edited)

So new version to test, with your version re-added and without hack.

czm97176, 2020-09-06:
@Smanar 

Sorry for the back and forth, i just updated to e9aaaee. it's still not able to show in phoscon as light.

Smanar, 2020-09-06:
Can you make tries without deleting the device.
Or using the deconz trick
- set phoscon in permit join (add new light)
- go on basic clustaer 0x0000
- read attributes (need perhaps wake up the device in same time)

czm97176, 2020-09-06:
@Smanar 

WOW. it just refreshed by clicking the white box on "READ SIMPLE DESCRIPTOR / READ NODE DESCRIPTOR"

[image]

Thanks a lot !!!.

so by asking to change the device type, How can i ask for other Dev opinion? i google and see there has a attribute call  "resourcelinks" , can it help ? i am not very sure .

Smanar, 2020-09-06:
>WOW. it just refreshed by clicking the white box on "READ SIMPLE DESCRIPTOR / READ NODE DESCRIPTOR"

And phoscon was in permit join at this moment ?

For the type, it s a long story, it s easy to do on my side, but it result fake information in the API, on some other devs realy don't like that.
The problem is without this "hacks" the device can't work natively in third application.

I have updated the code with the hack, so you will be able to test the device with HA (the device type will be "Window covering device")

But
- I can remove the hack if someone is against. (I m using so much hack for tuya ...)
- I have updated the code 10 mn ago with removing a hack that have an impact on inclusion, it works fine on router, but idk for battery device, you probably need to use the same method than previously.

And on previous issue, you say it s working, but you can see the device on phoscon ? And you still haven't tried the command ?

czm97176, 2020-09-06:
> 
> 
> > WOW. it just refreshed by clicking the white box on "READ SIMPLE DESCRIPTOR / READ NODE DESCRIPTOR"
> 
> And phoscon was in permit join at this moment ?

Yes, actually i press "add light" as the same time, not sure which of the command are the key of success.

> 
> For the type, it s a long story, it s easy to do on my side, but it result fake information in the API, and some other devs realy don't like that, it s not something clean.
> The problem is without this "hacks" the device can't work natively in third application.
> 
> I have updated the code with the hack, so you will be able to test the device with HA (the device type will be "Window covering device")
> 
> But
> 
>     * I can remove the hack if someone is against. (I m using so much hack for tuya ...)
> 
>     * I have updated the code 10 mn ago with removing a hack that have an impact on inclusion, it works fine on router, but idk for battery device, you probably need to use the same method than previously.
> 
> 
> And on previous issue, you say it s working, but you can see the device on phoscon ? And you still haven't tried the command ?

i see the device on phoscon, i didn't tried the command. i just try to toggle the cover by script in HA (even in wrong type)
i don't have the track and remote on hand, so i don't have the motion in full picture but motor can spin on both direction. However, i am still not able to stop the cover. 
 
[image]

i will try your updated code very soon. Will let you know once it done.

czm97176, 2020-09-06:
@Smanar 

Quick update, i can't compile the program in latest version.

[image]

Smanar, 2020-09-06:
Corrected.

But you can add this device to HA even the type is not "window covering" ?

Yeah for stop command, try directly the api command 
`curl -H 'Content-Type: application/json' -X PUT -d '{"stop": true}' http://IP:PORT/api/KEY/lights/ID/state`

From my memory I have already see this problem for HA.

czm97176, 2020-09-06:
> 
> 
> Corrected.
> 
> But you can add this device to HA even the type is not "window covering" ? So you don't need the hack ?
> 
> Yeah for stop command, try directly the api command
> `curl -H 'Content-Type: application/json' -X PUT -d '{"stop": true}' http://IP:PORT/api/KEY/lights/ID/state`
> 
> From my memory I have already see this problem for HA.

As i hosted HA and Home Bridge (homebridge-hue with plugin )...
i don't know how to "override" the device type in both platform and it should support "position" as well.
i need this hack because deCONZ is expose it as "switch" , i don't have proper user interface to open close and stop the motor.
as it works on zigbee2mqtt out of the box, so i really don't know how to do with deconz. 
i want it in deCONZ because it's should more reliable.  

https://www.zigbee2mqtt.io/devices/TS0601_curtain.html

Anyway, thanks for your great work !! i try will the new code ! thanks a lot ~

czm97176, 2020-09-06:
@Smanar , Sorry , i got failure in compiling source code again .

[image]

Smanar, 2020-09-06:
Corrected and tested this time, it compile.

Tell me if it s better. and I think position is working too (it work on another device) but need calibration, and idk for your device.

czm97176, 2020-09-06:
@Smanar 

Yeah. it's working now. Thank you so much! i would like ask three more question.

1. Does this devices has battery information in the payload ? is it possible ?
2. The API does not reflect the status to close   (open: false)  or open (open: true) after i executed comment and wait the motor spinning has finished. it's always showing open = false.

`curl -H 'Content-Type: application/json' -X PUT -d '{"open": true}' http://IP:PORT/api/API_KEY/lights/2/state
 curl -H 'Content-Type: application/json' -X PUT -d '{"open": false}' http://IP:PORT/api/API_KEY/lights/2/state`

this is the first time i bought this curtain motor. 
Does it require the rail/tracker installed and calibrated by remote to get the open/close status ?

3. As you said this hack may not up forever, may i know will this change merge to the official branch and how do i know they will drop the override hack if they are not happy with it ? Can i have work around ? 

Thanks
Gordon

Smanar, 2020-09-06:
1- It possible yes, but will be in another entriy, we can't use battery for "light", will take a look tomorrow if I can find the magic tuya command on the net (it s the first battery powered covering I m seing), else if it exists, it will be visible using debug mode.
2- Yep I think it s probably because of the missing calibration, because I m using the same command for another device without problem, and setlevel command too need calibration.
3 - I will ask tommorow to have more opinions, the next version will be out in less than 10 days, so I will try to make something the more clean I can in the next days.

ebaauw, 2020-09-06:
> it s the first battery powered covering I m seing

Check out the IKEA blinds and the Xiaomi B1.  I create ZHABattery for these as new sensor resource type.

czm97176, 2020-09-07:
> 2- Yep I think it s probably because of the missing calibration, because I m using the same command for another device without problem, and setlevel command too need calibration.

Thanks. i will report once it installed on track and calibration. 
btw, i see the lift is changing from 0 to 100 or from 100 to 0 when i execute the open:true/false command. 
Maybe it use the "lift" to determine it's open/close state instead  ? Not sure

[image]

On the other hand, i see there has a command to allow "reverse_direction" on similar tuya curtain in zigbee2mqtt,
Can we have the same options as well ?

Thank you so much

Smanar, 2020-09-07:
Ha ?
Good catch, there is a  problem here, yes lift can be used and it is from my memory, need to take a look. Will have a new version today.

For the reverse command, I can use it as hidden command
Using for exemple.
`PUT, PATCH /api/<apikey>/lights/<id>`

But it s something usefull ? you have 2 mounting possible ?

For the battery, I haven't find information on internet, so I will use the same tuya value than for thermostat device, if it don't work, we will need to use the debug mode to find the tuya value.

BTW ebaawn have find a probable critic bug with my inclusion system, so need to make some test on my side first with random stuff, no problem yet, but need some days to have a final version.

czm97176, 2020-09-07:
> 
> 
> Ha ?
> Good catch, there is a problem here, yes lift can be used and it is from my memory, need to take a look. Will have a new version today.
> But as "lift" is updated and not "bri" I think the problem is from tuya code.
> 
> For the reverse command, I can use it as hidden command
> Using for exemple.
> `PUT, PATCH /api/<apikey>/lights/<id>`
> 
> But it s something usefull ? you have 2 mounting possible ?
> 
> For the battery, I haven't find information on internet, so I will use the same tuya value than for thermostat device, if it don't work, we will need to use the debug mode to find the tuya value.
> 
> BTW ebaawn have find a probable critic bug with my inclusion system, so need to make some test on my side first with random stuff, no problem yet, but need some days to have a final version.

Thanks, No worry, take your time,

FYI i notice that the "lift" is changed immediately after command is execute even the motor is still spinning
In addition, when i tried to click "open 45%" in homekit, it will change the "lift" to 45 as well.

BTW, i apologize if i gave false information on "open" state checking idea.
because you maybe right.  it can report accurate after calibration. 
anyway, i need to buy a curtain track ASAP and get it installed 
so i can give you more solid finding.

Smanar, 2020-09-07:
Np the "open" field is aready corrected, bug was from me
The reverse command too

But I have problem for the battery, because all information are on the same cluster, the easier way will be using the ZHAtuya sensor, but this device is not implemented in third application.

Smanar, 2020-09-07:
Ok done it complile, a warning but don't block the compilation.

The "open" is corrected
To change direction use PUT /api/<apikey>/lights/<id> {"reverse":true}
The device will have a new zhabattery device, but not sure it will work, I m using the same one than for thermostat sensor.

As I don't find information on the net, the only solution is running deconz in debug mode, --dbg-error=2, all other ==0.
And you will see in log 
>Tuya debug 5 ........................................

czm97176, 2020-09-07:
> 
> 
> Ok done it complile, a warning but don't block the compilation.
>
> The "open" is corrected

i just updated to the latest version. the open is working now , thanks
so by the current information , this devices has no "Current Position" status but only "Target position" ?

> To change direction use PUT `/api/<apikey>/lights/<id>` with {"reverse":true}
> The device will have a new zhabattery device, but not sure it will work, I m using the same one than for thermostat sensor.
 
Regarding the battery status, there is no obvious change in the API. Please take a look.
[6F18D4EFD6.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5183551/6F18D4EFD6.json.txt)

> As I don't find information on the net, the only solution is running deconz in debug mode, --dbg-error=2, all other ==0.
> And you will see in log
> 
> > Tuya debug 5 ........................................

As i am using docker on deconz... it don't provide dbg-error option (i tried DEBUG_INFO = 2 , but no luck) ...
not sure how can i activate debug in docker mode ....ahhhhh

[image]

> Need to wait for the one used for battery. Take a look on the tuya app, some device haven't battery level but just an alarm for low battery (not the same)

i don't get it in this part, What is mean of "Need to wait for the one used for battery. Take a look on the tuya app" ? 
you mean someone has tuya gateway?

Smanar, 2020-09-07:
>so by the current information , this devices has no "Current Position" status but only "Target position" ?

Yep I think the target is more usefull than the current, but in reality I think "bri" is updated during the move.

You need to re-include the device for the battery device creation.

Don't worry, if you haven't the tuya app or docker ^^, it will be just harder.
Just enable logging by defaut, and hoping to see the good value, you can try to power cycle the device too (if it s possible without removing battery). The problem is we don't now when the device will send the report, can be 1 time every hour or every week (or never)

czm97176, 2020-09-07:
> 
> 
> > so by the current information , this devices has no "Current Position" status but only "Target position" ?
> 
> Yep I think the target is more usefull than the current, but in reality I think "bri" is updated during the move.
> 
> You need to re-include the device for the battery device creation.

i tried to delete/rejoin/reset/read descriptor/read cluster. i did what i have learn above.  
May i know what is the process of re-include? Please and Thanks 

[image]

> Don't worry, if you haven't the tuya app or docker ^^, it will be just harder.
> Just enable logging by defaut, and hoping to see the good value, you can try to power cycle the device too (if it s [...]

By looking the log information , What kind of wording that might relevant to it ?

Smanar, 2020-09-07:
The device for battery will be in "sensor" API section, and I have just corrected something 5mn ago, I think it will be fine now.

On log you will see 
>Tuya debug 5 : Status: ...................................

It s all the informations send by the tuya device, you will have them when you use the device and during periodic report. and if the device send battery it will use this way too.

czm97176, 2020-09-07:
> The device for battery will be in "sensor" API section, and I have just corrected something 5mn ago, I think it will be fine now.
>
> On log you will see
>
>
> It s all the informations send by the tuya [...]

Thanks you, Sorry i got this error in compile program 

[image]

Smanar, 2020-09-07:
oups sorry, I m making dinner in same time.
Updated.

czm97176, 2020-09-07:
> 
> 
> oups sorry, I m making dinner in same time.
> Updated.

Opps. Sorry to disturbing you..... 

i just updated to 1d93f36 and tried to re-include the device .... but i still can't find any thing mention battery
Attached is the API file
[6F18D4EFD6.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5184228/6F18D4EFD6.json.txt)
 
please take your time, no hurry , take care & stay healthy

Smanar, 2020-09-07:
Lol, np, it s just I have done the modification without testing.

But BTW I don't understand why it s not working, I have the same issue on another tuya device ATM, device is not created.

I have made some change to test, but if with them it s still not working, can you enable log during inclusion ? Only --dbg-info=2, you can set other = 0.

czm97176, 2020-09-08:
> Lol, np, it s just I have done the modification without testing.
>
> But BTW I don't understand why it s not working, I have the same issue on another tuya device ATM, device is not created.
>
> I have [...]

i have 1 thing i am quite aware of .... this device will disappear in the API and phoscon every time i restart the deconz 
in fact, this device is changed as a "unknown device". 
i don't need to re-pairing this but i have to click the "read cluster/read node descriptor" to make it back to normal
i don't have this issue on my light strip and those 4 sensor. 
Would you please take a look?

[image]

Smanar, 2020-09-08:
What is this device ? It s a tuya switch/remote ? You have the basic attribute ?

You don't need to set phoscon in permit join in same time ? to make it appear ? (To know if the api can include new device).

>this device will disappear in the API and phoscon every time i restart the deconz

You mean the name disapear, not the node ? It can be just the device is badly included in the API. Or not included at all.

BTW the battry sensor is still invisible in the API ?

czm97176, 2020-09-08:
> 
> 
> What is this device ? It s a tuya switch/remote ? You have the basic attribute ?

i mean the Tuya window cover.

> You don't need to set phoscon in permit join in same time ? to make it appear ? (To know if the api can include new device).

i don't need to permit join to make it appear , Every time i restarted the deCONZ,  Tuya window cover appear in deCONZ (the name in Hex Code) but not in the API and Phoscon. Once i press "read cluster" and "read node descriptor" , the name will change to On/Off light OR the name i set in phoscon before AND it will appear in API and phoscon.

basically , i have to re-read the cluster to make the Tuya right in API/phoscon every time i have restarted the deCONZ.

> > this device will disappear in the API and phoscon every time i restart the deconz
> 
> You mean the name disapear, not the node ? It can be just the device is badly included in the API. Or not included at all.

this issue only happen on the Tuya window cover, but no the sensor and light script

> BTW the battry sensor is still invisible in the API ?

Sorry... No

Smanar, 2020-09-08:
Arf, sorry haven't reconized it, I realy have problem with battery covering.

I think I have found the problem, you can see "Tuya debug 7 : Missing manufacture name" in loop in log after the restart ?
I will make a new code today, but if you can send me your log during inclusion will be usefull for the battery ?

czm97176, 2020-09-08:
> Arf, sorry haven't reconized it, I realy have problem with battery covering.
>
> I think I have found the problem, you can see "Tuya debug 7 : Missing manufacture name" in loop in log after the restart [...]

i can't see it in Tuya deug on my log , please find attached log file

[log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5188678/log.txt)

Btw, As i said tuya window cover is disapear evertime i restarted so it is not possible to make it work out VNC to control deCONZ (as i am using docker & raspberry OS without GUI  ) ...it mean that i can't use startup command `-platform minimal` ....  

so i am not sure the -dbg_error=2 can work without startup command `-platform minimal`

Smanar, 2020-09-08:
There is nothing realy usefull inside ^^

I think you can edit Environment Variables, but by defaut DEBUG_INFO=1, so I don't understand why you don't have more usefull lines. For exemple you miss all lines from deconz initialisation.

>Btw, As i said tuya window cover is disapear evertime i restarted so it is not possible to make it work out VNC to control deCONZ (as i am using docker & raspberry OS without GUI ) ...it mean that i can't use startup command -platform minimal ....

Yep I think the code is blocked, waiting for manufacture name, I trigger myself this request, but the "blocking code" in in "discovery" part, so I don't understand why this part is blocking after a restart.

czm97176, 2020-09-08:
> There is nothing realy usefull inside ^^
> 
> I think you can edit Environment Variables, but by defaut DEBUG_INFO=1, so I don't understand why you don't have more usefull lines. For exemple you miss all lines from deconz initialisation.
> 
> > Btw, As i said tuya window cover is disapear evertime i restarted so it is not possible to make it work out VNC to control deCONZ (as i am using docker & raspberry OS without GUI ) ...it mean that i can't use startup command -platform minimal ....
> 
> Yep I think the code is blocked, waiting for manufacture name, I trigger myself this request, but the "blocking code" is in "discovery" part, so I don't understand why this part is blocking after a restart.

@Smanar i got the debug message by setting debug_info=1

[image]

Smanar, 2020-09-08:
Ok I can reproduce the problem here too.
But still searching a solution.

czm97176, 2020-09-08:
> Ok I can reproduce the problem here too.
> But still searching a solution.

Thank you ! Regarding the tuya missing issue , i put a video on youtube to demonstrate how it looks and how i include it back manually . please take a look 

https://www.youtube.com/watch?v=B4TrwQU1Hc8

czm97176, 2020-09-09:
@Smanar i know you may still working on it. i have updated to the latest version and this is the debug message .
Unfortunely, the tuya is still disapear after restart and no battery information is shown. 

Anyway, thank you so much

[image]

Smanar, 2020-09-09:
Yep, I m on the code atm it s moving a lot ^^, but not succed.
I can retreive data from the database, so if you have already included the device, no more problem, but I have problem for the first inclusion, not possibe without deconz still.

czm97176, 2020-09-09:
> inclusion
Sorry, no offsense.
Just to make sure we are on the same page. What do you mean of `f you have already included the device, no more problem` ? the device still disapear every time i restart deCONZ , is this mean first inclusion ? 
btw, i guess the battery function is not possible at this moment yet ? 

Thanks
Gordon

Smanar, 2020-09-09:
Ok so i have found a "compromise"

Tuya devices will now need 2 inclusions.

The first inclusion will be bad, idk what you can have as entries. But after this one deconz store all informations in the db.
But now for tuya device the code make a search in the database with the mac adress to find a manufacture code, and if he found one, after that the inclusion will be nice.

So for you, it will be fine, because you device is already in the database after a reboot.

Problem idk with wich one frequency deconz store data.

So now you will have in your log during restart
```
18:56:00:364 Tuya debug 7 : Missing manufacture name for 0x00212effff029e9d
18:56:00:364 sql exec SELECT manufacturername FROM nodes WHERE mac LIKE '00:21:2e:ff:ff:02:9e:9d%' COLLATE NOCASE
18:56:00:364 DB SELECT manufacturername FROM nodes WHERE mac LIKE '00:21:2e:ff:ff:02:9e:9d%' COLLATE NOCASE: dresden elektronik
```

I m testing it on my side, I think it will be fine. So for the battery I need log strating with 
>Tuya debug 5 : Status: %d Transid:

And for information I know already DP for value 0x0104 0x0202 0x0203

czm97176, 2020-09-10:
HI,

i search the log and i got this from yesterday (the date before i updated to d8cea3f , maybe 8616c16 )

[image]

btw, the device missing issue is fixed ! Thank you !

Smanar, 2020-09-10:
This log was taken at startup ?

Because it s strange you have so much "missing manufacture name".

I don't find information about this one,
1025 > 0x04 0x01 mean an enum value, like for a "table state", so it can't be a battery level but can be a lowbattery state.

czm97176, 2020-09-10:
> 
> 
> This log was taken at startup ?
> Because it s strange you have so much "missing manufacture name".

Sorry i may misleading you, i filter out the log by "tuya", so the log entry is not display at all in sequence.
The "tuya debug 4" and "tuya debug 5"  happened on 18:26 and 18:30 but the startup time is 13:04.
and it only happened once from these couple days

> I don't find information about this one,
> 1025 > 0x04 0x01 mean an enum value, like for a "table state", so it can't be a battery level but can be a lowbattery state.

Maybe i try to charge the battery and see if there any things comes out

czm97176, 2020-09-10:
@Smanar i got another "Tuya log"

[image]

[23 more replies]

closed 2020-10-01 by manup