dresden-elektronik/deconz-rest-plugin #3226: Philips Hue: Motion detector. Value for temperature and light level is always 0 and will not be updated.

DenBir, 2020-09-02:
## Describe the question or issue you are having:
The Philips Hue motion detector sensor shows always 0 value for temperature and light level and will not be updated.
FW version, motion and battery condition are displayed correctly.

If I connect the same motion sensor with the Philips Hue bridge, then I see the correct values for temperatur and light level.
> Test with Android App - Hue Essentials

## Screenshots
deCONZ App:
[image]

Android App - Hue Essentials: Compare Hue Bridge vs. ConBee II:
[image]

## Environment
deCONZ App:
[image]

deCONZ GUI:
[image]

 - Host system: x86 Server with Proxmox as hypervisor
 - Running method: Virtual Machine - Debian GNU/Linux 10 (4.19.0-10-amd64)
 - Firmware version: 26580700
 - deCONZ version: 2.05.78
 - Device: ConBee II
 - Do you use an USB extension cable: no

- Philips Hue motion sensor FW: 6.1.1.27575

### deCONZ Logs
Here I tried to test the light sensor with a flashlight:
[debug.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5161691/debug.log)

The values for "temperature" and "lux" are available!
[image]

## Additional context
deCONZ GUI Nodes:
[image]

SwoopX, 2020-09-02:
Just wait. It might take a while for the values to be updated, depending on the actual attribute reporting configuration.

DenBir, 2020-09-02:
I'm waiting since yesterday...

Other sensors as Aqara Temperature Humidity Sensor works fine and the value for temperature was also immediately displayed correctly.

Mimiix, 2020-09-02:
This looks a awful lot like https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2968

They are both on .78.

@ebaauw @SwoopX Could any of your changes affect this?

KennethLavrsen, 2020-09-02:
It sure looks the same.
For the record. I am since my report now on 2.05.80.

ebaauw, 2020-09-02:
Can you double-check that the values update when you read the attributes of the corresponding clusters in the GUI?

We’ve had numerous issues about this: sometimes the REST API plugin doesn’t setup (all) the required bindings and/or attribute reporting after pairing a device.  The more configuration that needs to be done, the bigger the chances you’re hit by this.  For the Hue motion sensor that’s five (!) clusters; for the Xiaomi it’s none at all, as their firmware is hard-coded to report to the coordinator.

As mentioned with the previous issues: I don’t know what’s causing this exactly; the pairing logic is cursed and even some temporary interference could cause this.  Fixing this would require a complete refactor of the pairing logic.

It’s common for the API plugin to do the setup in a later state.  Typically reading the attributes of the corresponding cluster triggers this; you might also try and power-cycle the sensor, or repair it _without resetting it_ (as that would delete the bindings and reporting config that were already created).  Alternatively, do the setup manually in the GUI.

DenBir, 2020-09-02:
I've double-check the values in the deCONZ GUI. Booth values are "0":
[image]
[image]

How can I repair the device?
How can I setup the device manually in the GUI?

SwoopX, 2020-09-02:
**How to manually bind and enable attribute reporting on the example of the current position for windows covering devices**

**1. Manually bind a cluster**
- In deconz GUI, select _Panels_ and then _Bind dropbox_.
- For your device as well as the coordinator (blue node), press the most right bullet to expand the available clusters.
- For the coordinator: Drag&drop 01 Home Automation Endpoint as destination.
- For the device: Select the windows covering cluster as source, also drag&drop it. Take note that this must be a server cluster (colored blue).
- For grey colored nodes, make sure the device is awake by triggering one of its functions or pushing the physical button once.
- Press `bind`.

**2. Manually set up attribute reporting**
Then you need to set up attribute reporting. In most cases, there are default settings already defined and should be placed into action either after a deconz restart or power-cycling the device. However, if values do not get updated upon the taken action or not at all, attributes reporting has presumably not been set up properly.

Select windows covering cluster of your device, double click on 0x0008 (current position).
On the lower end of the dialog, enter the values

Min. report interval: 1
Max. report interval: 300
Reportable change: 1

then press `write config`. For grey colored nodes, make sure the device is awake by triggering one of its functions or pushing the physical button once.

DenBir, 2020-09-02:
Thanks for this information, but I think I can't this correct reproduce.

I tried this steps to bind 0400 and 0402:
[image]

[image]
[image]

The ID for 0x0008 can I found only by "Basic" on the endpoint 01 and 02.
Was the "click way" correct? If I click on "write config", then I get the message "unsupported attribute".
[image]

But I check again the sensor and now I see the correct values!
[image]

To be honest, I don't quite understand.
But it works!

Was it really just the additional binding?

Thank you for the great support!

ebaauw, 2020-09-02:
> I've double-check the values in the deCONZ GUI. Booth values are "0":

You need to press _Read_.  Might take a few tries.

> Was it really just the additional binding?

The REST API plugin probably picked up the response to the bind request and sent the report configuration.

You might want to double-check (_Read Config_ of) the _Battery Percentage Remaining_ attribute of the _Power Configuration_ cluster (should be 7200/7200/0), and of the _LED indication_ and _Usertest_ attributes of the _Basic_ cluster (should both be 5/7200).

DenBir, 2020-09-03:
The value (after click read) for Battery Percentage Remaining is **200**.
[image]

The value (after click read) for LED indication and Usertest is **false**.
[image]

ebaauw, 2020-09-03:
> The value (after click read) for Battery Percentage Remaining is 200.

Then it has probably been setup correctly.  Double-check the reporting config in the _Attribute Editor_ window to be sure.

> The value (after click read) for LED indication and Usertest is false.

That doesn't mean much, as these are the default values.  Also, I don't think the GUI reads these attributes when you press _Read_ (it only reads the first couple of attributes on devices without _Receiver On When Idle_).

closed 2020-09-03 by DenBir