dresden-elektronik/deconz-rest-plugin #3227: Deconz plugin to control devices

Rania-M, 2020-09-03:
Hi,
i would like to use the deconz plugin to control home devices using a qt application, i will possibly need to add new devices to the plugin, that being said, i would like to know what is the main used method, do you guys add the plugin directly in the code (as a .pri project inside a .pro one), or do you modify the plugin, compile it in the raspberry and then call it from the application in a REST request kind of way, i started with the first idea but i don't think it's supposed to be this way
If anyone could answer me quickly that would be great !

ebaauw, 2020-09-03:
The IDEA of the REST API plugin is to provide a REST API to external applications.

If you want to integrate control into the deCONZ GUI, you cannot use the REST API.  Use the C++ interface to create your own plugin.  Unfortunately, the documentation for that interface has gone MIA, but is self-documented in the header files, that the `deconz-dev` package installs into `/usr/include/deconz`.  See the "deCONZ for Dummies" in the WiKi.  There's also a link to a sample plugin on GitHub.

I fail to understand why you would want a Qt application to control home devices.  You can use the rules provided by the REST API for automations, and there's plenty of home automation systems that integrate with deCONZ using the REST API.

Rania-M, 2020-09-03:
Thank you for your answer, i would like to know where can i find those REST API rules you are talking about, as for the automation, i need to control the devices in a certain way which is why i will have to use an external application to call the plugin , i downloaded the plugin though as i may have to add new devices to it

closed 2020-10-04 by stale[bot]