dresden-elektronik/deconz-rest-plugin #3229: Xiaomi sensors (Motion/Water) not reporting temp to HA

pergolafabio, 2020-09-03:
## Describe the bug
Not sure if its a bug, but all my xiaomi sensors like water leak and motion are reporting also temperature in phoscon webapp
But in HA i only see the state + battery, not the temp value

## Environment
 - Host system: (Raspberry Pi / PC / NAS)  : HassOS VM
 - Running method: HA Deconz add-on 
 - Firmware version: 26330500
 - deCONZ version: 2.05.79
 - Device: ConBee I
 - Do you use an USB extension cable: Yes

Mimiix, 2020-09-03:
Thanks for opening with the template! Appreciate it :)

The RESTApi should report it to HA as Phoscon uses the same api. However, these temperature's are not really accurate. They are inacurate and not reliable.

pergolafabio, 2020-09-03:
No ;)

Yeah I noticed now, temp is totally wrong and useless indeed...
I hope my humidity sensors are correct though

closed 2020-09-03 by pergolafabio

Mimiix, 2020-09-03:
@pergolafabio 

Oh those are! The temp sensors seem to be some type of "internal temperature".  The Dedicated sensors (humid , or if the device is a actual temp sensor) are pretty acurate!

pergolafabio, 2020-09-03:
Thank God ;)