dresden-elektronik/deconz-rest-plugin #3230: deCONZ service could not been starte because of Qt issues

thomasklein1975, 2020-09-04:
Hi,

maybe somebody can help me. Untill now i was able to start service. Seems so that some package are missing.
I already tried "sudo apt -f install" after following instructions from website. Port changed for testing from 80 to 8090 
Even GUI on Raspi could not be started while using vnc.

Here are the logs:

`Sep 04 08:01:40 pi systemd[1]: deconz-gui.service: main process exited, code=exited, status=1/FAILURE
Sep 04 08:01:40 pi systemd[1]: Unit deconz-gui.service entered failed state.
Sep 04 08:02:10 pi systemd[1]: deconz-gui.service holdoff time over, scheduling restart.
Sep 04 08:02:10 pi systemd[1]: Stopping deCONZ: ZigBee gateway -- GUI/REST API...
Sep 04 08:02:10 pi systemd[1]: Starting deCONZ: ZigBee gateway -- GUI/REST API...
Sep 04 08:02:10 pi systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
Sep 04 08:02:10 pi deCONZ[2460]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5SerialPort.so.5: no version information available (required by /usr/bin/deCONZ
Sep 04 08:02:10 pi deCONZ[2460]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Network.so.5: no version information available (required by /usr/bin/deCONZ)
Sep 04 08:02:10 pi deCONZ[2460]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/bin/deCONZ)
Sep 04 08:02:10 pi deCONZ[2460]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5: no version information available (required by /usr/bin/deCONZ)
Sep 04 08:02:10 pi deCONZ[2460]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Gui.so.5: no version information available (required by /usr/bin/deCONZ)
Sep 04 08:02:10 pi deCONZ[2460]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Widgets.so.5: no version information available (required by /usr/bin/deCONZ)
Sep 04 08:02:10 pi deCONZ[2460]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Gui.so.5: no version information available (required by /usr/lib/libdeCONZ.so.
Sep 04 08:02:10 pi deCONZ[2460]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5SerialPort.so.5: no version information available (required by /usr/lib/libdeC
Sep 04 08:02:10 pi deCONZ[2460]: /usr/bin/deCONZ: /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5: no version information available (required by /usr/lib/libdeCONZ.so
Sep 04 08:02:10 pi systemd[1]: deconz-gui.service: main process exited, code=exited, status=1/FAILURE
Sep 04 08:02:10 pi systemd[1]: Unit deconz-gui.service entered failed state.
Sep 04 08:02:18 pi systemd[1]: Starting deCONZ: ZigBee gateway -- GUI/REST API...
Sep 04 08:02:40 pi systemd[1]: deconz-gui.service holdoff time over, scheduling restart.
Sep 04 08:02:40 pi systemd[1]: Stopping deCONZ: ZigBee gateway -- GUI/REST API...
Sep 04 08:02:40 pi systemd[1]: Starting deCONZ: ZigBee gateway -- GUI/REST API...
Sep 04 08:02:40 pi systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.`

It would be grat to get support/help. I want to ave my ikea smart home devices integrated in my fhem :-)

Greetings & thx in advance
Thomas

Mimiix, 2020-09-04:
Hi @thomasklein1975 

I am closing this issue as a issue template is not used. Please stick to the issue templates.

closed 2020-09-04 by Mimiix

thomasklein1975, 2020-09-04:
sorry didnt get it. what did i wrong?

Mimiix, 2020-09-04:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new/choose