dresden-elektronik/deconz-rest-plugin #3231: Hampton Bay Fan Module - Light dimmer

jjlawren, 2020-09-05:
## Describe the bug

I've installed a Hampton Bay fan controller and paired with my Raspbee. I'm able to turn the device/fan/light on and off and also control the fan speed, but I'm not able to control the light dimmer. It properly reports brightness levels via REST in `state["bri"]` when dimming using its remote.

I can publish values to the "bri" value over REST which updates the reported state, but it doesn't actually change the device's brightness. In Deconz when I try to use "Move to Level" in the 0008 Level Control cluster, i get `UNSUP_CLUSTER_COMMAND` in response. I'm unsure if this is a device or REST-API issue.

## Steps to reproduce the behavior

See above.

## Expected behavior

Light dimmer should be controllable over REST-API.

## Screenshots

## Environment

 - Host system: Raspberry Pi 3b+
 - Running method: Raspbian
 - Firmware version: 26350500
 - deCONZ version: 2.05.80
 - Device: RaspBee I

### deCONZ Logs

## Additional context

ebaauw, 2020-09-05:
> In Deconz when I try to use "Move to Level" in the 0008 Level Control cluster, i get UNSUP_CLUSTER_COMMAND in response.

That would mean the controller (firmware) doesn’t support dimming.

jjlawren, 2020-09-05:
On the fan module device or the Raspbee?

ebaauw, 2020-09-05:
The fan module.

jjlawren, 2020-09-07:
According to [this SmartThings thread](https://community.smartthings.com/t/hampton-bay-zigbee-fan-controller/47463/697), it should be possible to control after it's been dimmed using the RF remote at least once. I've tried this and haven't had success. Could there be another step or refresh necessary on the Deconz side if the device capabilities change?

ebaauw, 2020-09-08:
Until the module reacts to _Move to Level_, there’s nothing deCONZ can do.

If I read the SmartThing thread, you need to set the dip switch on the module.  Maybe you need to power cycle the module after that?

I’m sorry, the module is not usable outside North America, and the don’t have an international version, so I cannot help you with that.

jjlawren, 2020-09-17:
Found something that might help. The "Move to Level" command is not supported, but I _am_ able to adjust the brightness of the light by using the "Move to Level (with On/Off)" command in the desktop app. I can set various values for 'Level' and it updates accordingly each time I call the command.

Setting the value to 0 or 255 changes to off/on immediately. Setting to any value in-between has a ~1s transition. The "Transition time" option for the command doesn't appear to have any effect.

However, I've tried to replicate this with the REST API and did not have success. Here's how it behaves:

1. Send `{"bri": 50, "on": true}` -> Turns on light at ~20% brightness
2. Send `{"bri": 250, "on": true}` -> No effect
3. Send `{"on": false}` -> Turns off light
4. Send `{"bri": 250, "on": true}` -> Turns on light at ~20% brightness (?)

ebaauw, 2020-09-17:
> The "Move to Level" command is not supported, but I am able to adjust the brightness of the light by using the "Move to Level (with On/Off)" command in the desktop app.

That’s a first.  I changed the setting of the light state to be more in line with the Hue bridge, so an `{"on": true, "bri": 200}` sends a _Move to Level (with On/Off)_ for level 2 with a transition time of 0, followed by a _Move to Level_ for level 200 with a default transition time of 0.4s.

Could you try adding the module to a group and sending the body to the group action instead?  That still sends a _Move to Level (with On/Off)_ to level 200 with the default transition time.

jjlawren, 2020-09-18:
Yep, that works. Targeting the group with the REST API calls behaves similarly to controlling it with the "Move to Level (with On/Off" command in the GUI. Does this device need special handling in the REST API?

aofc, 2020-09-30:
it supports bri_inc commands and has brightness range of 50 (0-50: off) to 254 (setting it to 255 reports back 254).

jjlawren, 2020-10-22:
I believe this issue is still valid.

jjlawren, 2020-11-14:
@ebaauw based on what's been provided, can the dimming handling be improved for this device?

ebaauw, 2020-11-14:
In theory: yes.  In practice: I'm no longer touching that code; every change seems to do more harm than good.  Really need a capabilities model as proposed in the APIv2 discussions.

closed 2020-12-31 by stale[bot]